espressif/esp-idf #9533: [v4.4.2][ESP32-S3] memory leak in readdir() for large directories

chipweinberger, 2022-08-09:
ESP-IDF v4.4.2
ESP32-S3 with 2MB PSRAM 

**When calling readdir on large directories (>380 subfolders), a boundless amount of internal memory will be used. In my case, 160KB of internal RAM were used before it crashed.**

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/9285770/sdkconfig.txt)

<details>
  <summary>Logs</summary>
```
I (36268) pd api file: sending directory (detailed): '/sd/JAMC/2022/device-b1xqnf'
V (36278) pd file utils: pd_fileExists /sd/JAMC/2022/device-b1xqnf
V (36288) pd file utils: --pd_fileExists YES /sd/JAMC/2022/device-b1xqnf
V (36298) pd file utils: pd_fileExists /sd/JAMC/2022/device-b1xqnf
V (36308) pd file utils: --pd_fileExists YES /sd/JAMC/2022/device-b1xqnf
V (36308) pd file utils: pd_isDirectory /sd/JAMC/2022/device-b1xqnf
V (36318) pd file utils: --pd_isDirectory YES
V (36318) pd file utils: pd_opendir /sd/JAMC/2022/device-b1xqnf/
V (36328) pd file utils: --pd_opendir dir: 0x3de64a9c
V (36328) pd file utils: pd_readdir dir: 0x3de64a9c
V (36338) pd file utils: --pd_readdir dp: 0x0x3de64bec
V (36338) pd file utils: pd_readdir dir: 0x3de64a9c
V (36338) pd file utils: --pd_readdir dp: 0x0x3de64bec
V (36348) pd file utils: pd_readdir dir: 0x3de64a9c
V (36358) pd file utils: --pd_readdir dp: 0x0x3de64bec
V (36358) pd file utils: pd_readdir dir: 0x3de64a9c
V (36358) pd file utils: --pd_readdir dp: 0x0x3de64bec
V (36358) pd file utils: pd_readdir dir: 0x3de64a9c
V (36378) pd file utils: --pd_readdir dp: 0x0x3de64bec
V (36378) pd file utils: pd_readdir dir: 0x3de64a9c
V (36378) pd file utils: --pd_readdir dp: 0x0x3de64bec
V (36388) pd file utils: pd_readdir dir: 0x3de64a9c
V (36388) pd file utils: --pd_readdir dp: 0x0x3de64bec
V (36398) pd file utils: pd_readdir dir: 0x3de64a9c
V (36408) pd file utils: --pd_readdir dp: 0x0x3de64bec
V (36408) pd file utils: pd_readdir dir: 0x3de64a9c
V (36408) pd file utils: --pd_readdir dp: 0x0x3de64bec
V (36408) pd file utils: pd_readdir dir: 0x3de64a9c
V (36418) pd file utils: --pd_readdir dp: 0x0x3de64bec
V (36428) pd file utils: pd_readdir dir: 0x3de64a9c
V (36428) pd file utils: --pd_readdir dp: 0x0x3de64bec
V (36438) pd file utils: pd_readdir dir: 0x3de64a9c
V (36438) pd file utils: --pd_readdir dp: 0x0x3de64bec
V (36438) pd file utils: pd_readdir dir: 0x3de64a9c
V (36448) pd file utils: --pd_readdir dp: 0x0x3de64bec
V (36458) pd file utils: pd_readdir dir: 0x3de64a9c
V (36458) pd file utils: --pd_readdir dp: 0x0x3de64bec
V (36458) pd file utils: pd_readdir dir: 0x3de64a9c
V (36468) pd file utils: --pd_readdir dp: 0x0x3de64bec
V (36478) pd file utils: pd_readdir dir: 0x3de64a9c
V (36478) pd file utils: --pd_readdir dp: 0x0x3de64bec
V (36478) pd file utils: pd_readdir dir: 0x3de64a9c
V (36488) pd file utils: --pd_readdir dp: 0x0x3de64bec
V (36498) pd file utils: pd_readdir dir: 0x3de64a9c
V (36498) pd file utils: --pd_readdir dp: 0x0x3de64bec
V (36498) pd file utils: pd_readdir dir: 0x3de64a9c
V (36508) pd file utils: --pd_readdir dp: 0x0x3de64bec
V (36508) pd file utils: pd_readdir dir: 0x3de64a9c
V (36508) pd file utils: --pd_readdir dp: 0x0x3de64bec
V (36528) pd file utils: pd_readdir dir: 0x3de64a9c
V (36528) pd file utils: --pd_readdir dp: 0x0x3de64bec
V (36528) pd file utils: pd_readdir dir: 0x3de64a9c
V (36528) pd file utils: --pd_readdir dp: 0x0x3de64bec
V (36538) pd file utils: pd_readdir dir: 0x3de64a9c
V (36548) pd file utils: --pd_readdir dp: 0x0x3de64bec
V (36548) pd file utils: pd_readdir dir: 0x3de64a9c
V (36548) pd file utils: --pd_readdir dp: 0x0x3de64bec
V (36558) pd file utils: pd_readdir dir: 0x3de64a9c
V (36558) pd file utils: --pd_readdir dp: 0x0x3de64bec
V (36568) pd file utils: pd_readdir dir: 0x3de64a9c
[... 662 lines cut ...]
0x420b53f1: httpd_parse_req at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_http_server/src/httpd_parse.c:659 (discriminator 15)

0x420b5534: httpd_req_new at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_http_server/src/httpd_parse.c:787

0x420b2e7a: httpd_sess_process at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_http_server/src/httpd_sess.c:412 (discriminator 15)

0x420b1f10: httpd_process_session at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_http_server/src/httpd_main.c:178 (discriminator 15)

0x420b284f: httpd_sess_enum at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_http_server/src/httpd_sess.c:50 (discriminator 1)

0x420b21d7: httpd_server at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_http_server/src/httpd_main.c:228

0x420b2260: httpd_thread at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/esp_http_server/src/httpd_main.c:250 (discriminator 15)

0x4038704e: vPortTaskWrapper at /Volumes/User/MBP-Google-Drive/jamcorder/firmware/esp-idf/components/freertos/port/xtensa/port.c:131

ELF file SHA256: ae69775eeace0cef

I (3766) esp_core_dump_flash: Save core dump to flash...
I (3772) esp_core_dump_flash: Erase flash 32768 bytes @ 0xc000
I (4317) esp_core_dump_flash: Write end offset 0x7164, check sum length 4
I (4317) esp_core_dump_flash: Core dump has been saved to flash.
CPU halted.
```
<details>

chipweinberger, 2022-08-09:
I added some debug code to track memory over time. 

We begin at 258247 bytes free, and end at 88183 bytes free at the last readdir.
```
I (12991) pd api file: sending directory (detailed): '/sd/JAMC/2022/device-b1xqnf'
// opendir()
// followed by lots of calls to readdir
malloc 512 bytes caps 8, fr 258247
free 257719
malloc 512 bytes caps 8, fr 258247
... ~180 more calls....
free 85863
malloc 512 bytes caps 8, fr 84599
free 84071
malloc 512 bytes caps 8, fr 82807
```

After closedir, **2000** calls to free are made! crazy.
```
// closedir()
free 81463
free 81463
free 81503
... ~2000 more....
free 258379
free 258435
free 258483
```

**Debug Code**
```
IRAM_ATTR void heap_caps_free( void *ptr)
{
    esp_rom_printf(DRAM_STR("free %u bytes\n"), heap_caps_get_free_size(MALLOC_CAP_INTERNAL));
    ...
}

IRAM_ATTR void *heap_caps_malloc( size_t size, uint32_t caps){

    esp_rom_printf(DRAM_STR("malloc %u bytes caps %u, fr %u\n"), size, caps, heap_caps_get_free_size(MALLOC_CAP_INTERNAL));
  ..
}
```

chipweinberger, 2022-08-09:
This is a super critical issue for me.

igrr, 2022-08-09:
@chipweinberger Unfortunately i haven't been able to reproduce the issue based on your description, yet.

Could you please share some code which we can use to reproduce the issue?

Here's what I have tried:

1. On a PC, create a directory `test` on an SD card, fill it with 1000 files, named testN.bin with N=1...1000.
2. On the ESP, use SD card example, modifying it to run the following code:
    ```c++
    DIR* dir = opendir("/sdcard/test");
    assert(dir != NULL);

    while (true) {
        struct dirent* de = readdir(dir);
        if (!de) {
            break;
        }
        printf("Found subdirectory: %s. Heap: %" PRIu32 "\n", de->d_name, esp_get_free_heap_size());
    }
    closedir(dir);
    ```

The heap size was stable at 212960 bytes throughout the test. (This was on an S2 board.)

Do you think you can reduce your application to something similarly simple, which still reproduces the issue?

chipweinberger, 2022-08-09:
Appreciate the effort Ivan.

Yes let me see If I can repro on a more minimal example.

perhaps S3 + my sdkconfig is required.

chipweinberger, 2022-08-09:
**You are right. There is no issue. It was my fault... sigh.**

I was adding the readdir filenames to a cJSON Object, and I completely overlooked how much memory cJSON requires to store these strings. 

My Code:
```
    // iterate
    pd_filename_t filepathOut;
    while (pd_folder_iter_next(dirIter, &filepathOut, error)) {

        cJSON* jFile = cJSON_CreateObject(); // THIS IS ALLOCATING WITH INTERNAL MEMORY!
        cJSON_AddStringToObject(jFile, PD_KEY_FILENAME, pd_basename(filepathOut.v));
        cJSON_AddItemToArray(jArray, jFile);
    }
```

Sorry for not testing on my side further. I will close this.

chipweinberger, 2022-08-09:
As an aside, I now need a way to force cJSON to use SPI memory =)

closed 2022-08-09 by chipweinberger