espressif/esp-idf #9534: When the A2DP-sink uses PSRAM as a buffer, an error occurs when saving the pairing code to NVS

maplerian, 2022-08-09:
- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

## Environment

- Development Kit:      none
- Kit version (for WroverKit/PicoKit/DevKitC):  none
- Module or chip used: ESP32
- IDF version (run ``git describe --tags`` to find it):
    v5.0-dev-3654-gc2ccc383da
- Build System:        idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
     (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System:     Linux
- Using an IDE?: Yes(VScode)
- Power Supply:         external 3.3V

## Problem Description
During Bluetooth pairing, the pairing code cannot be saved, which will cause the device to restart and fails to be saved.

### Expected Behavior
Correctly maintain matchmaking codes to NVS

### Actual Behavior
The device unexpectedly restarts when saving the matching code. Procedure

### Steps to reproduce
1. Select BT/BLE will first malloc the memory from the PSRAM in SDKConfig
2. Select Use dynamic memory allocation in BT/BLE stack in SDKConfig
3. Set Bluetooth event (callback to application) task stack size 4096
4. Enable PSRAM Config

### Code to reproduce this issue
a2dp-sink demo

## Debug Logs

```
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6936
load:0x40078000,len:15360
load:0x40080400,len:3840
0x40080400: _init at ??:?

entry 0x4008064c
I (161) boot: ESP-IDF v5.0-dev-3654-gc2ccc383da-dirty 2nd stage bootloader
I (161) boot: compile time 10:36:28
I (163) boot: chip revision: 3
I (167) boot.esp32: SPI Speed      : 80MHz
I (171) boot.esp32: SPI Mode       : DIO
I (176) boot.esp32: SPI Flash Size : 8MB
W (181) boot.esp32: PRO CPU has been reset by WDT.
W (186) boot.esp32: WDT reset info: PRO CPU PC=0x40081b8d
0x40081b8d: panicHandler at /home/maple/env/esp/esp-idf/components/esp_system/port/panic_handler.c:216

W (192) boot.esp32: WDT reset info: APP CPU PC=0x40081b8d
0x40081b8d: panicHandler at /home/maple/env/esp/esp-idf/components/esp_system/port/panic_handler.c:216

I (198) boot: Enabling RNG early entropy source...
I (204) boot: Partition Table:
I (207) boot: ## Label            Usage          Type ST Offset   Length
I (215) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (222) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (230) boot:  2 factory          factory app      00 00 00010000 00100000
[... 160 lines cut ...]
I (77) boot:  2 factory          factory app      00 00 00010000 00100000
I (84) boot: End of partition table
I (88) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1dda0h (122272) map
I (134) esp_image: segment 1: paddr=0002ddc8 vaddr=3ffbdb60 size=02250h (  8784) load
I (137) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=7e950h (518480) map
I (296) esp_image: segment 3: paddr=000ae978 vaddr=3ffbfdb0 size=01628h (  5672) load
I (298) esp_image: segment 4: paddr=000affa8 vaddr=40080000 size=190b8h (102584) load
I (338) esp_image: segment 5: paddr=000c9068 vaddr=50000000 size=00010h (    16) load
I (350) boot: Loaded app from partition at offset 0x10000
I (350) boot: Disabling RNG early entropy source...
I (362) quad_psram: This chip is ESP32-D0WD
I (363) esp_psram: Found 8MBit SPI RAM device
I (363) esp_psram: Speed: 80MHz
I (367) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
I (374) cpu_start: Pro cpu up.
I (378) cpu_start: Starting app cpu, entry point is 0x400814e4
0x400814e4: call_start_cpu1 at /home/maple/env/esp/esp-idf/components/esp_system/port/cpu_start.c:147
```

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder)
- [x] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)
[a2dp_sink.zip](https://github.com/espressif/esp-idf/files/9286232/a2dp_sink.zip)

maplerian, 2022-08-09:
There is also a problem that the SPI write exception is triggered when the Bluetooth thread is running on CPU1

maplerian, 2022-08-09:
```
I (1492) BT_AV: Get delay report value: delay_value: 1200 * 1/10 ms
I (1492) BT_AV: Set delay report value: success, delay_value: 1250 * 1/10 ms
W (6742) BT_HCI: hcif conn complete: hdl 0x81, st 0x0
W (6742) BT_HCI: hcif link supv_to changed: hdl 0x81, supv_to 8000
I (7152) BT_AV: ESP_BT_GAP_CFM_REQ_EVT Please compare the numeric value: 524963
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Core  0 register dump:
PC      : 0x4008bfad  PS      : 0x00060435  A0      : 0x8008d3cb  A1      : 0x3ffb7080  
0x4008bfad: spi_flash_op_block_func at /home/maple/env/esp/esp-idf/components/spi_flash/cache_utils.c:132 (discriminator 1)

A2      : 0x00000000  A3      : 0x00000001  A4      : 0x3ffbf970  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00060f23  A8      : 0x00000000  A9      : 0x00000001  
A10     : 0x3ffc15ec  A11     : 0x00001000  A12     : 0x8400124c  A13     : 0x3ffbf924  
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000001  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x4008bfaa:0x3ffb70800x4008d3c8:0x3ffb70a0 0x40094fc5:0x3ffb70c0 
0x4008bfaa: spi_flash_op_block_func at /home/maple/env/esp/esp-idf/components/spi_flash/cache_utils.c:131

0x4008d3c8: ipc_task at /home/maple/env/esp/esp-idf/components/esp_system/esp_ipc.c:81

0x40094fc5: vPortTaskWrapper at /home/maple/env/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:141

ELF file SHA256: 00f86c59bc0c4b88

Rebooting...
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6936
load:0x40078000,len:15360
load:0x40080400,len:3840
0x40080400: _init at ??:?

entry 0x4008064c
I (26) boot: ESP-IDF v5.0-dev-3654-gc2ccc383da-dirty 2nd stage bootloader
I (27) boot: compile time 10:36:28
I (28) boot: chip revision: 3
I (32) boot.esp32: SPI Speed      : 80MHz
`

maplerian, 2022-08-09:
Also, all threads involved in FLASH reading and writing must have a stack size of at least/more than 4096.
Threads using the ESP_LOG component must also have a thread stack size of at least/more than 3072 to work properly.
There is also an I2C problem, which is also a problem.
These two problems make me very upset. I have to open a thread for FLASH writing and I2C reading and writing, and I have to use printf instead of ESP_LOG component.

maplerian, 2022-08-09:
Pairing code saving is occasionally successful, but the probability is very small. Successful saving is usually the first saving after the firmware is swiped in

blueMoods, 2022-08-09:
Hi @maplerian 

Thanks for your information,  The first issue have been fixed, refer to https://github.com/espressif/esp-idf/commit/9a6ccb7f579a9d0162b32908cf9e57f89f771985 , Please test with it, thanks.

maplerian, 2022-08-09:
```
W (17967) BT_HCI: hcif conn complete: hdl 0x80, st 0x0
W (17977) BT_HCI: hcif link supv_to changed: hdl 0x80, supv_to 8000
I (18377) BT_AV: ESP_BT_GAP_CFM_REQ_EVT Please compare the numeric value: 379409
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Core  0 register dump:
PC      : 0x400935c3  PS      : 0x00060835  A0      : 0x800837f8  A1      : 0x3ffbe070  
0x400935c3: xQueueReceiveFromISR at /home/maple/env/esp/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1953 (discriminator 1)

A2      : 0x3f800e84  A3      : 0x00000000  A4      : 0x3ffbe0c0  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0xbad00bad  A8      : 0x80087291  A9      : 0x3ffbe1a0  
A10     : 0x00000018  A11     : 0x00000000  A12     : 0x3ffc6ce0  A13     : 0x3ffc6c90  
A14     : 0x4008c774  A15     : 0x3ffc6d40  SAR     : 0x00000020  EXCCAUSE: 0x00000007  
0x4008c774: flash_end_flush_cache at /home/maple/env/esp/esp-idf/components/spi_flash/esp_flash_api.c:149

EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace:0x400935c0:0x3ffbe0700x400837f5:0x3ffbe0a0 0x4008d887:0x3ffbe0c0 0x4008db43:0x3ffbe0f0 0x4008d831:0x3ffbe120 0x400849c9:0x3ffbe140 0x40084fba:0x3ffbe160 0x40086c69:0x3ffbe1a0 0x4008728e:0x3ffbe1c0 0x4008adce:0x3ffbe1f0 0x4008b783:0x3ffbe210 0x40082e2d:0x3ffbe230 0x4008c299:0x3ffc6bf0 0x4008c6be:0x3ffc6c20 0x40097f66:0x3ffc6c50 0x4008c7a1:0x3ffc6c70 0x4008cf4f:0x3ffc6c90 0x40128e79:0x3ffc6ce0 0x4012bf87:0x3ffc6d00 0x4012c5e1:0x3ffc6d20 0x4012c84a:0x3ffc6d40 0x4012b393:0x3ffc6db0 0x4012b8d5:0x3ffc6e30 0x4012bd99:0x3ffc6ea0 0x4012add7:0x3ffc6ec0 0x40107025:0x3ffc6ef0 0x400f841f:0x3ffc6f50 0x400dab1d:0x3ffc6f70 0x400f8715:0x3ffc6fc0 0x400f8b46:0x3ffc7110 0x400f3745:0x3ffc71b0 0x400f4c55:0x3ffc71d0 0x40094fc5:0x3ffc71f0 
0x400935c0: xQueueReceiveFromISR at /home/maple/env/esp/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1953 (discriminator 1)

0x400837f5: semphr_take_from_isr_wrapper at /home/maple/env/esp/esp-idf/components/esp_wifi/esp32/esp_adapter.c:288

0x4008d887: coex_bb_reset_unlock at ??:?

0x4008db43: coex_core_release at ??:?

0x4008d831: coex_bt_release at ??:?

0x400849c9: coex_bt_release_wrapper at /home/maple/env/esp/esp-idf/components/bt/controller/esp32/bt.c:1133

0x40084fba: coex_sw_event_from_isr at ??:?

0x40086c69: coex_classic_bt_release at ld_fm.c:?

0x4008728e: r_ld_fm_frame_isr at ??:?

0x4008adce: r_rwbt_isr at ??:?

0x4008b783: r_rwbtdm_isr_wrapper at intc.c:?

0x40082e2d: _xt_lowint1 at /home/maple/env/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1117

0x4008c299: mmu_ll_read_entry at /home/maple/env/esp/esp-idf/components/hal/esp32/include/hal/mmu_ll.h:111
 (inlined by) is_page_mapped_in_cache at /home/maple/env/esp/esp-idf/components/spi_flash/flash_mmap.c:466

0x4008c6be: spi_flash_check_and_flush_cache at /home/maple/env/esp/esp-idf/components/spi_flash/flash_mmap.c:498

0x40097f66: memspi_host_flush_cache at /home/maple/env/esp/esp-idf/components/spi_flash/memspi_host_driver.c:131

0x4008c7a1: flash_end_flush_cache at /home/maple/env/esp/esp-idf/components/spi_flash/esp_flash_api.c:159

0x4008cf4f: esp_flash_write at /home/maple/env/esp/esp-idf/components/spi_flash/esp_flash_api.c:922 (discriminator 1)

0x40128e79: esp_partition_write at /home/maple/env/esp/esp-idf/components/spi_flash/partition_target.c:80

0x4012bf87: nvs::NVSPartition::write(unsigned int, void const*, unsigned int) at /home/maple/env/esp/esp-idf/components/nvs_flash/src/nvs_partition.cpp:59

0x4012c5e1: nvs::Page::writeEntry(nvs::Item const&) at /home/maple/env/esp/esp-idf/components/nvs_flash/src/nvs_page.cpp:101

0x4012c84a: nvs::Page::writeItem(unsigned char, nvs::ItemType, char const*, void const*, unsigned int, unsigned char) at /home/maple/env/esp/esp-idf/components/nvs_flash/src/nvs_page.cpp:247

0x4012b393: nvs::Storage::writeMultiPageBlob(unsigned char, char const*, void const*, unsigned int, nvs::VerOffset) at /home/maple/env/esp/esp-idf/components/nvs_flash/src/nvs_storage.cpp:221 (discriminator 4)

0x4012b8d5: nvs::Storage::writeItem(unsigned char, nvs::ItemType, char const*, void const*, unsigned int) at /home/maple/env/esp/esp-idf/components/nvs_flash/src/nvs_storage.cpp:320

0x4012bd99: nvs::NVSHandleSimple::set_blob(char const*, void const*, unsigned int) at /home/maple/env/esp/esp-idf/components/nvs_flash/src/nvs_handle_simple.cpp:52

0x4012add7: nvs_set_blob at /home/maple/env/esp/esp-idf/components/nvs_flash/src/nvs_api.cpp:425

[... 70 lines cut ...]
I (914) heap_init: Initializing. RAM available for dynamic allocation:
I (921) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (927) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (933) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (939) heap_init: At 3FFC32A8 len 0001CD58 (115 KiB): DRAM
I (945) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (952) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (958) heap_init: At 400990B8 len 00006F48 (27 KiB): IRAM
I (964) esp_psram: Adding pool of 4096K of external SPI memory to heap allocator
I (973) spi_flash: detected chip: generic
I (977) spi_flash: flash io: dio
I (982) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (992) esp_psram: Reserving pool of 4K of internal memory for DMA/internal allocations
I (1042) BTDM_INIT: BT controller compile version [5688ed5]
I (1042) system_api: Base MAC address is not set
I (1042) system_api: read default base MAC address from EFUSE
I (1052) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
`

> 你好@maplerian
> 
> 感谢您的信息，第一个问题已修复，参考[9a6ccb7](https://github.com/espressif/esp-idf/commit/9a6ccb7f579a9d0162b32908cf9e57f89f771985)，请测试，谢谢。

已测试, 但是还是失败了

maplerian, 2022-08-09:
很抱歉, 刚才只是修改了蓝牙的文件, 现在将WIFI相关的文件也修改后,可以了, 第一个问题修复了

maplerian, 2022-08-09:
第二个问题,  蓝牙也可以在CPU1运行下正确保存配对码了

maplerian, 2022-08-09:
希望官方能尽快修复FLASH写和ESP_LOG组件相关的线程栈占用大问题

0xjakob, 2022-08-16:
@maplerian When you say "FLASH reading and writing", you mean NVS reading/writing, right?

Could you open an issue for each of the memory size problems, please? It's better for everyone involved if we have one issue per problem. 

In general, printf-based functions will use a considerable amount of stack (1 or 1.5K is normal for these). If a lot of data is already saved on the stack when you call `printf` or `ESP_LOG`, then this will add up.

maplerian, 2022-08-18:
> @maplerian当您说“FLASH读写”时，您的意思是NVS读写，对吗？
Yes, during development I often encountered that writes to NVS require a lot of thread stack space

0xjakob, 2023-03-15:
@maplerian May I ask, has this been resolved? Did you try to increase the stack size?

maplerian, 2023-03-16:
As for the part of printf, I directly replaced it by carbon copy, and the rest parts have been solved. Personally, I don't agree with the excessive stack size occupied by printf, because the memory usage is very sensitive for the projects I use. Personally, I don't want it to occupy too much memory, but I can save a little

closed 2023-03-16 by maplerian

0xjakob, 2023-03-16:
@maplerian Thanks for the feedback! In the future, we may be able to consider it. For now, we are using the newlib libc implementation of `printf`. newlib is an upstream library (like gcc, e.g.), making it not difficult to change.