espressif/esp-idf #9536: Android BLE Prov App leaves ESP device paired

bschick, 2022-08-09:
## Environment

- Module or chip used:  ESP32-C3-MINI-1-N4
- ESP-IDF v4.4.1
- Build System:        idf.py
- Operating System:    macOS
- Using an IDE?: No
- Power Supply:    USB
- ESP BLE Prove: Androidv2.0.13

## Problem Description

I am using the latest Android ESP BLE Prov mobile app with ESP-IDF code derived from https://github.com/espressif/esp-idf/tree/30e8f19f5a/examples/provisioning/wifi_prov_mgr.

The first BLE WIFI provisioning is successful, but after provisioning WIFI succeeds the ESP device is left paired with the Android device. If the ESP device is reset with wifi_prov_mgr_reset_provisioning(), and the BLE Prov mobile app is run again, the next attempt to provision the same ESP device will fail because there is an existing pairing. The failed provisioning seems to remove the paring and the following attempt will succeed, but that seems like a lucky coincidence rather than an explicit design.

## Expected Behavior

The Android ESP BLE Prov mobile app can be used to repeatedly provision WIFI without errors or manually removing already paired ESP device.

## Potential Solutions

1. I think the best solution would be to not cause permanent bonding/paring to occur. I believe the Android device itself may be initiating the bond (aka pairing) due to some characteristic returned by the ESP device. See "Initiating a bond" in here: https://punchthrough.com/android-ble-guide/#Bonding-with-a-BLE-device

The reason I believe it is Android OS initiated is that the ESP BLE mobile app does not call createBond() and WIFI_PROV_BLE_BONDING is not set in the ESP32 firmware code. Also, using ESP-IDF 3.x this problem did not exist and a bond/pairing was not established. That makes me think the ESP-IDF library has changed some characteristic or other property that now triggers bonding (should be tested on iOS also).

2. Next option would be to allow the bond to be created, but change the mobile app to check for an existing pairing and reuse it. I believe that would avoid the error and disconnect that currently results from calling "BluetoothDevice:connectGatt"  and *BluetoothGatt#discoverServices* on a device that is already paired. 

I have not tested this, but I assume that would be done by determining the device was already paired and then using *BluetoothGatt#connect* rather than *BluetoothDevice:connectGatt*.

3. Use java reflection to always remove the bond during connection shutdown. This uses a hidden Android SDK method, but it seems to be well used and it still works when I tested it. In BLETransport.java in the BluetoothGattCallback's STATE_DISCONNECTED handlerm, add the following:

```  
try {
  Method m = currentDevice.getClass().getMethod("removeBond", (Class[]) null);
  m.invoke(currentDevice, (Object[]) null);
} catch (Exception e) {
  Log.e(TAG, "removeBond failed: " + e.getMessage());
}
```

jacek12345, 2022-08-09:
I had the same problem on iOS app, but disabling WIFI_PROV_BLE_BONDING helps.
Duble check if have this disabled.

bschick, 2022-08-10:
@jacek12345 I mentioned that in the potential solutions #1... WIFI_PROV_BLE_BONDING is not set. I doubled checked sdkconfig:
# CONFIG_WIFI_PROV_BLE_BONDING is not set

jacek12345, 2022-08-10:
@bschick I suggest update ESP-IDF to 4.4.2 branch. 
I found: "Fixed bond not saved properly after re-pairing" in this branch.

bschick, 2022-08-10:
@jacek12345 Thanks for the pointer, while I don't believe the specific change you mentioned fixed he problem it did lead me to a different commit that did. My suspicion in the original report was correct... some ESP-IDF change in the past forced BT characteristics to be encrypted which triggers Android devices (and I'd assume iOS) to initiate bonding on their own. 

The change below merged into v4.4 added support for removing forced characteristic encryption 
https://github.com/espressif/esp-idf/commit/7a01fe7a713a33ba41d942fccb39c5fe30e3568e

**Disabling the new CONFIG_WIFI_PROV_BLE_FORCE_ENCRYPTION flag stopped the undesired bond.** 

I assume the actual provisioning data passed of BLE (wifi ssid and password) are encrypted given that WIFI_PROV_BLE_SEC_CONN is still set.

bschick, 2022-08-11:
Even though I found a solution, I think this Issue still needs to be addressed. Currently on Android, and I'd assume iOS, a pairing is created when using the default settings of the ESP prov manager example:  https://github.com/espressif/esp-idf/tree/30e8f19f5a/examples/provisioning/wifi_prov_mgr

Then when the Mobile App is used to provision the same device again, an error results since the app doesn't handle ian established pairing. A few adjustments could be made to make the mobile app and the ESP example work better together:

1. Mobile app should be changed to reuse an existing pairing. 
2. The ESP example should be configured with CONFIG_WIFI_PROV_BLE_FORCE_ENCRYPTION disable by default so no paring occurs.
3. ESP Prov manager configs should be changed to enable CONFIG_WIFI_PROV_BLE_BONDING when CONFIG_WIFI_PROV_BLE_FORCE_ENCRYPTION is enabled since the Mobile device will save the bond regardless

rahult-github, 2023-05-12:
Changes pertaining to disabling CONFIG_WIFI_PROV_BLE_FORCE_ENCRYPTION by default are done which i believe should suffice to fix this issue.

rahult-github, 2024-02-27:
> Even though I found a solution, I think this Issue still needs to be addressed. Currently on Android, and I'd assume iOS, a pairing is created when using default configs for the ESP prov manager example: https://github.com/espressif/esp-idf/tree/30e8f19f5a/examples/provisioning/wifi_prov_mgr
> 
> When the Mobile App is used to provision the same ESP device a 2nd time, an error results since the app doesn't handle an already paired ESP device. A few adjustments could be made to make the mobile app and the ESP example work better together:
> 
> 1. Mobile app should be changed to reuse an existing pairing.
> 2. The ESP example should be configured with CONFIG_WIFI_PROV_BLE_FORCE_ENCRYPTION disable by default so no paring occurs.
> 3. ESP Prov manager configs should be changed to enable CONFIG_WIFI_PROV_BLE_BONDING when CONFIG_WIFI_PROV_BLE_FORCE_ENCRYPTION is enabled since the Mobile device will save the bond regardless

For point 1,  since app side changes are not tracked on IDF side, i request you to please raise issue in   https://github.com/espressif/esp-idf-provisioning-android 

For point 2 , change has bene done. 

For point 3, this is configurable to cater to different use cases. So user can select / deselect the option as per their use.

closed 2024-02-27 by rahult-github