espressif/esp-idf #9537: Loop on audio connnect on the hfp_ag example with QC 35 earphones

luisesn, 2022-08-09:
- Development Kit:     TTGO
- Module or chip used:  ESP32
- IDF version: v4.4.2
- Build System:  vscode espidf plugin
- Compiler version: latest at this time
- Operating System:     Windows
- (Windows only) environment type: powershell
- Using an IDE?: Yes, vscode
- Power Supply:         USB

## Problem Description
The code builds perfectly and I can connect with the device, but when I try to connect the audio (`cona` command) it gets stuck in a loop, jumping between the audio codecs but never selecting one.

(I have selected HCI on both bluetooth and bluedroid menuconfig options)

### Expected Behavior
From the readme file:
```
W (1066280) BT_APPL: BTA_AG_SCO_CODEC_ST: Ignoring event 1
I (1067200) BT_APP_HF: APP HFP event: BCS_EVT
I (1067200) BT_APP_HF: --AG choose codec mode: CVSD Only
E (1067230) BT_BTM: btm_sco_connected, handle 180
I (1067240) BT_APP_HF: APP HFP event: AUDIO_STATE_EVT
I (1067240) BT_APP_HF: --Audio State connected
```

### Actual Behavior
```
Connect Audio
I (114394) BT_BTM: BTM_SetPowerMode: pm_id 0 BDA: a577ab80 mode:0x0
I (114394) BT_APP_HF: APP HFP event: AUDIO_STATE_EVT
I (114404) BT_APP_HF: --Audio State connecting
I (114414) BT_BTM: BTM_SetPowerMode: pm_id 0 BDA: a577ab80 mode:0x0
I (114414) BT_BTM: BTM_SetEScoMode -> mode 2
I (114414) BT_BTM: BTM_CreateSco -> (e)SCO Link for ACL handle 0x0080, Desired Type 2
I (114434) BT_BTM: txbw 0x1f40, rxbw 0x1f40, lat 0xd, voice 0x43, retrans 0x02, pkt 0x0388
E (114434) BT_BTM: btm_sco_connected, handle 80
I (114444) BT_BTM: BTM_SetPowerMode: pm_id 0 BDA: a577ab80 mode:0x0
I (114444) BT_APP_HF: APP HFP event: BCS_EVT
I (114454) BT_BTM: BTM_SetPowerMode: pm_id 0 BDA: a577ab80 mode:0x0
I (114464) BT_BTM: BTM_SetEScoMode -> mode 2
I (114464) BT_BTM: BTM_CreateSco -> (e)SCO Link for ACL handle 0x0080, Desired Type 2
I (114474) BT_BTM: txbw 0x1f40, rxbw 0x1f40, lat 0x8, voice 0x43, retrans 0x02, pkt 0x03c8
E (114484) BT_BTM: btm_sco_connected, handle 80
I (114484) BT_BTM: BTM_SetPowerMode: pm_id 0 BDA: a577ab80 mode:0x0
I (114464) BT_APP_HF: --Consequence of codec negotiation: Use MSBC
I (114504) BT_APP_HF: APP HFP event: BCS_EVT
I (114514) BT_BTM: BTM_SetPowerMode: pm_id 0 BDA: a577ab80 mode:0x0
I (114514) BT_BTM: BTM_SetEScoMode -> mode 2
I (114514) BT_BTM: BTM_CreateSco -> (e)SCO Link for ACL handle 0x0080, Desired Type 2
I (114534) BT_BTM: txbw 0x1f40, rxbw 0x1f40, lat 0xc, voice 0x60, retrans 0x02, pkt 0x033d
E (114534) BT_BTM: btm_sco_connected, handle 80
I (114544) BT_BTM: BTM_SetPowerMode: pm_id 0 BDA: a577ab80 mode:0x0
I (114504) BT_APP_HF: --Consequence of codec negotiation: Use MSBC
I (114554) BT_APP_HF: APP HFP event: BCS_EVT
I (114554) BT_APP_HF: --Consequence of codec negotiation: Use CVSD
I (114564) BT_BTM: BTM_SetPowerMode: pm_id 0 BDA: a577ab80 mode:0x0
I (114574) BT_BTM: BTM_SetEScoMode -> mode 2
I (114574) BT_BTM: BTM_CreateSco -> (e)SCO Link for ACL handle 0x0080, Desired Type 2
I (114594) BT_BTM: txbw 0x1f40, rxbw 0x1f40, lat 0x8, voice 0x43, retrans 0x02, pkt 0x03c8
E (114594) BT_BTM: btm_sco_connected, handle 80
I (114604) BT_BTM: BTM_SetPowerMode: pm_id 0 BDA: a577ab80 mode:0x0
I (114604) BT_APP_HF: APP HFP event: BCS_EVT
I (114614) BT_APP_HF: --Consequence of codec negotiation: Use MSBC
I (114624) BT_BTM: BTM_SetPowerMode: pm_id 0 BDA: a577ab80 mode:0x0
I (114634) BT_BTM: BTM_SetEScoMode -> mode 2
I (114634) BT_BTM: BTM_CreateSco -> (e)SCO Link for ACL handle 0x0080, Desired Type 2
I (114634) BT_BTM: txbw 0x1f40, rxbw 0x1f40, lat 0xc, voice 0x60, retrans 0x02, pkt 0x033d
E (114654) BT_BTM: btm_sco_connected, handle 80
I (114654) BT_BTM: BTM_SetPowerMode: pm_id 0 BDA: a577ab80 mode:0x0
I (114654) BT_APP_HF: APP HFP event: BCS_EVT
I (114674) BT_APP_HF: --Consequence of codec negotiation: Use CVSD
I (114684) BT_BTM: BTM_SetPowerMode: pm_id 0 BDA: a577ab80 mode:0x0
I (114684) BT_BTM: BTM_SetEScoMode -> mode 2
I (114694) BT_BTM: BTM_CreateSco -> (e)SCO Link for ACL handle 0x0080, Desired Type 2
I (114694) BT_BTM: txbw 0x1f40, rxbw 0x1f40, lat 0x8, voice 0x43, retrans 0x02, pkt 0x03c8
E (114704) BT_BTM: btm_sco_connected, handle 80
```

### Steps to reproduce

1. Build hfg_ac example
2. Connect with QC 35 II earphones
3. Try to connect audio

BetterJincheng, 2022-08-15:
@luisesn   Please attach your sdkconfig.

luisesn, 2022-08-16:
Here it goes:

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/9348441/sdkconfig.txt)

BetterJincheng, 2022-08-16:
@luisesn In menuconfig `Component config > Bluetooth > Controller Options > BR/EDR Sync(SCO/eSCO) Max Connections` ,set the sco connection to 1 (or >= 1) and try again.

luisesn, 2022-08-16:
And now it's working! :)

Thanks!

Maybe you should add that to the example README.md? Can I send an PR with it?

BetterJincheng, 2022-08-16:
@luisesn  We will fix it right now. Thanks for your reporting!

closed 2022-08-19 by espressif-bot