espressif/esp-idf #9538: Incorrect WiFi status when router password added extra letter

Rbhagwat62, 2022-08-09:
## Environment

- Development Kit:      custom hardware
- Kit version (for WroverKit/PicoKit/DevKitC): NA
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.3.2
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt.
- Using an IDE?: VS code
- Power Supply:  external

## Problem Description
We are using esp_wifi_sta_get_ap_info API get the current Wi-Fi status. 

Consider test case ESP is connected with Router whose ssid = NETGEAR and password = qwerty. We configured ESP with router ssid and Password, its get connected and gives ESP_OK status when we call esp_wifi_sta_get_ap_info.

Now if we change router password = qwertyu then we get Wi-Fi disconnect event and it shows below in the log continuously.
wifi: new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
wifi:state: init -> auth (b0)
wifi:state: auth -> assoc (0)
wifi:state: assoc -> run (10)
Wi-Fi disconnected, trying to reconnect...
Wi-Fi disconnected, trying to reconnect...

When we do the query by calling esp_wifi_sta_get_ap_info it some time gives ESP_OK and sometimes give ESP_ERR_WIFI_NOT_CONNECT

In our case ESP is connected with another system. When the second system sends query about the network status then we call the esp_wifi_sta_get_ap_info. We tried sending query at 5s and 10s interval but still it randomly gives ESP_OK though WiFi is not yet connected. Because of that another system keep showing connected for some time and then disconnect.

What is the correct way to use esp_wifi_sta_get_ap_info API for above test case?

zhangyanjiaoesp, 2022-08-11:
@Rbhagwat62 
When `wifi:state: init -> auth (b0)` is shown, it means the AP's info has been saved. when `wifi:state: run -> init (xxx)` is shown, it means the connection failed, and the AP's info has been cleared. 
Because you use a wrong password to connect to the AP, and run reconnect when connect fail, so the wifi's state will change all the time. when you call `esp_wifi_get_ap_info()` API every 5s or 10s, you can't determine the state of wifi, so the API some time gives ESP_OK and sometimes give ESP_ERR_WIFI_NOT_CONNECT.

Suggest you call `esp_wifi_get_ap_info()` after every connect attempt.

Rbhagwat62, 2022-08-22:
@zhangyanjiaoesp 
Thanks for the response.

calling esp_wifi_get_ap_info() after every connect attempt, seems working as per expectation.

closed 2022-08-22 by Rbhagwat62