espressif/esp-idf #9539: esp_event.h: No such file or directory

thxmxx, 2022-08-09:
I'm trying to link my project (idf 5) with `esp_modem` but I'm getting the following error:

`... managed_components/espressif__esp_modem/include/cxx_include/esp_modem_primitives.hpp:17:10: fatal error: esp_event.h: No such file or directory`

### idf_component.yml:
```
## IDF Component Manager Manifest File
dependencies:
  espressif/esp_modem: "==0.1.18"
  espressif/coap: "==4.3.1-rc.1"
  ## Required IDF version
  idf:
    version: ">=4.4.0"
  # # Put list of dependencies here
  # # For components maintained by Espressif:
  # component: "~1.0.0"
  # # For 3rd party components:
  # username/component: ">=1.0.0,<2.0.0"
  # username2/component2:
  #   version: "~1.0.0"
  #   # For transient dependencies `public` flag can be set.
  #   # `public` flag doesn't have an effect dependencies of the `main` component.
  #   # All dependencies of `main` are public by default.
  #   public: true
```

### Main CMakeLists.txt

```
idf_component_register(SRCS "bt.c" "main.c"
                    INCLUDE_DIRS "." REQUIRES esp_event driver nvs_flash bt esp_timer GSM)

target_compile_options(${COMPONENT_LIB} PRIVATE "-Wno-format" "-Wno-implicit-function-declaration" "-Wno-cpp" "-Wno-unused-variable")
```

Can someone help me?

igrr, 2022-08-09:
The fix was recently committed by @suren-gabrielyan-espressif in https://github.com/espressif/esp-protocols/commit/6e4e4fab1d2c6c63c796a81b5dce33d4d33892e4 but it seems the new release of esp-modem hasn't been made yet. We'll try to release 0.1.19 with the fix soon!

Alvin1Zhang, 2022-09-09:
Thanks for reporting, the issue seems to have been fixed when the https://github.com/espressif/esp-protocols/pull/112 gets merged, feel free to reopen.

closed 2022-09-09 by Alvin1Zhang

LHYYSY, 2023-04-07:
[image]

LHYYSY, 2023-04-07:
The IDF version is 5.0.1