espressif/esp-idf #9541: Add support for UDP-Lite (RFC 3828)

fredlarochelle, 2022-08-10:
With UDP, if the payload is corrupted by even a single bit, the packet is rejected by the checksum. UDP-Lite (link to the [wikipedia article](https://en.wikipedia.org/wiki/UDP-Lite) for more info) enables to delivery of a partially corrupted packet to the application layer where they can be further processed. 

For the ESP32, that functionality would be useful. Take a hypothetical example where ten consecutive sensor readings are sent as a payload. If only the bits from one reading are corrupted, with some programming logic (or even advanced forward error correction), the nine other values could be used. No need to reject the whole packet.

Adding that functionality would be useful for VoIP applications and data streaming, ...

Currently, that functionality could probably be implemented using Wifi Sniffer Mode by getting the 802.11 CRC error frame. That is not an ideal solution since promiscuous mode reduce the throughput greatly. It would be better for the wifi driver to support UDP-Lite to simply "let through" the corrupted packet.

In brief, I propose that the Esp-IDF should have a way to activate support for UDP-Lite (RFC 3828), adding that support shouldn't be too complicated I would imagine, the wifi driver simply needs to "not check the checksum", maybe a function could return a boolean flag that tells if the checksum has passed or failed with the payload.

fredlarochelle, 2022-08-10:
If it is of any help, LwIP seems to be already supporting UDD-Lite -> https://github.com/espressif/esp-lwip/blob/2.1.2-esp/src/core/udp.c

xueyunfei998, 2022-08-18:
hi @fredlarochelle 

Thank you for offering such good advice.

However, it may not be appropriate to implement this function in wifi lib.

Because of the packets received by wifi, there are management packets, control packets and data packets.
If crc is turned off, in some complex environments, some erroneous packets will be received. However, wifi lib does not have UDP-like error correction function, which will make wifi unable to handle these erroneous packets.

For tcp packets, if the crc detection function is turned off, in some responsible environments, the wrong packets will be received, which will cause the TCP window to be updated incorrectly, and a large number of retransmissions may occur.

fredlarochelle, 2022-08-19:
UDP-Lite is a connectionless protocol like UDP. There wouldn't be any retransmission.

A " packet " in UDP comprises the datagram header and the data payload. For IPv4, you have two possible configurations; 1) with the checksum turned on for both the datagram header or 2) the data payload or turned off for both (I can't find information if the ESP-IDF supports the second configuration, but it is in the standard). For IPv6, the checksum is mandatory.

UDP-Lite is sort of a middle ground between those 2 configurations (and supports IPv4 and IPv6). Different configurations are available, but the most interesting is where the checksum is enabled only for the datagram header and disabled for the data payload. In brief, you have a "partial checksum" with UDP-Lite. 

With a checksum on the datagram header, the wifi would be able to handle different errors and it would then be up to the user to decide what to do with the data in the payload in the application layer.

I think this precision might appease your concerns!

xueyunfei998, 2022-08-19:
[image]

You can enable udp check in menuconfig.
Turn off ipv4 udp check because wifi has already been checked, and turn off ipv4 check in order to improve performance.

closed 2024-01-12 by Alvin1Zhang