espressif/esp-idf #9543: simple_ota_example  esp-tls-mbedtls: Failed to set client configurations, returned [0x8017] (ESP_ERR_MBEDTLS_SSL_SETUP_FAILED)

DreamPWJ, 2022-08-11:
[image]

When I do the ota upgrade, there is an error as shown in the figure, I want to know how this pem is generated? What is the role of existence code?

mahavirj, 2022-08-11:
@DreamPWJ 

Please try `simple_ota` example with certificate bundle feature from:

https://github.com/espressif/esp-idf/blob/afc6fba7f5619af82623f22052685bbcc6fb5665/examples/system/ota/simple_ota_example/main/simple_ota_example.c#L95

This way you need not generate and attach any other server certificate. Please let us know if you still run into any issues.

DreamPWJ, 2022-08-12:
[image]
doesn't work, same error  @mahavirj

mahavirj, 2022-08-12:
@DreamPWJ Please share details as per GH issue template, along with your `sdkconfig` and sample app code.

DreamPWJ, 2022-08-12:
https://github.com/DreamPWJ/embedded-monorepo/blob/develop/projects/esp32-demo/lib/ota/ota.cpp   @mahavirj