espressif/esp-idf #9546: Zigbee Examples: esp_zigbee_rcp -> "zb_esp_platform_config_t" not available - can not be found

ESP32DE, 2022-08-11:
Zigbee Example : esp_zigbee_rcp

[zb_esp_plattform_config_t](https://github.com/espressif/esp-idf/blob/master/examples/zigbee/esp_zigbee_rcp/main/esp_zigbee_rcp.c#L84-L88) not available - can not be found

used HW: ESP32-H2-DevKitC-1 V2.1
[image]

used ESP-IDF: Master (last)

used [esp_zboss_lib](https://github.com/espressif/esp-zboss-lib)

can you please provide me the struct /  point me to the typdef  "zb_esp_platform_config_t" ?
is it in a Blob Lib ? 

thank you.

ESP32DE, 2022-08-11:
**zb_esp**_platform_config_t  

[**esp_zb**_plattform_config_t](https://github.com/espressif/esp-zboss-lib/blob/801c5eb2c1db7338e09e7ae5581909989902e053/include/zb_config_platform.h#L118-L121)

[all types](https://github.com/espressif/esp-zboss-lib/blob/801c5eb2c1db7338e09e7ae5581909989902e053/include/zb_config_platform.h#L74-L116) are started with **zb_esp** only the plattform_config_t is swapped in **esp_zb** :)

typo? internal use? can you please check?
thank you.

btw: 
what does the [external version](https://github.com/espressif/esp-zboss-lib/blob/801c5eb2c1db7338e09e7ae5581909989902e053/include/zb_config_platform.h#L123) actually do / usecase?

likunqiao097304, 2022-08-12:
@ESP32DE Hi, I am assuming you are using esp-zboss-lib 0.1.0 version from the link you provided. The latest version is 0.1.0 which we change from zb_esp_platform_config_t to esp_zb_platform_config_t for naming alignment.

However, if you use the esp-idf zigbee example, the idf_component.yml will automatically help you to download correct on, which is 0.0.6 version. 

So in a word, either you change the name as you did, or delete your esp-zboss-lib and let example itself to download 0.0.6 version. Hope I answer your question.

ESP32DE, 2022-08-12:
@likunqiao097304 
Thank you very much for the detailed help and explanation. I had already thought of something like that, but I wanted to make sure with the question. I saw [the change in 0.1.0](https://github.com/espressif/esp-zboss-lib/commit/ac6348b5c65b887ab0c814573029b40a83d9da0c#diff-8d68bb029f186ad30be2095a623754425d9394e393e4665813b5555a9d8f1cbf) also zb_esp_platform_serial_config_set
zb_esp_macsplit_get_version() and other.

Can be that up to 0.0.6 was esp internal / skipped from 0.0.5 to 0.1.0 and [not online](https://github.com/espressif/esp-zboss-lib/commits/master/include/zb_config_platform.h).
[image]

However - Request can be closed - if you want - can also remain open as information for others, depending on the situation. I can close too - let me know.

Again, thank you very much, solved for me.

chshu, 2022-08-12:
It was updated to 0.0.6 within this commit: https://github.com/espressif/esp-zboss-lib/commit/c741c1b45c673229a1acddfe66c275b2c48457d1

The current Zigbee examples in IDF should download 0.0.6 by default, and will be updated to use 0.1.0 soon.
Since it's not an issue, please help to close if no other questions, thanks.

closed 2022-08-12 by ESP32DE