espressif/esp-idf #9547: esp_image: Checksum failed.

hipal31, 2022-08-12:
## Environment

- Module or chip used:  ESP32C3
- IDF version : v4.3.2
- Build System:         idf.py
- Compiler version :8.4.0
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply:  external 3.3V

## Problem Description

I'm getting  `esp_image: Checksum failed.` error during boot of controller. I did OTA using esp_https_ota and after successful upgrade device was working properly. But after 2 months I got this error that controller can't boot due to checksum failed.  And out of 100 devices 4 devices are getting this kind of error. Which means it's hard to reproduce this.

### Expected Behavior

It should work properly even after 2-3 months or a year.

### Actual Behavior

After 2-3 months it's getting checksum failed error during boot.

## Debug Logs

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d153a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x1944
load:0x403ce000,len:0x8d4
load:0x403d0000,len:0x2dc0
entry 0x403ce000
[0;32mI (35) boot: ESP-IDF v4.3.2-dirty 2nd stage bootloader[0m
[0;32mI (35) boot: compile time 16:52:33[0m
[0;32mI (35) boot: chip revision: 3[0m
[0;32mI (37) boot_comm: chip revision: 3, min. bootloader chip revision: 2[0m
[0;32mI (44) boot.esp32c3: SPI Speed      : 80MHz[0m
[0;32mI (49) boot.esp32c3: SPI Mode       : DIO[0m
[0;32mI (54) boot.esp32c3: SPI Flash Size : 4MB[0m
[0;32mI (58) boot: Enabling RNG early entropy source...[0m
[0;32mI (64) boot: Partition Table:[0m
[0;32mI (67) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (75) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000[0m
[0;32mI (82) boot:  1 nvs              WiFi data        01 02 00010000 00006000[0m
[0;32mI (90) boot:  2 otadata          OTA data         01 00 00016000 00002000[0m
[0;32mI (97) boot:  3 phy_init         RF data          01 01 00018000 00001000[0m
[0;32mI (105) boot:  4 ota_0            OTA app          00 10 00020000 001c2000[0m
[0;32mI (112) boot:  5 ota_1            OTA app          00 11 001f0000 001c2000[0m
[0;32mI (120) boot:  6 fctry            WiFi data        01 02 003b2000 00006000[0m
[0;32mI (127) boot: End of partition table[0m
[0;32mI (132) boot_comm: chip revision: 3, min. application chip revision: 2[0m
[0;32mI (139) esp_image: segment 0: paddr=00020020 vaddr=3c130020 size=43074h (274548) map[0m
[0;32mI (188) esp_image: segment 1: paddr=0006309c vaddr=3fc90000 size=0580ch ( 22540) load[0m
[0;32mI (193) esp_image: segment 2: paddr=000688b0 vaddr=40380000 size=07768h ( 30568) load[0m
[0;32mI (200) esp_image: segment 3: paddr=00070020 vaddr=42000020 size=12ec08h (1240072) map[0m
[0;32mI (389) esp_image: segment 4: paddr=0019ec30 vaddr=40387768 size=086d8h ( 34520) load[0m
[0;32mI (395) esp_image: segment 5: paddr=001a7310 vaddr=50000000 size=00010h (    16) load[0m
[0;31mE (396) esp_image: Checksum failed. Calculated 0x12 read 0x32[0m
[0;31mE (401) boot: OTA app partition slot 0 is not bootable[0m
[0;31mE (407) esp_image: image at 0x1f0000 has invalid magic byte (nothing flashed here?)[0m
[0;31mE (415) boot: OTA app partition slot 1 is not bootable[0m
[0;31mE (421) boot: No bootable app partitions in the partition table[0m
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d153a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x1944
load:0x403ce000,len:0x8d4
load:0x403d0000,len:0x2dc0
entry 0x403ce000
[0;32mI (35) boot: ESP-IDF v4.3.2-dirty 2nd stage bootloader[0m
[0;32mI (35) boot: compile time 16:52:33[0m
[0;32mI (35) boot: chip revision: 3[0m
[0;32mI (37) boot_comm: chip revision: 3, min. bootloader chip revision: 2[0m
[0;32mI (44) boot.esp32c3: SPI Speed      : 80MHz[0m
[0;32mI (49) boot.esp32c3: SPI Mode       : DIO[0m
[0;32mI (54) boot.esp32c3: SPI Flash Size : 4MB[0m
[0;32mI (58) boot: Enabling RNG early entropy source...[0m
[0;32mI (64) boot: Partition Table:[0m
[0;32mI (67) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (75) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000[0m
[0;32mI (82) boot:  1 nvs              WiFi data        01 02 00010000 00006000[0m
[0;32mI (90) boot:  2 otadata          OTA data         01 00 00016000 00002000[0m
[0;32mI (97) boot:  3 phy_init         RF data          01 01 00018000 00001000[0m
[0;32mI (105) boot:  4 ota_0            OTA app          00 10 00020000 001c2000[0m
[0;32mI (112) boot:  5 ota_1            OTA app          00 11 001f0000 001c2000[0m
[0;32mI (120) boot:  6 fctry            WiFi data        01 02 003b2000 00006000[0m
[0;32mI (127) boot: End of partition table[0m
[0;32mI (132) boot_comm: chip revision: 3, min. application chip revision: 2[0m
[0;32mI (139) esp_image: segment 0: paddr=00020020 vaddr=3c130020 size=43074h (274548) map[0m
[0;32mI (188) esp_image: segment 1: paddr=0006309c vaddr=3fc90000 size=0580ch ( 22540) load[0m
[0;32mI (193) esp_image: segment 2: paddr=000688b0 vaddr=40380000 size=07768h ( 30568) load[0m
[0;32mI (200) esp_image: segment 3: paddr=00070020 vaddr=42000020 size=12ec08h (1240072) map[0m
[0;32mI (389) esp_image: segment 4: paddr=0019ec30 vaddr=40387768 size=086d8h ( 34520) load[0m
[0;32mI (395) esp_image: segment 5: paddr=001a7310 vaddr=50000000 size=00010h (    16) load[0m
[0;31mE (396) esp_image: Checksum failed. Calculated 0x12 read 0x32[0m
[0;31mE (401) boot: OTA app partition slot 0 is not bootable[0m
[0;31mE (407) esp_image: image at 0x1f0000 has invalid magic byte (nothing flashed here?)[0m
[0;31mE (415) boot: OTA app partition slot 1 is not bootable[0m
[0;31mE (421) boot: No bootable app partitions in the partition table[0m
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d153a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x1944
load:0x403ce000,len:0x8d4
load:0x403d0000,len:0x2dc0
entry 0x403ce000
[0;32mI (35) boot: ESP-IDF v4.3.2-dirty 2nd stage bootloader[0m
[0;32mI (35) boot: compile time 16:52:33[0m
[0;32mI (35) boot: chip revision: 3[0m
[0;32mI (37) boot_comm: chip revision: 3, min. bootloader chip revision: 2[0m
[0;32mI (44) boot.esp32c3: SPI Speed      : 80MHz[0m
[0;32mI (49) boot.esp32c3: SPI Mode       : DIO[0m
[0;32mI (54) boot.esp32c3: SPI Flash Size : 4MB[0m
[0;32mI (58) boot: Enabling RNG early entropy source...[0m
[0;32mI (64) boot: Partition Table:[0m
[0;32mI (67) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (75) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000[0m
[0;32mI (82) boot:  1 nvs              WiFi data        01 02 00010000 00006000[0m
[0;32mI (90) boot:  2 otadata          OTA data         01 00 00016000 00002000[0m
[0;32mI (97) boot:  3 phy_init         RF data          01 01 00018000 00001000[0m
[0;32mI (105) boot:  4 ota_0            OTA app          00 10 00020000 001c2000[0m
[0;32mI (112) boot:  5 ota_1            OTA app          00 11 001f0000 001c2000[0m
[0;32mI (120) boot:  6 fctry            WiFi data        01 02 003b2000 00006000[0m
[0;32mI (127) boot: End of partition table[0m
[0;32mI (132) boot_comm: chip revision: 3, min. application chip revision: 2[0m
[0;32mI (139) esp_image: segment 0: paddr=00020020 vaddr=3c130020 size=43074h (274548) map[0m
[0;32mI (188) esp_image: segment 1: paddr=0006309c vaddr=3fc90000 size=0580ch ( 22540) load[0m
[0;32mI (193) esp_image: segment 2: paddr=000688b0 vaddr=40380000 size=07768h ( 30568) load[0m
[0;32mI (200) esp_image: segment 3: paddr=00070020 vaddr=42000020 size=12ec08h (1240072) map[0m
[0;32mI (389) esp_image: segment 4: paddr=0019ec30 vaddr=40387768 size=086d8h ( 34520) load[0m
[0;32mI (395) esp_image: segment 5: paddr=001a7310 vaddr=50000000 size=00010h (    16) load[0m
[0;31mE (396) esp_image: Checksum failed. Calculated 0x12 read 0x32[0m
[0;31mE (401) boot: OTA app partition slot 0 is not bootable[0m
[0;31mE (407) esp_image: image at 0x1f0000 has invalid magic byte (nothing flashed here?)[0m
[0;31mE (415) boot: OTA app partition slot 1 is not bootable[0m
[0;31mE (421) boot: No bootable app partitions in the partition table[0m
```

## Other items if possible

- sdkconfig file : [sdkconfig](https://github.com/espressif/esp-idf/files/9313796/sdkconfig.txt)

hipal31, 2022-08-17:
Can anyone help?

hipal31, 2022-11-09:
@JiaLWang Can you please help me or connect with the right person for this issue?

JiaLWang, 2022-11-09:
> [External: This email originated outside Espressif]

Hi Hipal,

Is it using wifi OTA?

Best regards,
Jialiang Wang 王家亮
Software Engineer｜Espressif Systems

www.espressif.com<https://www.espressif.com/>
Wi-Fi & Bluetooth MCUs and AIoT Solutions I Espressif Systems<https://www.espressif.com/>
Espressif Systems (688018.SH) is a public multinational, fabless semiconductor company established in 2008, with offices in China, the Czech Republic, India, Singapore and Brazil. We have a passionate team of engineers and scientists from all over the world, focused on developing cutting-edge Wi-Fi-and-Bluetooth, low-power, AIoT solutions.
www.espressif.com
#303A, Block 2, 690 Bibo Road,
Zhang Jiang High-Tech Park, Shanghai, China

hipal31, 2022-11-09:
@JiaLWang yes it is using wifi ota. But problem occurs without ota as well with  new code flashed.

JiaLWang, 2022-11-09:
@Alvin1Zhang  Would you please help with it? Thanks.

hipal31, 2022-11-15:
@JiaLWang @Alvin1Zhang Please look into the problem

hipal31, 2022-11-26:
@Alvin1Zhang Can you please help me with this?

ginkgm, 2022-11-28:
@hipal31 

firstly please help read out the flash ID and the status register when the device is broken, and post it here:

1. install & export in IDF 
2. run `esptool.py flash_id`
3. run `esptool.py read_flash_status`

and then do you have enough broken devices? if so, please try if re-flash the device can make it work again. (to see if this is a irreversible issue)

Michael

hipal31, 2022-11-28:
@ginkgm Thanks for the reply.

Until now I received 15 devices with same issue. I did re-flash the device and kept it for testing. After re-flashing It's working.

Right now I don't have broken device at present with the issue to read flash id. I'll send once I receive a new broken device.

Regards,
Hipal

hipal31, 2023-01-04:
> @hipal31
>
> firstly please help read out the flash ID and the status register when the device is broken, and post it here:
>
> 1. install & export in IDF
> 2. run `esptool.py flash_id`
> 3. run `esptool.py [...]

Hi @ginkgm ,
Recently I received a broken device, in which 
1. After running `esptool.py flash_id` command, result is : 
```
Connecting....
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 34:b4:72:4f:25:3c
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```
2.  After running `esptool.py read_flash_status` command, result is : 
```
Connecting...
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 34:b4:72:4f:25:3c
Stub is already running. No upload is necessary.
Status value: 0x0200
Hard resetting via RTS pin...
```

3. After that I run `idf.py erase-flash` and `idf.py flash` commands to re-flash module and it's working fine after re-flashing.

Please let me know if anything needs to be performed at my side and help me find out solution.

Regards,
Hipal

hipal31, 2023-01-09:
@ginkgm please look into the issue.

ginkgm, 2023-01-09:
@hipal31 ,

This looks like a random problem, which may be related to many factors (power supply, PCB, SW, or the flash itself). We have a team dealing with this kind of issues. Please help raise a ticket here:

https://cqms.espressif.com/ticker/add

Please also include this Github issue in the ticket, and our support team will contact with you.

hipal31, 2023-01-09:
@ginkgm can I have your email for it?

ginkgm, 2025-03-12:
Thank you. I think we have nothing else can do online. Issue closed.

closed 2025-03-12 by ginkgm