espressif/esp-idf #9548: BLE LPN does not receive message from friend if address of message is 0xffff(to all)

Shane555, 2022-08-12:
## Environment
- Development Kit:      [ESP32-C3-DevKitM-1]
- Kit version (for DevKitC): |v3|
- IDF version: v5.1.0
- Build System:         [idf.py]
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System:     [Ubuntu 20.04 LTS]
- Using an IDE: [VSCode]
- Power Supply:         [USB]

## Problem Description

I am running a 3 node network to test the functionality of LPN in ble mesh. The first node is a on/off client that periodically sends a GET message to all of the other nodes in the network. The second node is a generic server friend node. The last node is a generic server low power node. The message never reaches the LPN in any case. The LPN is always either in `BLE_MESH_LPN_RECV_DELAY` or `BLE_MESH_LPN_WAIT_UPDATE` which i believe is always so because it keeps polling the friend node but the friend node does not think there is a matching address for 0xFFFF to the LPN. 
The nodes are all provisioned using nrF mesh app. The code are all extracted and modified based off the ble_mesh_node example
 

### Code to reproduce this issue
in onoffclient's example
```cpp
  common.opcode = ESP_BLE_MESH_MODEL_OP_GEN_ONOFF_GET;
  common.model = onoff_client.model;
  common.ctx.net_idx = store.net_idx;
  common.ctx.app_idx = store.app_idx;
  common.ctx.addr = 0xFFFF; /* to all nodes */ //or 0xC000 if testing group addressing
  common.ctx.send_ttl = 3;
  common.ctx.send_rel = false;
  common.msg_timeout = 0; /* 0 indicates that timeout value from menuconfig will be used */
  common.msg_role = ROLE_NODE;

```
run a loop in client example
```cpp
  while (1)
  {
    example_ble_mesh_send_gen_onoff_get();
    vTaskDelay(20000 / portTICK_PERIOD_MS);
  }
```

What you will notice is the friend node side keeps saying it cannot find matching LPN for address 0xffff which should be sent to all LPN connected by default. Once i modified the code to send to a group address instead and have the LPN subscribe to the group address, the LPN can receive messages meant for it again.

JiaLWang, 2022-08-23:
Hi @Shane555 

We will add an option to LPN to implement the above functionality soon. The current LPN needs to send a Friend Subscription List Add message to subscribe to 0xFFFF.

Thanks.

Shane555, 2022-08-24:
Hi, thanks for the clarification. I understand that there are sleep modes such as light sleep and deep sleep. I understand they are compatible with BLE in esp but i am not sure if they can be implemented in BLE mesh. In particular, the low power nodes. Is this implemented in the mesh support for esp?

JiaLWang, 2022-08-24:
Hi @Shane555 

Here is the the bugfix patch, you can test with it. Please let me know if there is any problem with the patch.
[patch.zip](https://github.com/espressif/esp-idf/files/9412111/patch.zip)
You need to enable the option in the menuconfig. 
 (`Component config->ESP BLE Mesh Support->Support for Friend feature->Automatically subscribe all nodes address`)

Thanks.

Shane555, 2022-08-24:
Hi @JiaLWang  i have added the patch and tested it. It seems to work as expected now with the friend node handling the message correctly for the LPN. By the way, do you know if the LPN is compatible with light sleep/deep sleep? I am not sure if the automatic light sleep functionality in https://github.com/espressif/esp-idf/issues/947#issuecomment-500312453 works for BLE mesh in particular. Thanks!

JiaLWang, 2022-08-24:
Hi @Shane555 
Thanks for your contribution. We will merge it ASAP.
For ESP32, the LPN compatible light sleep/deep sleep and we tested it. But for esp32c3, we have not done relevant tests yet. Theoretically it can also work.
Thanks.

Alvin1Zhang, 2022-11-21:
Thanks for reporting, fix is available at https://github.com/espressif/esp-idf/commit/cfbdc55717282fcb4cfd1e0fb2ca5a2e27378fff and also back ported to other release branches, feel free to reopen.

closed 2022-11-21 by Alvin1Zhang