espressif/esp-idf #9549: nvs partition is not encrypted after enable CONFIG_NVS_ENCRYPTION

hyundolee27, 2022-08-12:
Before submitting a new issue, please follow the checklist and try to find the answer.

## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  9ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):
   v4.3.1-3-g6fc07a1a68
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     Linux
- (Windows only) environment type:
- Using an IDE?:  YES
- Power Supply:         USB

## Problem Description

1. write NVS partition (not encrypted)
2. enable CONFIG_FLASH_ENCRYPTION_ENABLED and CONFIG_NVS_ENCRYPTION
3. app bulid - flash and monitor by idf.py

In this condtition, I saw the bootloader stop for a long time and then restart.
flash encryption was successfully done.
nvs_flash_generate_keys, nvs_flash_secure_init, nvs_flash_secure_init_partition is successfully called.

But NVS partition is not encrypted after first booting(other flash data are encrypted after first booting)
Failed to read NVS Data (becaust NVS data was not encrypted, the attempt to decrypt it failed)

### Expected Behavior
nvs is succesfully encrypted, and success to read nvs data

### Actual Behavior
nvs is not encrypted, and failed to read nvs data

### Steps to reproduce

1. write NVS partition (not encrypted)
2. enable CONFIG_FLASH_ENCRYPTION_ENABLED and CONFIG_NVS_ENCRYPTION
3. app bulid - flash and monitor by idf.py

negativekelvin, 2022-08-13:
You have to generate the encrypted nvs and keys off device.  Encryption in place is not supported for nvs.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_partition_gen.html#to-generate-encrypted-nvs-partition

mahavirj, 2022-08-18:
@hyundolee27 As mentioned by @negativekelvin, NVS partition must be pre-encrypted before flashing for your case. 

For more details please refer to:
- [Design documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html#nvs-encryption)
- [Example](https://github.com/espressif/esp-idf/tree/master/examples/security/flash_encryption)

Closing this, please feel free to re-open if you still run into any issues.

closed 2022-08-18 by mahavirj