espressif/esp-idf #9550: Errors not showing in v4.4.2 but do in v4.4.1

versamodule, 2022-08-12:
Moved to a new PC and installed VSC and added ESP-IDF VS Code Extension.
Everything works fine but it no longer shows errors.  No squiggles, and on the file/folder tree the file with the issue does not turn red. 
Uninstalling v4.4.2 and installing v4.4.1 everything works fine.  Going back to 4.4.2 again, same problem comes back. 
Using blink example.

versamodule, 2022-08-15:
No change, same problem.

brianignacio5, 2022-08-16:
We've made some fixes in [this PR](https://github.com/espressif/vscode-esp-idf-extension/pull/779) ! Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/331310595.zip)

Related link: [how to Install extension from vsix](https://code.visualstudio.com/docs/editor/extension-marketplace#_install-from-a-vsix)

versamodule, 2022-08-18:
ESP-IDF VS Code Extension was updated to version 1.5.0. 
Now its showing the errors again.

closed 2022-08-18 by versamodule