espressif/esp-idf #9551: Used Flash size for Visual Studio Code

versamodule, 2022-08-12:
When you do a build in VSC, you get this nicely formatted output. 

```
Total sizes:
Used static IRAM:   67838 bytes ( 294402 remain, 18.7% used)
      .text size:   66811 bytes
   .vectors size:    1027 bytes
Used stat D/IRAM:   87865 bytes (  85063 remain, 50.8% used)
      .data size:   10241 bytes
      .bss  size:   77624 bytes
Used Flash size :  219411 bytes
      .text     :  159343 bytes
      .rodata   :   59812 bytes
Total image size:  297490 bytes (.bin may be padded larger)
```

In sdkconfig we set the flash size. It would be nice if "Used Flash size" also showed % used like IRAM & D/IRAM.

dobairoland, 2022-08-15:
Thanks for the suggestion.

fhrbata, 2024-06-27:
Hello @versamodule, the `esp-idf-size` tool has been revamped. The updated version, which is the default from ESP-IDF v5.3 onwards, displays flash usage information similarly to other memory types by mapping memory ranges from the link map file into the flash memory. Below is an example output for `esp32`.
```
                             Memory Type Usage Summary                              
┏━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━┳━━━━━━━━━━┳━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━┓
┃ Memory Type/Section   ┃ Used [bytes] ┃ Used [%] ┃ Remain [bytes] ┃ Total [bytes] ┃
┡━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━╇━━━━━━━━━━╇━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━┩
│ Flash Code            │        80686 │     2.41 │        3261618 │       3342304 │
│    .text              │        80686 │     2.41 │                │               │
│ IRAM                  │        51843 │    39.55 │          79229 │        131072 │
│    .text              │        50815 │    38.77 │                │               │
│    .vectors           │         1027 │     0.78 │                │               │
│ Flash Data            │        38248 │     0.91 │        4156024 │       4194272 │
│    .rodata            │        37992 │     0.91 │                │               │
│    .appdesc           │          256 │     0.01 │                │               │
│ DRAM                  │        11236 │     6.22 │         169500 │        180736 │
│    .data              │         8988 │     4.97 │                │               │
│    .bss               │         2248 │     1.24 │                │               │
│ RTC SLOW              │           24 │     0.29 │           8168 │          8192 │
│    .rtc_slow_reserved │           24 │     0.29 │                │               │
└───────────────────────┴──────────────┴──────────┴────────────────┴───────────────┘
Total image size: 179764 bytes (.bin may be padded larger)
```

In earlier versions of ESP-IDF, you can switch to the refactored version by setting or exporting the `ESP_IDF_SIZE_NG` environmental variable e.g. with `export ESP_IDF_SIZE_NG=1`.

Thank you!

closed 2024-06-27 by dobairoland