espressif/esp-idf #9552: esp_timer parameter can not set to one hour,It will cause watchdog timeout.

aboyuanna, 2022-08-13:
## Environment

- IDF version (run ``git describe --tags`` to find it):v4.4.2
- Build System: idf.py
- Operating System:Windows

## Problem Description

### Code to reproduce this issue

#include <stdio.h>
#include <string.h>
#include <unistd.h>
#include "esp_timer.h"
#include "esp_log.h"
#include "esp_sleep.h"
#include "sdkconfig.h"

static void periodic_timer_callback(void *arg);
static void oneshot_timer_callback(void *arg);

static const char *TAG = "example";

void app_main(void)
{
    const esp_timer_create_args_t periodic_timer_args = {
        .callback = &periodic_timer_callback,
        .name = "periodic"};

    esp_timer_handle_t periodic_timer;
    ESP_ERROR_CHECK(esp_timer_create(&periodic_timer_args, &periodic_timer));

    const esp_timer_create_args_t oneshot_timer_args = {
        .callback = &oneshot_timer_callback,
        .arg = (void *)periodic_timer,
        .name = "one-shot"};
    esp_timer_handle_t oneshot_timer;
    ESP_ERROR_CHECK(esp_timer_create(&oneshot_timer_args, &oneshot_timer));

    ESP_ERROR_CHECK(esp_timer_start_periodic(periodic_timer, 3600 * 1000 * 1000));
    ESP_ERROR_CHECK(esp_timer_start_once(oneshot_timer, 3600 * 1000 * 1000));
    ESP_LOGI(TAG, "Started timers, time since boot: %lld us", esp_timer_get_time());
}

static void periodic_timer_callback(void *arg)
{
    int64_t time_since_boot = esp_timer_get_time();
    ESP_LOGI(TAG, "Periodic timer called, time since boot: %lld us", time_since_boot);
}

static void oneshot_timer_callback(void *arg)
{
    int64_t time_since_boot = esp_timer_get_time();
    ESP_LOGI(TAG, "One-shot timer called, time since boot: %lld us", time_since_boot);
}

## Debug Logs

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6660
load:0x40078000,len:14848
ho 0 tail 12 room 4
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (29) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (29) boot: compile time 15:05:19
I (29) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (96) boot_comm: chip revision: 3, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=07a3ch ( 31292) map
I (123) esp_image: segment 1: paddr=00017a64 vaddr=3ffb0000 size=02340h (  9024) load
I (126) esp_image: segment 2: paddr=00019dac vaddr=40080000 size=0626ch ( 25196) load
I (141) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=14a24h ( 84516) map
I (172) esp_image: segment 4: paddr=00034a4c vaddr=4008626c size=051d0h ( 20944) load
I (181) esp_image: segment 5: paddr=00039c24 vaddr=50000000 size=00010h (    16) load
I (186) boot: Loaded app from partition at offset 0x10000
I (187) boot: Disabling RNG early entropy source...
I (202) cpu_start: Pro cpu up.
I (203) cpu_start: Starting app cpu, entry point is 0x4008100c
0x4008100c: call_start_cpu1 at C:/Users/HJYVM10/esp/esp-idf/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (217) cpu_start: Pro cpu start user code
I (217) cpu_start: cpu freq: 160000000
I (217) cpu_start: Application information:
I (221) cpu_start: Project name:     esp_timer_example
I (227) cpu_start: App version:      1
I (232) cpu_start: Compile time:     Aug 13 2022 15:19:50
I (238) cpu_start: ELF file SHA256:  d65a8710e015c443...
I (244) cpu_start: ESP-IDF:          v4.4.2
I (249) heap_init: Initializing. RAM available for dynamic allocation:
I (256) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (262) heap_init: At 3FFB2C30 len 0002D3D0 (180 KiB): DRAM
I (268) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (274) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (281) heap_init: At 4008B43C len 00014BC4 (82 KiB): IRAM
I (288) spi_flash: detected chip: generic
I (292) spi_flash: flash io: dio
I (297) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (10306) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10306) task_wdt:  - IDLE (CPU 0)
E (10306) task_wdt: Tasks currently running:
E (10306) task_wdt: CPU 0: esp_timer
E (10306) task_wdt: CPU 1: IDLE
E (10306) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x400D6647:0x3FFB0B800x400828A5:0x3FFB0BA0 0x4000BFED:0x3FFB4760 0x40087EA6:0x3FFB4770 0x40082112:0x3FFB4790 0x400D4573:0x3FFB47B0 0x400D458D:0x3FFB47E0 0x40087BF5:0x3FFB4800
0x400d6647: task_wdt_isr at C:/Users/HJYVM10/esp/esp-idf/components/esp_system/task_wdt.c:183 (discriminator 3)

0x400828a5: _xt_lowint1 at C:/Users/HJYVM10/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111

0x40087ea6: vPortClearInterruptMaskFromISR at C:/Users/HJYVM10/esp/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:571
 (inlined by) vPortExitCritical at C:/Users/HJYVM10/esp/esp-idf/components/freertos/port/xtensa/port.c:319

0x40082112: vPortExitCriticalSafe at C:/Users/HJYVM10/esp/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:607
 (inlined by) timer_list_unlock at C:/Users/HJYVM10/esp/esp-idf/components/esp_timer/src/esp_timer.c:310

0x400d4573: timer_process_alarm at C:/Users/HJYVM10/esp/esp-idf/components/esp_timer/src/esp_timer.c:377

0x400d458d: timer_task at C:/Users/HJYVM10/esp/esp-idf/components/esp_timer/src/esp_timer.c:386 (discriminator 1)

0x40087bf5: vPortTaskWrapper at C:/Users/HJYVM10/esp/esp-idf/components/freertos/port/xtensa/port.c:131

E (10306) task_wdt: Print CPU 1 backtrace

Backtrace:0x400842C1:0x3FFB11800x400828A5:0x3FFB11A0 0x4000BFED:0x3FFB6760 0x40087EA6:0x3FFB6770 0x400D68A4:0x3FFB6790 0x400D68AF:0x3FFB67C0 0x400D1829:0x3FFB67E0 0x400866C6:0x3FFB6800 0x40087BF5:0x3FFB6820
0x400842c1: esp_crosscore_isr at C:/Users/HJYVM10/esp/esp-idf/components/esp_system/crosscore_int.c:92

0x400828a5: _xt_lowint1 at C:/Users/HJYVM10/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111

0x40087ea6: vPortClearInterruptMaskFromISR at C:/Users/HJYVM10/esp/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:571
 (inlined by) vPortExitCritical at C:/Users/HJYVM10/esp/esp-idf/components/freertos/port/xtensa/port.c:319

0x400d68a4: esp_task_wdt_reset at C:/Users/HJYVM10/esp/esp-idf/components/esp_system/task_wdt.c:330

0x400d68af: idle_hook_cb at C:/Users/HJYVM10/esp/esp-idf/components/esp_system/task_wdt.c:80

0x400d1829: esp_vApplicationIdleHook at C:/Users/HJYVM10/esp/esp-idf/components/esp_system/freertos_hooks.c:51 (discriminator 1)

0x400866c6: prvIdleTask at C:/Users/HJYVM10/esp/esp-idf/components/freertos/tasks.c:3973 (discriminator 1)

0x40087bf5: vPortTaskWrapper at C:/Users/HJYVM10/esp/esp-idf/components/freertos/port/xtensa/port.c:131

E (15306) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (15306) task_wdt:  - IDLE (CPU 0)
E (15306) task_wdt: Tasks currently running:
E (15306) task_wdt: CPU 0: esp_timer
E (15306) task_wdt: CPU 1: IDLE
E (15306) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x400D6647:0x3FFB0B800x400828A5:0x3FFB0BA0 0x4000BFED:0x3FFB4760 0x40087EA6:0x3FFB4770 0x40082112:0x3FFB4790 0x400D4573:0x3FFB47B0 0x400D458D:0x3FFB47E0 0x40087BF5:0x3FFB4800
0x400d6647: task_wdt_isr at C:/Users/HJYVM10/esp/esp-idf/components/esp_system/task_wdt.c:183 (discriminator 3)

0x400828a5: _xt_lowint1 at C:/Users/HJYVM10/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111

0x40087ea6: vPortClearInterruptMaskFromISR at C:/Users/HJYVM10/esp/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:571
 (inlined by) vPortExitCritical at C:/Users/HJYVM10/esp/esp-idf/components/freertos/port/xtensa/port.c:319

0x40082112: vPortExitCriticalSafe at C:/Users/HJYVM10/esp/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:607
 (inlined by) timer_list_unlock at C:/Users/HJYVM10/esp/esp-idf/components/esp_timer/src/esp_timer.c:310

0x400d4573: timer_process_alarm at C:/Users/HJYVM10/esp/esp-idf/components/esp_timer/src/esp_timer.c:377

0x400d458d: timer_task at C:/Users/HJYVM10/esp/esp-idf/components/esp_timer/src/esp_timer.c:386 (discriminator 1)

0x40087bf5: vPortTaskWrapper at C:/Users/HJYVM10/esp/esp-idf/components/freertos/port/xtensa/port.c:131

igrr, 2022-08-13:
Looks like you are running into an integer overflow when calculating 3600*1000*1000 (the operations are treated as 32-bit by default).
Can you please try changing that to `3600LLU * 1000 * 1000`?

aboyuanna, 2022-08-13:
> Looks like you are running into an integer overflow when calculating (the operations are treated as 32-bit by default). Can you please try changing that to ?`3600*1000*1000``3600LLU * 1000 * 1000`

Thanks，the problem has been solved and I will close it.

closed 2022-08-13 by aboyuanna