espressif/esp-idf #9555: [BLE5] periodic advertising

chegewara, 2022-08-13:
Hi team,
I dont actually know if its bug or just bluetooth specs, because periodic advertising with bluedroid seems to work.
The problem i see is receiving periodic advertising with 1-2 packets late, where periodic advertising period is 2000ms, which is 3-4s later than advertising.
Probably it is just my "expectations" differ from how API works. Here is what i see:
- set periodic advertising with incrementing number, which is logged on 1 device
- about 2 seconds later receive synced periodic advertising with value which is less by 1 than number from previous step
- i would expect it to be the same value, but it is from 2 previous periodic advertising, not the current one, which means its from about 4000ms ago

It is not a problem, because i dont need real time data to be advertising, but i just am trying to understand and eventually if its really a bug to report it.

Thanks

chegewara, 2022-08-13:
Maybe not related question, but what coding is used in periodic advertising over coded PHY?
`ESP_BLE_GAP_PHY_OPTIONS_PREF_S2_CODING` or `ESP_BLE_GAP_PHY_OPTIONS_PREF_S8_CODING`
and is there an option to set it up?
I am asking because with S3 devkits i am able to achieve only about 100m+, when it should be possible to achieve at least 10 times farther.
https://esp32.com/viewtopic.php?f=19&t=21011#p77540

chegewara, 2022-08-13:
2 more things:
- sending periodic advertising with length 251 bytes is split (should not be 251, at least receiving buffer is 251 bytes size)
```
periodic adv report, sync handle 1 data status 1 data len 241 rssi -9
periodic adv report, sync handle 1 data status 0 data len 10 rssi -9
```
- sending periodic advertising with length 252 bytes is:
-- generating error, but still is sending data:
```
E (41945) BT_BTM: LE PA SetData: cmd err=0xc
E (41945) BT_HCI: CC evt: op=0x203f, status=0xc
```
-- data in advertising buffer is not set/replaced with new data, due to to this error
-- data is advertised anyway (old data), but the length is incorrect (should be 252, but its 300 bytes in total)
```
periodic adv report, sync handle 1 data status 1 data len 241 rssi -9
periodic adv report, sync handle 1 data status 0 data len 59 rssi -9
```

esp-cjh, 2022-08-15:
Hi @chegewara,
I did the test on esp32s3 and didn't find the problem you said. The periodic adv interval is 2000ms.
[image]

chegewara, 2022-08-15:
Yes, with `len == 28`, did you test with len 252+ bytes and updated value?
Like i said, first packet set for periodic advertising does not spit any errors. Then updating advertising packets with length > 251 bytes will not update advertising data and advertise only old data.
Also observer part is having problems with receiving fine periodic advertising shorter than 488 bytes, which you didnt test.

esp-cjh, 2022-08-15:
OK, I will test with length > 251.

esp-cjh, 2022-08-17:
Hi, @chegewara 
Refer to Core Spec 5.2 Vol4 PartE 7.8.62, If periodic advertising is currently enabled for the specified advertising set and Operation does not have the value 0x03, the Controller shall return the error code Command Disallowed (0x0C).
So you can't update advertising data with length > 251.

esp-cjh, 2022-08-17:
And what's the problem with receiving periodic adv data shorter than 488 bytes? I test with length = 400 and found no problems.

chegewara, 2022-08-17:
> Hi, @chegewara Refer to Core Spec 5.2 Vol4 PartE 7.8.62, If periodic advertising is currently enabled for the specified advertising set and Operation does not have the value 0x03, the Controller [...]

Ok, i will stick to len < 251, but im not sure its correct.

Here is another part of this exact section:
> If Operation indicates the start of new data (values 0x01 or 0x03), then any
existing partial or complete data shall be discarded. If the
Advertising_Data_Length parameter is 0, then Operation shall be 0x03; this
indicates that any existing partial or complete data shall be deleted and no new
data provided.

so, this command should remove all existing periodic adv data:
`esp_ble_gap_config_periodic_adv_data_raw(0, 0, data);`
instead we get this error:
`E (9411) BT_BTC: btc_gap_ble_arg_deep_copy 41 no mem`

chegewara, 2022-08-17:
> And what's the problem with receiving periodic adv data shorter than 488 bytes? I test with length = 400 and found no problems.

Sorry, it was typo, i meant longer than 488 bytes. This part of my previous report:
> setting 1st advertising packet with length 1000 bytes is working on advertiser side (no error), but there is some issue, because on observer side i see such errors:

and here is scanner/observer error:
```
periodic adv report, sync handle 1 data status 1 data len 241 rssi -9
E (2382238) BT_HCI: btu_ble_periodic_adv_report_evt, Invalid params, the adv len is to short.
periodic adv report, sync handle 1 data status 1 data len 247 rssi -9
E (2382255) BT_HCI: btu_ble_periodic_adv_report_evt, Invalid params, the adv len is to short.
periodic adv report, sync handle 1 data status 1 data len 247 rssi -9
E (2382273) BT_HCI: btu_ble_periodic_adv_report_evt, Invalid params, the adv len is to short.
periodic adv report, sync handle 1 data status 1 data len 247 rssi -9
periodic adv report, sync handle 1 data status 0 data len 9 rssi -9
```

BTW is there any API to get this:
```
LE Read Maximum Advertising Data Length Command
```
Core Spec 5.2 Vol4 PartE 7.8.57 (page 1355)

esp-cjh, 2022-08-17:
Hi @chegewara,
The problem you said we have fixed. But the changes have not been synced to github.

closed 2022-08-17 by chegewara