espressif/esp-idf #9556: Install.ps1 fails on v4.4.2

AEFeinstein, 2022-08-13:
## Environment

- Development Kit: none
- Kit version (for WroverKit/PicoKit/DevKitC): n/a
- Module or chip used:  n/a
- IDF version (run ``git describe --tags`` to find it):
    // v4.4.2
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // n/a
- Operating System: Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: No
- Power Supply: n/a

## Problem Description

In b28d7e6850936a9e5bbf542fa1d55adf6f9ae2f3 both ` --features=${FEATURES}` was added to the command to install the Python environment, and the variable was defined:
```powershell
$FEATURES = (python "$IDF_PATH/tools/install_util.py" extract features "$args")
```
But the variable is used and not defined in https://github.com/espressif/esp-idf/blob/v4.4.2/install.ps1, which leads to an error.

### Expected Behavior

The Python environment is installed

### Actual Behavior

The Python environment is not installed

### Steps to reproduce

1. On Windows, delete the local IDF folder
1. Delete the local tools folder
1. Clone the v4.4.2 tag
1. Run `install.ps1`

### Code to reproduce this issue

```powershell
git clone -b v4.4.2 --recursive https://github.com/espressif/esp-idf.git esp-idf-v4.4.2
cd esp-idf-v4.4.2/
install.ps1
```

## Debug Logs

```
Setting up Python environment
usage: idf_tools.py [-h] [--quiet] [--non-interactive] [--tools-json TOOLS_JSON] [--idf-path IDF_PATH] {list,check,export,install,download,install-python-env} ...
idf_tools.py: error: unrecognized arguments: --features=
```

dobairoland, 2022-08-15:
Hi @AEFeinstein. Thank you for letting us know. We'll fix it soon.

igrr, 2022-08-23:
The fix for release/v4.4 branch is available in https://github.com/espressif/esp-idf/commit/a1f03817975d29fc332e81d0420859bb8677053e and will be part of the next v4.4.3 bugfix release.

https://github.com/espressif/esp-idf/commit/5d1a51db65e2a9de71f2361d47317b4e5daf27b0 fixes the same bug for release/v4.3.

AEFeinstein, 2022-08-23:
Excellent. Is there a timetable for 4.4.3?

igrr, 2022-08-24:
Not at the moment, I'm afraid. Currently we are working on v4.2.4 bugfix release, next will be v4.3.4. Once these two releases are done, we'll start working on v4.4.3.

AEFeinstein, 2022-08-24:
Okay. Good luck on all the releases!

closed 2022-08-25 by dobairoland