espressif/esp-idf #9557: Hardware - No information regarding internal ESP32 capacitances

francis2tm, 2022-08-15:
This is not related to esp-idf, but usually you don't answer to questions made in the forum (own experience...).
THis is a question already raised in https://esp32.com/viewtopic.php?t=19972&start=10. I'm designing a custom board, I'm not using a module, I'm using the ESP32-D0WDR2-V3 SoC and I'm dimensioning the capacitors for the 40MHz crystal (I'm planning on using the 
ECS-400-10-37B-CKY-TR3 crystal). Quoting the big question in this thread:

> What is the input capacitance of the ESP32 internal Pierce oscillator?
> 
> Normally I would expect such a specification in the Datasheet or similar, but there seems to be no mention of it in any document from Espressif and also no information on the forums.
> 
> The Input Capacitance of the ESP32 has to be calculated in to the Load matching that the crystal sees.
> So:
> CL_desired = CL_crystal = ((Cap_XTAL_P + CL_esp_internal_p) * (Cap_XTAL_N + CL_esp_internal_N)) / (Cap_XTAL_P + CL_esp_internal_p + Cap_XTAL_N + CL_esp_internal_N) + CL_traces
> Or easier:
> CL_desired = CL_crystal = (Cap_XTAL_P + CL_esp_internal_p)//(Cap_XTAL_N + CL_esp_internal_N) + CL_traces
> 
> Without information form ESP about ESP32 input capacitance this formula will never be solvable.
> Hint: CL_traces are the PCB traces and typically amount to some 2pF.
> 
> Any updated from an ESP rep would be welcomed

I need to know the internal capacitances of the XTAL_P and XTAL_N pins, so that I can correctly calculate Cap_XTAL_P and Cap_XTAL_N values...
Any help regarding this is welcomed

krzychb, 2022-08-25:
The capacitance of pins is provided in the chip datasheet. Below is the value from [ESP32 Series Datasheet v3.8](https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf)
[image]

francis2tm, 2022-08-25:
Hey @krzychb,

The Cin of each pin is relative with GND? Or do I assume a series capacitor?

In short, in order to calculate the values of C1 and C2, is this rational correct?

[image]

krzychb, 2022-08-26:
Hey @francis2tm, the Cin is relative to GND.

francis2tm, 2022-08-26:
Thanks @krzychb again for the swift reply,

Therefore, the equation is (assuming C1 = C2, Cin = 2pF, Cstray = 2pF and Cload = 10pF) (// = paralell)
(C1 // C2 // Cin // Cin) + Cstray = Cload <=> C1 = C2 = 5.46pF
Is my rational correct?
It is my first time choosing load caps for crystals

Esp-Doc, 2025-02-07:
Hi @francis2tm , please check the input capacitance explained in the [Hardware Design Guidelines](https://docs.espressif.com/projects/esp-hardware-design-guidelines/en/latest/esp32/schematic-checklist.html#external-crystal-clock-source-compulsory). Thanks.

closed 2025-02-25 by Alvin1Zhang