espressif/esp-idf #9558: What is the correct usage of Fatfs in ESP32WROOM32D? Fatfs mount error 101 after some period.

Irfan93, 2022-08-15:
Environment

-     Development Kit: custom hardware
-     Kit version:
-     Module or chip used: ESP-WROOM-32D Module
-     IDF version (run git describe --tags to find it): v4.3.2 release
-     Build System: CMake | IDF.PY
-     Compiler version (run xtensa-esp32-elf-gcc --version to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
-     Operating System: Linux Kubuntu 22.04
-     Using an IDE?: No
-     Power Supply: external power

Hello everyone,

I am super new here in esp32 development and currently working on a project inherited from someone else.

Currently the fatfs partition is set to an allocated space of 4MiB with about 1.5MiB files occupying the fatfs.img (also fatfs image created to be 4096 bytes per sector). Initially the strategy used for OTA was to write the updated files(1.8MiB) to the fatfs.img and on reboot the older files are removed while the update files are renamed to be the current files. This method seems to cause fatfs corruption where the fatfs can't be mounted after about 2 or 3 OTA.

Now the new strategy implemented in effort to tackle the previous problem, we are reformatting the fatfs image every time an OTA is initiated. The reformat is done through f_mkfs function. A wrapper for f_mkfs was created based on its usage in the esp_vfs_fat_spiflash_mount such that:
```

esp_err_t esp_vfs_fat_spiflash_format_mount(const char* base_path,
    const char* partition_label,
    const esp_vfs_fat_mount_config_t* mount_config,
    wl_handle_t* wl_handle)
{
    esp_err_t result = ESP_OK;
    const size_t workbuf_size = 4096;
    void *workbuf = NULL;

    esp_partition_subtype_t subtype = partition_label ?
            ESP_PARTITION_SUBTYPE_ANY : ESP_PARTITION_SUBTYPE_DATA_FAT;
    const esp_partition_t *data_partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA,
                                                subtype, partition_label);
    if (data_partition == NULL) {
        ESP_LOGE(TAG, "Failed to find FATFS partition (type='data', subtype='fat', partition_label='%s'). Check the partition table.", partition_label);
        printf("Failed to find fatfs partition \r\n");
        return ESP_ERR_NOT_FOUND;
    }

    result = wl_mount(data_partition, wl_handle);
    if (result != ESP_OK) {
        ESP_LOGE(TAG, "failed to mount wear levelling layer. result = %i", result);
        printf(" failed to mount wear levelling layer \r\n");
        return result;
    }
    // connect driver to FATFS
    BYTE pdrv = 0xFF;
    if (ff_diskio_get_drive(&pdrv) != ESP_OK) {
        ESP_LOGD(TAG, "the maximum count of volumes is already mounted");
        printf(" the maximum count of volumes is already mounted \r\n");
        return ESP_ERR_NO_MEM;
    }
    ESP_LOGD(TAG, "using pdrv=%i", pdrv);
    char drv[3] = {(char)('0' + pdrv), ':', 0};

    result = ff_diskio_register_wl_partition(pdrv, *wl_handle);
    if (result != ESP_OK) {
        ESP_LOGE(TAG, "ff_diskio_register_wl_partition failed pdrv=%i, error - 0x(%x)", pdrv, result);
        printf(" ff_diskio_register_wl_partition failed pdrv \r\n");
        goto fail;
    }
    FATFS *fs;
    result = esp_vfs_fat_register(base_path, drv, mount_config->max_files, &fs);
    if (result == ESP_ERR_INVALID_STATE) {
        // it's okay, already registered with VFS
    } else if (result != ESP_OK) {
        ESP_LOGD(TAG, "esp_vfs_fat_register failed 0x(%x)", result);
        printf(" esp_vfs_fat_register failed \r\n");
        goto fail;
[... 120 lines cut ...]
[18:37:45:895] Fatfs path: /fatfs ␍␍␊
[18:37:45:917] 1017 sector total drive space. ␍␊
[18:37:45:917] 578 sector available. ␍␊
[18:37:45:917] SUCCESS mounted: fatfs␍␍␊
[18:37:45:949] SUCCESS unmounted: fatfs␍␍␊
[18:42:46:027] Format flag 0 ␍␍␊
[18:42:46:027] Fatfs path: /fatfs ␍␍␊
[18:42:46:076] fileSystem.c PartitionMount 938 ERROR_MEMORY Failed to mount: /fatfs error(101)␍␍␊
[18:42:46:107] fileSystem.c PartitionunMount 957 ERROR_MEMORY Failed to unMount: /fatfs error(103)␍␍␊
[18:42:46:107] fileSystem.c FatfsunMount 877 ERROR_FAIL FATFS unmounting failed ␍␍␊
[18:42:46:140] Format flag 0 ␍␍␊
[18:42:46:140] Fatfs path: /fatfs ␍␍␊
[18:42:46:227] fileSystem.c PartitionMount 938 ERROR_MEMORY Failed to mount: /fatfs error(101)␍␍␊
[18:42:46:227] fileSystem.c PartitionunMount 957 ERROR_MEMORY Failed to unMount: /fatfs error(103)␍␍␊
[18:42:46:227] fileSystem.c FatfsunMount 877 ERROR_FAIL FATFS unmounting failed ␍␍␊
[18:47:46:273] Format flag 0 ␍␍␊
[18:47:46:289] Fatfs path: /fatfs ␍␍␊
[18:47:46:363] fileSystem.c PartitionMount 938 ERROR_MEMORY Failed to mount: /fatfs error(101)␍␍␊
[18:47:46:363] fileSystem.c PartitionunMount 957 ERROR_MEMORY Failed to unMount: /fatfs error(103)␍␍␊
[18:47:46:363] fileSystem.c FatfsunMount 877 ERROR_FAIL FATFS unmounting failed ␍␍␊
[18:47:46:411] Format flag 0 ␍␍␊
[18:47:46:411] Fatfs path: /fatfs ␍␍␊
[18:47:46:492] fileSystem.c PartitionMount 938 ERROR_MEMORY Failed to mount: /fatfs error(101)␍␍␊
[18:47:46:492] fileSystem.c PartitionunMount 957 ERROR_MEMORY Failed to unMount: /fatfs error(103)␍␍␊
[18:47:46:492] fileSystem.c FatfsunMount 877 ERROR_FAIL FATFS unmounting failed ␍␍␊

igrr, 2022-08-15:
Hi @Irfan93, if your application is running out of heap memory after doing some operation repeatedly, this could be caused by a memory leak. Please check this section of the IDF programming guide: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#heap-tracing. You can try enabling heap tracing to find the code which is allocating but not freeing the memory.

Irfan93, 2022-08-16:
Hey @igrr . Thanks for the reply. I came to the same conclusion even though my heap space before mounting and after unmounting, remains the same as I am freeing the malloced space correctly but I assume there could be some sort memory de-fragmentation happening.

For reference, currently I am implementing the logic to pull the required data from fatfs to live in flash array as to prevent the mount/unmount every 5 minutes. On top of that, a reboot event in case of the memory error when a fatfs operation is concerned.

closed 2023-02-20 by Alvin1Zhang