espressif/esp-idf #9559: 关于上行流量控制的窗口调整问题

FPOOOOOOO, 2022-08-15:
## Problem Description
[在官方网站手册上有这样一句话：](https://docs.espressif.com/projects/esp-idf/zh_CN/v4.4.1/esp32/api-guides/esp-wifi-mesh.html?highlight=mesh#mesh-network-performance)

ESP-WIFI-MESH 依赖父节点来控制其直接子节点的上行数据流。为了防止父节点的消息缓冲因上行传输过载而溢出，父节点将为每个子节点分配一个称为 接收窗口 的上行传输配额。 每个子节点均必须申请接收窗口才允许进行上行传输。接收窗口的大小可以动态调整。完成从子节点到父节点的上行传输包括以下步骤：

请问一下这个接收窗口在哪？我要如何调整来改变上行的流量控制呢？

## Other items if possible

我希望每个节点的上行数据流控制在1M，如果叶节点上行数据流控制在1M的话，那么中间节点的上行数据流时候就会有N个叶节点往上转发的数据流经过，那么就没法控制在1M了。

shenjun7, 2022-08-16:
可以通过 `esp_mesh_set_xon_qsize()` 设定 `xon_qsize`, 上行窗口数与 `xon_qsize` 成正比, 与最大连接数成反比．

FPOOOOOOO, 2022-08-16:
> 可以通过 `esp_mesh_set_xon_qsize()` 设定 `xon_qsize`, 上行窗口数与 `xon_qsize` 成正比, 与最大连接数成反比．

你好，请问最大连接数是指idf menuconfig里预先设置的连接数（6）吗？
[image]

shenjun7, 2022-08-17:
是的

Alvin1Zhang, 2022-08-24:
感谢反馈，后续如仍有问题，可以重新开启此问题，谢谢。

closed 2022-08-24 by Alvin1Zhang

FPOOOOOOO, 2022-11-01:
> 可以通过 `esp_mesh_set_xon_qsize()` 设定 `xon_qsize`, 上行窗口数与 `xon_qsize` 成正比, 与最大连接数成反比．
还想请问一下这个xon_qsize最大能设置为多少，以及xon_qsize*max connection是否为一个定值，定值是多少？
[image]

FPOOOOOOO, 2022-11-01:
> 可以通过 `esp_mesh_set_xon_qsize()` 设定 `xon_qsize`, 上行窗口数与 `xon_qsize` 成正比, 与最大连接数成反比．

还想请问一下这个xon_qsize最大能设置为多少，以及xon_qsize*max connection是否为一个定值，定值是多少？
@shenjun7 
[image]

zhangyanjiaoesp, 2023-01-13:
@FPOOOOOOO 
Please refer this:

> Using esp_mesh_set_xon_qsize() users may configure the RX queue size, default:32. If this size is too large, and esp_mesh_recv_toDS() isn't called in time, there is a risk that a great deal of memory is occupied by the pending packets. If this size is too small, it will impact the efficiency on upstream. How to decide this value depends on the specific application scenarios.

https://github.com/espressif/esp-idf/blob/8a7f6af6256f4c954053e08e819f55b93ddc1ff1/components/esp_wifi/include/esp_mesh.h#L736

You can set xon_qsize to 128, but you should consider the memory occupy and application scenario also.  The `xon_qsize` represents the RX queue size of this node, the average number of window allocated to a child node is： `wnd = xon_qsize / (2 * max_connection + 1)`. However, the window of each child node is not strictly equal to the average value, is affected by traffic also.

FPOOOOOOO, 2023-01-13:
> @FPOOOOOOO Please refer this:
>
>
> https://github.com/espressif/esp-idf/blob/8a7f6af6256f4c954053e08e819f55b93ddc1ff1/components/esp_wifi/include/esp_mesh.h#L736
>
> You can set xon_qsize to 128, but you [...]

Yeah, thank you. I have already set xon_qsize to 128 in idf.py menuconfig-MWifi. And I have tried to max number to 255, If I got the number to 256 then the log will be warning. I think this can explain my throughput issue also. 
[ISSUE](https://github.com/espressif/esp-mdf/issues/301)