espressif/esp-idf #9560: Memory fragmentation after open_memstream()

HarryHpd, 2022-08-15:
open memstream is reducing minimum heapsize in my application dramatically.

For example from 151.989 Bytes to 86.440 Bytes

This can be reproduced by...

FILE *stream;
	char *buf;
	size_t len;

	stream = open_memstream (&buf, &len);
	cleanupFileBuffer(stream, &buf);
.......
void cleanupFileBuffer(FILE *fp, char **buf) {
    fflush(fp);
    fclose(fp);
    free(*buf);
}

Free Heap size is not affected by this issue.

I am using Espressif IDE updatet to the latest version.

ESP-IDF is: 4.4.2

igrr, 2022-08-15:
Hi @HarryHpd,
Indeed `open_memstream` is not very efficient in its use of heap. It reallocates the destination buffer on every write operation which fills up the buffer: https://github.com/espressif/newlib-esp32/blob/78c7e93abd86181307ba9a050f43a353f8853e62/newlib/libc/stdio/open_memstream.c#L111-L126. Here `realloc` function is called, which means that the old buffer is kept around and the new buffer is allocated somewhere else in heap, then finally the old buffer is released. Since the buffer size grows every time, this leads to fragmentation.

To avoid this, you can either set the `len` to the expected maximum size of the buffer. This value will be used as a hint, and the buffer will be allocated with that specified size at once.

Alternatively, you can use `fmemopen`, which works with a fixed size buffer. Since you control the buffer in this case, you can also manage the heap fragmentation better.

HarryHpd, 2022-08-16:
...yes, fmemopen did the trick. You can close this incident.

Many thanks for your quick reply.

Harry

closed 2022-08-16 by igrr