espressif/esp-idf #9562: mbedtls in IDF collides with my own version

mzimmersfluidra, 2022-08-16:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32U|]
- IDF version (run ``git describe --tags`` to find it): v4.4.1
- Build System:         [|CMake|]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     [Windows|]
- (Windows only) environment type: [|PowerShell].
- Using an IDE?: [|Yes VS Code
- Power Supply:         [|external 5V|]

## Problem Description

I'm porting an old application with its own md5.c/.h files. When I try to build, I get an error
`C:/Espressif/frameworks/esp-idf-v4.4.1/components/mbedtls/mbedtls/library/md5.c:391: multiple definition of `mbedtls_md5'; esp-idf/main/libmain.a(md5.c.obj):C:\repos\iQPump-ESP32\fw\build/../main/md5.c:322: first defined here`
### Expected Behavior
To build properly.
### Actual Behavior
Build error above.

```cpp
    mbedtls_md5_init(&Zctx);
    mbedtls_md5_starts(&Zctx);
```
It's noteworthy that the linker finds the call to mbedtls_md5_init, but not the call to mbedtls_md5_starts (which I understand is deprecated).

I've tried removing my own files, and using those from the IDF, but I get an undefined reference to mbedtls_md5_starts (probably because it's flagged as deprecated). But I'd much prefer to use my own files.

How do I tell the build system that I want to use my own md5 files, and not those from the IDF?

Thanks...

mz

mahavirj, 2022-08-18:
@mzimmersfluidra 

Please try following:

- Disable following config option: https://github.com/espressif/esp-idf/blob/4d14c2ef2d9d08cd1dcbb68a8bb0d76a666e2b4b/components/mbedtls/Kconfig#L439

- Change following line to `#undef MBEDTLS_MD5_C`: https://github.com/espressif/esp-idf/blob/4d14c2ef2d9d08cd1dcbb68a8bb0d76a666e2b4b/components/mbedtls/port/include/mbedtls/esp_config.h#L1923

mahavirj, 2022-08-23:
@mzimmersfluidra If you could host your minimal project somewhere then I can help to check it out. Please create minimal application that contains your own `md5` implementation that you would like to use.

mzimmersfluidra, 2022-08-25:
@mahavirj thank you for the offer, but I believe I found the problem. My md5 implementation contained the routine:
`void mbedtls_md5( const unsigned char *input, size_t ilen, unsigned char output[16] )
`
Which is also defined in the IDF. I commented out this routine in my implementation, and things seem to work now.

I still don't understand why the other routines don't collide, but I guess my problem is resolved. Thank you for looking at it.

closed 2022-08-25 by mzimmersfluidra