espressif/esp-idf #9564: API from TWDT for getting the task that not reset the WDT

Eran125, 2022-08-16:
I'm using the TWDT to find a task or tasks that runs too much time.
While develop and debug I can use the terminal and the prints to find the problematics task, but in real time the device is in the field and not connected to PC. I'm saving the reset event and trigger in Log (in the Flash).
I like to add to the Log the tasks that caused the WDT reset, for this I need API to read the "twdt_config" list and check the "has_reset" variable.

This are my settings:
chip esp32s3
IDF 4.4
IDE VScode

Dazza0, 2022-08-17:
@Eran125 Have you considered using the `esp_task_wdt_isr_user_handler()`. It's a user provided callback that will run when the TWDT triggers.

From the callback, you can get the currently running tasks on each core via `xTaskGetCurrentTaskHandleForCPU()`.

Eran125, 2022-08-17:
Hi, 
I'm using the esp_task_wdt_isr_user_handler(), but I don't know how to get the busy task.
I try to print the current handle task as you suggest but I got idle task in both cpu.
ESP_EARLY_LOGE(TAG, "WDT Task:%s | %s",pcTaskGetName(xTaskGetCurrentTaskHandleForCPU(1)),pcTaskGetName(xTaskGetCurrentTaskHandleForCPU(0)));

Dazza0, 2022-08-19:
@Eran125 Could you attach a copy of the log when the TWDT times out? What does the log say w.r.t which task triggered the time out?

Eran125, 2022-08-22:
E (5377) task_wdt: Task watchdog got triggered. The following tasks did not reset the [22/08/22 - 07:07:33:541] watchdog in time:
E (5377) task_wdt:  - ESI_UART (CPU 0/1)
E (5377) task_wdt: Tasks currently running:
E (5377) task_wdt: CPU 0: IDLE
E (5377) task_wdt: CPU 1: IDLE
E (5377) Main: WDT Task:IDLE | IDLE
E (5377) task_wdt: Aborting.

Dazza0, 2022-08-22:
@Eran125 After subscribing `ESI_UART` task to the TWDT, did you make sure that `ESI_UART` task runs periodically? It looks like `ESI_UART` isn't running at all (thus isn't feeding the watchdog) as both cores are running the idle tasks.

Eran125, 2022-08-22:
I know the "ESI_UART" task isn't feed the TWDT, I did it on purpose. I'm learning how to handle with the TWDT, I need to handle it while the device is close and not connect to PC with terminal. after a reset I save the reason (software or power up or WDT etc.) in my log in the Flash but if the reason was WDT I must save also in the Flash the task that not feed the WDT.
As you can see in the print I send, the function "xTaskGetCurrentTaskHandleForCPU" return "IDLE" so it's not helping.
Please advise.

Dazza0, 2022-08-23:
@Eran125 Ah sorry, I get your requirements now. In that case, I think the easiest solution is to add another `weak` callback similar to `esp_task_wdt_isr_user_handler()`. Something like `esp_task_wdt_isr_user_get_timeout_task()` that gets called for each for each task that hasn't fed the TWDT on a timeout.

I'd prefer not to expose the TWDT's internal data structures (i.e., the task linked list) in case we end up changing the implementation.

I've attached the patch below based off of [v4.4.2](https://github.com/espressif/esp-idf/tree/v4.4.2) and the example below demonstrates how to use it. Please not that the `esp_task_wdt_isr_user_get_timeout_task()` handler is called from an ISR and inside a critical section, thus some restrictions apply (such as no blocking, and keeping it short to avoid a IWDT timeout).

```patch
From 0a0cf6220a75d40ff275be17a9da0bd69b6f8ca0 Mon Sep 17 00:00:00 2001
From: Darian Leung <darian@espressif.com>
Date: Tue, 23 Aug 2022 17:09:33 +0800
Subject: [PATCH] twdt: Add user handler to get timed out tasks

---
 components/esp_system/task_wdt.c | 6 ++++++
 1 file changed, 6 insertions(+)

diff --git a/components/esp_system/task_wdt.c b/components/esp_system/task_wdt.c
index cde04ce719..0124827a50 100644
--- a/components/esp_system/task_wdt.c
+++ b/components/esp_system/task_wdt.c
@@ -118,6 +118,11 @@ static void reset_hw_timer(void)
     }
 }
 
+void __attribute__((weak)) esp_task_wdt_isr_user_get_timeout_task(TaskHandle_t task_handle)
+{
+    ;
+}
+
 /*
  * This function is called by task_wdt_isr function (ISR for when TWDT times out).
  * It can be redefined in user code to handle twdt events.
@@ -155,6 +160,7 @@ static void task_wdt_isr(void *arg)
     ESP_EARLY_LOGE(TAG, "Task watchdog got triggered. The following tasks did not reset the watchdog in time:");
     for (twdttask=twdt_config->list; twdttask!=NULL; twdttask=twdttask->next) {
         if (!twdttask->has_reset) {
+            esp_task_wdt_isr_user_get_timeout_task(twdttask->task_handle);
             cpu=xTaskGetAffinity(twdttask->task_handle)==0?DRAM_STR("CPU 0"):DRAM_STR("CPU 1");
             if (xTaskGetAffinity(twdttask->task_handle)==tskNO_AFFINITY) {
                 cpu=DRAM_STR("CPU 0/1");
-- 
2.25.1

```

```c
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_rom_sys.h"

void esp_task_wdt_isr_user_get_timeout_task(TaskHandle_t task_handle)
{
    esp_rom_printf("Task %s timedout\n", pcTaskGetTaskName(task_handle));
}

void app_main(void)
{
    while (1)
    {
        // Loop to trigger a WDT timeout
        ;
    }
}
```

Eran125, 2022-08-24:
Hi,
This is good solution. I know this function called from ISR, in this cases I save the information in no-init RAM section and handle it (like save in Flash) after reset and system restart.
Is this patch will release in the next version or it already available?

Dazza0, 2022-09-28:
@Eran125 I'm closing this issue as we aren't planning to add this callback feature to IDF. Please take a look at the core dump feature to see if it meets your requirements.

closed 2022-09-28 by Dazza0

zzff-sys, 2024-04-26:
> @Eran125您是否考虑过使用`esp_task_wdt_isr_user_handler()`.这是用户提供的回调，将在 TWDT 触发时运行。
> 
> 从回调中，您可以通过 获取每个核心上当前正在运行的任务`xTaskGetCurrentTaskHandleForCPU()`。

I use esp_task_wdt_isr_user_handler () to declare the watchdog interrupt function, but it doesn't seem to enter the function when timeout occurs and keeps restarting. How do I use it correctly

Dazza0, 2024-04-30:
@zzff-sys Simply just define a global function named `esp_task_wdt_isr_user_handler()` and it will be called on a task watchdog timeout