espressif/esp-idf #9569: [UART] Get received parity bit value directly

marcoccio, 2022-08-16:
**Is your feature request related to a problem? Please describe.**

Interfacing with serial peripherals that use 9 bit words without parity, using 8 bit words capable hardware would be much easier if there was a way to directly read the 8 bit word and the "parity" bit received by the ESP32 RX hw.  

**Describe the solution you'd like**

I would like a function that simply returns the value of the "parity" bit received by the hardware so that later one can reconstruct the 9 bit word. 

**Describe alternatives you've considered**
I tried using the events generated by the UART driver, but it is slow and I could not get a consistent result. I tried reading the UART2 structure directly when a DATA event is generated but that didn't work either. Lastly I tried implementing my own ISR but i could not make it work due to my lack of knowledge. 

Sorry if this request is redundant, or if there is already a way, I could not for my life figure this out even googling for hours.

law-ko, 2022-12-15:
In addition, it seems like parity bits isn't working on my end. Setting one ESP32 on `UART_PARITY_ODD` and another as `UART_PARITY_EVEN` doesn't really trigger anything or error logs.

dazca, 2023-05-18:
I was unable to see how we can obtain the status of the 9th bit in every received packet. But we can access the byte and check if there has been a parity error. If we know the state in which we left the driver the last time we wrote (even=0 or odd=1), we can infer the 9th bit as follows:

`bit9set = byte_parity XOR serial_parity XOR parity_error`

The only thing I miss is the possibility of executing an event or ISR between each received packet. 

**In events**, we can use UART_DATA or UART_PARITY_ERR, but I am unsure of how these events work. Experimenting, it seems that UART_DATA is only triggered when we've finished reading, and UART_PARITY_ERR is triggered for every error. However, I haven't found a way to determine in which received byte we are when the error occurs.

avrguru, 2026-03-04:
The ESP32 UART FIFO's don't buffer the error flags (like Parity error) alongside the data, they simply assert an IRQ, and set a specific bit-flag in UART_INT_RAW_REG that can be examined inside your ISR. For the ESP32-S3, section 26.4.4 describes this in the Reference manual.  In part, "When the receiver detects a parity bit error in the data received, a UART_PARITY_ERR_INT interrupt
is generated, and the erroneous data are still stored into the RX FIFO."