espressif/esp-idf #9570: get size of a directory

farbodpm, 2022-08-17:
I am working on a project and I need the directory size. Maybe returning size in dirent structure be great.

`struct dirent {
    ino_t d_ino;          /*!< file number */
    uint8_t d_type;     /*!< not defined in POSIX, but present in BSD and Linux */
    uint32_t size;
#define DT_UNKNOWN  0
#define DT_REG      1
#define DT_DIR      2
#if __BSD_VISIBLE
#define MAXNAMLEN 255
    char d_name[MAXNAMLEN+1];   /*!< zero-terminated file name */
#else
    char d_name[256];
#endif
};`

o-marshmallow, 2022-08-17:
Hi @farbodpm ,

What do you mean by the directory size? Do you mean the size of a directory entry on the disk ? Or do you mean the size of all the files (and directories recursively) in the directory ?

farbodpm, 2022-08-18:
Hi, 
When U read a directory it would be great to have files size and folders. Like ll command in linux. I use this value to check my files size. checking files one by one would be unoptimized.

o-marshmallow, 2022-08-18:
On Ubuntu `ll` command is an alias to `ls -alF`. For the directories, this command doesn't show the real space usage, it shows the size of the directory entry itself takes on the disk: 4096 bytes. It is always 4KB, not matter what is in the directory.

If you want to retreive the space usage of the files and sub-directories contained in it, you have to use other commands, such as `du -sh`.  Such command will recursively calculate the size of all the files in the sub-directories. This is not done by the kernel, this is done in user space.

As such, in IDF the same pattern has to be followed, you'll need to recursively access the directories and sub-directories to calculate the size of a folder.

igrr, 2022-08-18:
I have edited the formatting of the code block in @farbodpm's original post. I think now I understand the suggestion — which is to add a `size` member to `struct dirent`. So the request is basically to report the size of every file when iterating over a directory. @farbodpm Is my understanding correct?

farbodpm, 2022-08-18:
> I have edited the formatting of the code block in @farbodpm's original post. I think now I understand the suggestion — which is to add a `size` member to `struct dirent`. So the request is basically [...]

That's exactly what I meant. And as @o-marshmallow said the  folders size can be fixed folder size or the exact amount. I think the exact amount is not optimized and also not essential.

igrr, 2022-08-18:
@farbodpm Do you have any example of an OS or a system where `struct dirent` has the `size` member? I have checked the definition of this structure for Linux and BSD, and both don't have such a field. Normally you need to call `stat` function to get `struct stat` for the given file, and the size will be available in the returned `struct stat`.
We would like to avoid deviating from the standard behavior here.

Do you have any issue calling `fstat` for every file in the directory, as you are iterating over the files anyway?

farbodpm, 2022-08-18:
yes ... . Littlefs as one of more famous library return the size of the file when you read directory. I think that's wise because in embedded world the file size is small and the cost of checking the directory files one by one is high because of low speed chips.

farbodpm, 2022-08-23:
@rrtandler I already implemented this feature. Can I help you with this?

RathiSonika, 2024-04-25:
Similar to #10220. Please find the solution in the commit 3d813af.

closed 2024-04-25 by RathiSonika