espressif/esp-idf #9574: ESP32 Crash 'Cache disabled but cached memory region accessed', uart.c:972 & ringbuf.c:32

mws-rmain, 2022-08-17:
## Environment

- PCB: Custom
- Module or chip used:  ESP32-WROOM-32U
- IDF version: v4.4.1-185-gc5deb7a91
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: Battery

## Problem Description

Very infrequently on bootup, I experience a random crash. The crash dump (see below) indicates 'Cache disabled, but cached memory region accessed', and the stack implicates a UART ringbuffer. 

In the design, three UARTs are in use: one for external IO (&diagnostics), and two connected to other devices. Is there some way I can confirm WHICH UART is involved in the crash (i.e., Crash dump registers, etc)?

The offending line of UART.C:972 is:
`if (pdFALSE == xRingbufferSendFromISR(p_uart->rx_ring_buf, p_uart->rx_data_buf, p_uart->rx_stash_len, &HPTaskAwoken)) {
`
Is it possible there might be a 'window' during early boot when cache might be temporarily disabled, when receipt of data over the UART might cause this condition?

Also:
CONFIG_UART_ISR_IN_IRAM=y (see attached sdkconfig file)

## Debug Logs

```
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed). 

Core  1 register dump:
PC      : 0x401be638  PS      : 0x00060035  A0      : 0x8008bedf  A1      : 0x3ffbeec0  
0x401be638: prvCheckItemFitsByteBuffer at /home/mainr/esp/esp-idf/components/esp_ringbuf/ringbuf.c:325

A2      : 0x3ffeb5d8  A3      : 0x3ffeb4d0  A4      : 0x00000078  A5      : 0x3ffbeef0  
A6      : 0x401be638  A7      : 0x3ffeb6bc  A8      : 0x80091c54  A9      : 0x3ffbeea0  
0x401be638: prvCheckItemFitsByteBuffer at /home/mainr/esp/esp-idf/components/esp_ringbuf/ringbuf.c:325

A10     : 0x3ffeb5d8  A11     : 0x00000078  A12     : 0x00060021  A13     : 0x00060023  
A14     : 0x00000003  A15     : 0x00060c23  SAR     : 0x00000017  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x401be635:0x3ffbeec00x4008bedc:0x3ffbeee0 0x40082ff1:0x3ffbef40 0x4008c96a:0x3ffb7c50 0x4008aa3e:0x3ffb7c70 0x4009228d:0x3ffb7c90 
0x401be635: prvAcquireItemNoSplit at ringbuf.c:?
0x4008bedc: uart_rx_intr_handler_default at /home/mainr/esp/esp-idf/components/driver/uart.c:972
0x40082ff1: _xt_lowint1 at /home/mainr/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111
0x4008c96a: spi_flash_op_block_func at /home/mainr/esp/esp-idf/components/spi_flash/cache_utils.c:122
0x4008aa3e: ipc_task at /home/mainr/esp/esp-idf/components/esp_ipc/src/esp_ipc.c:74
0x4009228d: vPortTaskWrapper at /home/mainr/esp/esp-idf/components/freertos/port/xtensa/port.c:131
```

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/9364441/sdkconfig.zip)

igrr, 2022-08-19:
Hi @mws-rmain,
It looks like the same issue as https://github.com/espressif/esp-idf/issues/9198. We have recently fixed it, the backport on release/v4.4 branch is available in https://github.com/espressif/esp-idf/commit/a940064748fcf9da26592516258fe65d2e52aa9a. Could you please try updating your branch to that commit and see if that solves the problem?

mws-rmain, 2022-08-19:
Thanks, Ivan.
It does indeed look like the same issue. I'll incorporate [a940064](https://github.com/espressif/esp-idf/commit/a940064748fcf9da26592516258fe65d2e52aa9a) & re-test. In my case, this issue was really difficult to capture in the first place, so I won't be able to offer definitive proof that it is fixed, but I'll test until it's statistically probable.
Thanks for the quick response - I don't know how I missed [#9198](https://github.com/espressif/esp-idf/issues/9198) in my pre-post search.

closed 2022-08-19 by mws-rmain