espressif/esp-idf #9579: assert failed: prvReturnItemDefault ringbuf.c:623 (pxCurHeader->xItemLen <= pxRingbuffer->xMaxItemSize)

hechaotudou, 2022-08-18:
I (9650) hh---dd: length---1---=456
I (9650) hh---dd:  level0=0     duration0=3881  level1=1        duration1=1376
I (9650) hh---dd:  level0=0     duration0=664   level1=1        duration1=1115
I (9660) hh---dd:  level0=0     duration0=664   level1=1        duration1=1090
I (9660) hh---dd:  level0=0     duration0=640   level1=1        duration1=393
I (9670) hh---dd:  level0=0     duration0=639   level1=1        duration1=394
I (9680) hh---dd:  level0=0     duration0=639   level1=1        duration1=394
I (9690) hh---dd:  level0=0     duration0=639   level1=1        duration1=1115
I (9690) hh---dd:  level0=0     duration0=640   level1=1        duration1=393
I (9700) hh---dd:  level0=0     duration0=639   level1=1        duration1=374
I (9710) hh---dd:  level0=0     duration0=663   level1=1        duration1=1116
I (9710) hh---dd:  level0=0     duration0=639   level1=1        duration1=1114
I (9720) hh---dd:  level0=0     duration0=640   level1=1        duration1=394
I (9730) hh---dd:  level0=0     duration0=639   level1=1        duration1=1115
I (9730) hh---dd:  level0=0     duration0=640   level1=1        duration1=393
I (9740) hh---dd:  level0=0     duration0=639   level1=1        duration1=394
I (9750) hh---dd:  level0=0     duration0=640   level1=1        duration1=1115
I (9750) hh---dd:  level0=0     duration0=639   level1=1        duration1=1095
I (9760) hh---dd:  level0=0     duration0=662   level1=1        duration1=394
I (9770) hh---dd:  level0=0     duration0=640   level1=1        duration1=1115
I (9770) hh---dd:  level0=0     duration0=664   level1=1        duration1=1090
I (9780) hh---dd:  level0=0     duration0=664   level1=1        duration1=369
I (9790) hh---dd:  level0=0     duration0=639   level1=1        duration1=394
I (9790) hh---dd:  level0=0     duration0=640   level1=1        duration1=1114
I (9800) hh---dd:  level0=0     duration0=664   level1=1        duration1=369
I (9810) hh---dd:  level0=0     duration0=639   level1=1        duration1=374
I (9810) hh---dd:  level0=0     duration0=663   level1=1        duration1=1115
I (9820) hh---dd:  level0=0     duration0=664   level1=1        duration1=369
I (9830) hh---dd:  level0=0     duration0=639   level1=1        duration1=393
I (9830) hh---dd:  level0=0     duration0=640   level1=1        duration1=393
I (9840) hh---dd:  level0=0     duration0=640   level1=1        duration1=394
I (9850) hh---dd:  level0=0     duration0=664   level1=1        duration1=368
I (9850) hh---dd:  level0=0     duration0=640   level1=1        duration1=393
I (9860) hh---dd:  level0=0     duration0=664   level1=1        duration1=349
I (9870) hh---dd:  level0=0     duration0=688   level1=1        duration1=369
I (9870) hh---dd:  level0=0     duration0=639   level1=1        duration1=393
I (9880) hh---dd:  level0=0     duration0=664   level1=1        duration1=369
I (9890) hh---dd:  level0=0     duration0=665   level1=1        duration1=368
I (9890) hh---dd:  level0=0     duration0=639   level1=1        duration1=394
I (9900) hh---dd:  level0=0     duration0=664   level1=1        duration1=369
I (9910) hh---dd:  level0=0     duration0=664   level1=1        duration1=369
I (9910) hh---dd:  level0=0     duration0=665   level1=1        duration1=348
I (9920) hh---dd:  level0=0     duration0=687   level1=1        duration1=369
I (9930) hh---dd:  level0=0     duration0=664   level1=1        duration1=369
I (9930) hh---dd:  level0=0     duration0=640   level1=1        duration1=1114
I (9940) hh---dd:  level0=0     duration0=666   level1=1        duration1=367
I (9950) hh---dd:  level0=0     duration0=640   level1=1        duration1=394
I (9950) hh---dd:  level0=0     duration0=665   level1=1        duration1=1089
I (9960) hh---dd:  level0=0     duration0=639   level1=1        duration1=394
I (9970) hh---dd:  level0=0     duration0=639   level1=1        duration1=374
I (9970) hh---dd:  level0=0     duration0=663   level1=1        duration1=1115
I (9980) hh---dd:  level0=0     duration0=664   level1=1        duration1=1090
I (9990) hh---dd:  level0=0     duration0=639   level1=1        duration1=394
I (9990) hh---dd:  level0=0     duration0=639   level1=1        duration1=393
I (10000) hh---dd:  level0=0    duration0=666   level1=1        duration1=368
I (10010) hh---dd:  level0=0    duration0=639   level1=1        duration1=393
I (10010) hh---dd:  level0=0    duration0=665   level1=1        duration1=368
I (10020) hh---dd:  level0=0    duration0=666   level1=1        duration1=347
I (10030) hh---dd:  level0=0    duration0=663   level1=1        duration1=1115
I (10030) hh---dd:  level0=0    duration0=664   level1=1        duration1=370
I (10040) hh---dd:  level0=0    duration0=664   level1=1        duration1=1089
I (10050) hh---dd:  level0=0    duration0=640   level1=1        duration1=393
I (10050) hh---dd:  level0=0    duration0=693   level1=1        duration1=340
I (10060) hh---dd:  level0=0    duration0=666   level1=1        duration1=367
I (10070) hh---dd:  level0=0    duration0=640   level1=1        duration1=393
I (10080) hh---dd:  level0=0    duration0=639   level1=1        duration1=374
I (10080) hh---dd:  level0=0    duration0=687   level1=1        duration1=370
I (10090) hh---dd:  level0=0    duration0=663   level1=1        duration1=1091
I (10100) hh---dd:  level0=0    duration0=665   level1=1        duration1=368
I (10100) hh---dd:  level0=0    duration0=665   level1=1        duration1=368
[... 105 lines cut ...]

I (190) cpu_start: App cpu up.
I (218) cpu_start: Pro cpu start user code
I (218) cpu_start: cpu freq: 160000000
I (218) cpu_start: Application information:
I (223) cpu_start: Project name:     aaa_ir_protocols
I (228) cpu_start: App version:      1
I (233) cpu_start: Compile time:     Aug 17 2022 23:26:23
I (239) cpu_start: ELF file SHA256:  409a30caa4ca4229...
I (245) cpu_start: ESP-IDF:          v4.4
I (250) heap_init: Initializing. RAM available for dynamic allocation:
I (257) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (263) heap_init: At 3FFB2CB0 len 0002D350 (180 KiB): DRAM
I (269) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (276) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (282) heap_init: At 4008C3D8 len 00013C28 (79 KiB): IRAM
I (289) spi_flash: detected chip: generic
I (293) spi_flash: flash io: dio
W (297) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (311) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (4310) hh---dd: length---1---=4
I (4310) hh---dd:  cmd---=0
I (22830) hh---dd: length---1---=4
I (22830) hh---dd:  cmd---=0

Alvin1Zhang, 2022-08-18:
Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

suda-morris, 2022-08-18:
@hechaotudou Could you share your initialization code of the RMT part? Note, ESP32's hardware can't receive RMT frame whose length is longer than the block size. And could you try increasing the block size and the ringbuffer size in your case?

FYI, on the 5.0 branch, we have added a new RMT driver, which can handle the incoming packets in an asynchronous way, see the example: https://github.com/espressif/esp-idf/blob/master/examples/peripherals/rmt/ir_nec_transceiver/main/ir_nec_transceiver_main.c

hechaotudou, 2022-08-18:
@Alvin1Zhang 

开发板：esp32-s(Nodemcu)

/********************code***************start****************/
/* IR protocols example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include <string.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "driver/rmt.h"
#include "ir_tools.h"
static const char *TAG = "hh---dd";
static rmt_channel_t example_rx_channel = CONFIG_EXAMPLE_RMT_RX_CHANNEL;

static void example_ir_rx_task(void *arg)
{
    uint32_t addr = 0;
    uint32_t cmd = 0;
    size_t length = 0;
    bool repeat = false;
    RingbufHandle_t rb = NULL;
    rmt_item32_t *items = NULL;
    
    rmt_config_t rmt_rx_config = RMT_DEFAULT_CONFIG_RX(CONFIG_EXAMPLE_RMT_RX_GPIO, example_rx_channel);
    rmt_rx_config.mem_block_num=6;
    rmt_config(&rmt_rx_config);
    rmt_driver_install(example_rx_channel, 1000, 0);
    ir_parser_config_t ir_parser_config = IR_PARSER_DEFAULT_CONFIG((ir_dev_t)example_rx_channel);
    ir_parser_config.flags |= IR_TOOLS_FLAGS_PROTO_EXT; // Using extended IR protocols (both NEC and RC5 have extended version)
    ir_parser_t *ir_parser = NULL;

    //get RMT RX ringbuffer
    rmt_get_ringbuf_handle(example_rx_channel, &rb);
    assert(rb != NULL);
    // Start receive
    rmt_rx_start(example_rx_channel, true);
    while (1) {
        items = (rmt_item32_t *) xRingbufferReceive(rb, &length, portMAX_DELAY);
    	ESP_LOGI(TAG,"length---1---=%d",length);

        length /= 4; // one RMT = 4 Bytes
        for(int aaa=0;aaa<length-1;aaa++){
        	cmd++;
            ESP_LOGI(TAG, " level0=%d	duration0=%d	level1=%d	duration1=%d",  items->level0, items->duration0 ,items->level1,items->duration1);
            items++;
        }
        ESP_LOGI(TAG, " cmd---=%d"	,  cmd);

        length=0;
        vRingbufferReturnItem(rb, (void *) items);
		 vTaskDelay(200 / portTICK_RATE_MS);
    }
    ir_parser->del(ir_parser);
    rmt_driver_uninstall(example_rx_channel);
    vTaskDelete(NULL);
}

void app_main(void)
{
    xTaskCreate(example_ir_rx_task, "ir_rx_task", 2048, NULL, 10, NULL);
[... 465 lines cut ...]
#define CONFIG_SYSTEM_EVENT_TASK_STACK_SIZE CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE
#define CONFIG_TASK_WDT CONFIG_ESP_TASK_WDT
#define CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU0 CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU0
#define CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU1 CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU1
#define CONFIG_TASK_WDT_TIMEOUT_S CONFIG_ESP_TASK_WDT_TIMEOUT_S
#define CONFIG_TCPIP_RECVMBOX_SIZE CONFIG_LWIP_TCPIP_RECVMBOX_SIZE
#define CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY CONFIG_LWIP_TCPIP_TASK_AFFINITY_NO_AFFINITY
#define CONFIG_TCPIP_TASK_STACK_SIZE CONFIG_LWIP_TCPIP_TASK_STACK_SIZE
#define CONFIG_TCP_MAXRTX CONFIG_LWIP_TCP_MAXRTX
#define CONFIG_TCP_MSL CONFIG_LWIP_TCP_MSL
#define CONFIG_TCP_MSS CONFIG_LWIP_TCP_MSS
#define CONFIG_TCP_OVERSIZE_MSS CONFIG_LWIP_TCP_OVERSIZE_MSS
#define CONFIG_TCP_QUEUE_OOSEQ CONFIG_LWIP_TCP_QUEUE_OOSEQ
#define CONFIG_TCP_RECVMBOX_SIZE CONFIG_LWIP_TCP_RECVMBOX_SIZE
#define CONFIG_TCP_SND_BUF_DEFAULT CONFIG_LWIP_TCP_SND_BUF_DEFAULT
#define CONFIG_TCP_SYNMAXRTX CONFIG_LWIP_TCP_SYNMAXRTX
#define CONFIG_TCP_WND_DEFAULT CONFIG_LWIP_TCP_WND_DEFAULT
#define CONFIG_TIMER_QUEUE_LENGTH CONFIG_FREERTOS_TIMER_QUEUE_LENGTH
#define CONFIG_TIMER_TASK_PRIORITY CONFIG_FREERTOS_TIMER_TASK_PRIORITY
#define CONFIG_TIMER_TASK_STACK_DEPTH CONFIG_FREERTOS_TIMER_TASK_STACK_DEPTH
#define CONFIG_TIMER_TASK_STACK_SIZE CONFIG_ESP_TIMER_TASK_STACK_SIZE
#define CONFIG_TOOLPREFIX CONFIG_SDK_TOOLPREFIX
#define CONFIG_UDP_RECVMBOX_SIZE CONFIG_LWIP_UDP_RECVMBOX_SIZE

/********************sdkconfig***************end****************/

hechaotudou, 2022-08-18:
@Alvin1Zhang 
大佬，提问者是菜鸟，还望海涵
## Environment
esp32-s(nodemcu)
release-v4.4

执行 /usr/bin/python3 /home/hctd/user/esp32/install/eclipse-workspace/esp-idf/tools/idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* xtensa-clang: LLVM for Xtensa (ESP32, ESP32-S2) based on clang (optional)
  - 12.0.1-d9341b81fc (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 and ESP32-S3 ULP coprocessors
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.20.3 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.11.0-esp32-20211220 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.10.2 (recommended, installed)

CDT Build environment variables
PATH: /root/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin:/root/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/bin:/root/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin:/root/.espressif/tools/xtensa-clang/12.0.1-d9341b81fc/xtensa-esp32-elf-clang/bin:/root/.espressif/tools/riscv32-esp-elf/esp-2021r2-8.4.0/riscv32-esp-elf/bin:/root/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/root/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/root/.espressif/tools/cmake/3.20.3/bin:/root/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin:/root/.espressif/tools/ninja/1.10.2/:/root/.espressif/python_env/idf4.4_py3.10_env/bin:/home/hctd/user/esp32/install/eclipse-workspace/esp-idf/tools:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin
XAUTHORITY: /run/user/1000/.mutter-Xwaylandauth.LRXPQ1
LC_MEASUREMENT: zh_CN.UTF-8
LC_MONETARY: zh_CN.UTF-8
OXYGEN_DISABLE_INNER_SHADOWS_HACK: 1
LC_TELEPHONE: zh_CN.UTF-8
LC_TIME: zh_CN.UTF-8
TERM: xterm-256color
LANG: zh_CN.UTF-8
SUDO_USER: hctd
OPENOCD_SCRIPTS: /root/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts
SUDO_GID: 1000
DISPLAY: :0
MAIL: /var/mail/root
COLORTERM: truecolor
LC_PAPER: zh_CN.UTF-8
LOGNAME: root
SUDO_UID: 1000
LC_NAME: zh_CN.UTF-8
LANGUAGE: zh_CN:zh:en_US:en
SHELL: /bin/bash
LC_ADDRESS: zh_CN.UTF-8
LIBOVERLAY_SCROLLBAR: 0
IDF_PATH: /xxx/xxx/xxx/xxx/xxx/eclipse-workspace/esp-idf
GDK_CORE_DEVICE_EVENTS: 1
USER: root
SUDO_COMMAND: user/esp32/install/eclipse/eclipse
LC_NUMERIC: zh_CN.UTF-8
IDF_COMPONENT_MANAGER: 1
LS_COLORS: rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36
IDF_PYTHON_ENV_PATH: /root/.espressif/python_env/idf4.4_py3.10_env
HOME: /root
LC_IDENTIFICATION: zh_CN.UTF-8

操作系统： linux
Java 运行时间版本： 16.0.2+7-67
Eclipse 版本： 4.21.0.v20210906-0500
Eclipse CDT 版本： 10.4.0.202106130052
IDF Eclipse 插件版本： 2.6.0.202208110735
ESP-IDF v4.4
为 IDF_PYTHON_ENV 指定的 Python： Python 3.10.4
[... 721 lines cut ...]
#define CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT CONFIG_VFS_SUPPRESS_SELECT_DEBUG_OUTPUT
#define CONFIG_SYSTEM_EVENT_QUEUE_SIZE CONFIG_ESP_SYSTEM_EVENT_QUEUE_SIZE
#define CONFIG_SYSTEM_EVENT_TASK_STACK_SIZE CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE
#define CONFIG_TASK_WDT CONFIG_ESP_TASK_WDT
#define CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU0 CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU0
#define CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU1 CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU1
#define CONFIG_TASK_WDT_TIMEOUT_S CONFIG_ESP_TASK_WDT_TIMEOUT_S
#define CONFIG_TCPIP_RECVMBOX_SIZE CONFIG_LWIP_TCPIP_RECVMBOX_SIZE
#define CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY CONFIG_LWIP_TCPIP_TASK_AFFINITY_NO_AFFINITY
[... cut ...]
#define CONFIG_TCP_SYNMAXRTX CONFIG_LWIP_TCP_SYNMAXRTX
#define CONFIG_TCP_WND_DEFAULT CONFIG_LWIP_TCP_WND_DEFAULT
#define CONFIG_TIMER_QUEUE_LENGTH CONFIG_FREERTOS_TIMER_QUEUE_LENGTH
#define CONFIG_TIMER_TASK_PRIORITY CONFIG_FREERTOS_TIMER_TASK_PRIORITY
#define CONFIG_TIMER_TASK_STACK_DEPTH CONFIG_FREERTOS_TIMER_TASK_STACK_DEPTH
#define CONFIG_TIMER_TASK_STACK_SIZE CONFIG_ESP_TIMER_TASK_STACK_SIZE
#define CONFIG_TOOLPREFIX CONFIG_SDK_TOOLPREFIX
#define CONFIG_UDP_RECVMBOX_SIZE CONFIG_LWIP_UDP_RECVMBOX_SIZE

hechaotudou, 2022-08-18:
> @hechaotudou你能分享你的RMT部分的初始化代码吗？注意，ESP32 的硬件无法接收长度大于块大小的 RMT 帧。您可以尝试在您的情况下增加块大小和环形缓冲区大小吗？
> 
> 仅供参考，在 5.0 分支上，我们添加了一个新的 RMT 驱动程序，它可以以异步方式处理传入的数据包，请参见示例：[https ://github.com/espressif/esp-idf/blob/master/examples/外围设备/rmt/ir_nec_transceiver/main/ir_nec_transceiver_main.c](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/rmt/ir_nec_transceiver/main/ir_nec_transceiver_main.c)

  rmt_config_t rmt_rx_config = RMT_DEFAULT_CONFIG_RX(CONFIG_EXAMPLE_RMT_RX_GPIO, example_rx_channel);
    rmt_rx_config.mem_block_num=6;
    rmt_config(&rmt_rx_config);
    rmt_driver_install(example_rx_channel, 1000, 0);
    ir_parser_config_t ir_parser_config = IR_PARSER_DEFAULT_CONFIG((ir_dev_t)example_rx_channel);
    ir_parser_config.flags |= IR_TOOLS_FLAGS_PROTO_EXT; // Using extended IR protocols (both NEC and RC5 have extended version)
    ir_parser_t *ir_parser = NULL;

hechaotudou, 2022-08-18:
@suda-morris 
  rmt_config_t rmt_rx_config = RMT_DEFAULT_CONFIG_RX(CONFIG_EXAMPLE_RMT_RX_GPIO, example_rx_channel);
    rmt_rx_config.mem_block_num=6;
    rmt_config(&rmt_rx_config);
    rmt_driver_install(example_rx_channel, 1000, 0);
    ir_parser_config_t ir_parser_config = IR_PARSER_DEFAULT_CONFIG((ir_dev_t)example_rx_channel);
    ir_parser_config.flags |= IR_TOOLS_FLAGS_PROTO_EXT; // Using extended IR protocols (both NEC and RC5 have extended version)
    ir_parser_t *ir_parser = NULL;

suda-morris, 2022-08-24:
Hi @hechaotudou Your code crashed at `vRingbufferReturnItem(rb, (void *) items);` because the `item` address has changed when you print the items. It's not the address that return from `xRingbufferReceive`.

closed 2022-08-25 by suda-morris