espressif/esp-idf #9580: A fatal error occurred: Failed to connect to ESP32: Invalid head of packet (0x08)

pooja079, 2022-08-18:
Hi
Please help someone I'm new to this, not able to flash the project:

C:\Users\WIN10\esp\hello_world>idf.py -p COM7 flash
Executing action: flash
Running ninja in directory c:\users\win10\esp\hello_world\build
Executing "ninja flash"...
[1/5] cmd.exe /C "cd /D C:\Users\WIN10\esp\hello_world\bui....bin C:/Users/WIN10/esp/hello_world/build/hello_world.bin"
hello_world.bin binary size 0x29730 bytes. Smallest app partition is 0x100000 bytes. 0xd68d0 bytes (84%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Users\WIN10\esp\hello_world\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v4.4/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/Users/WIN10/esp/hello_world/build/bootloader/bootloader.bin"
Bootloader binary size 0x62b0 bytes. 0xd50 bytes (12%) free.
[2/3] cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v4.../esp-idf-v4.4/components/esptool_py/run_serial_tool.cmake"
esptool.py esp32 -p COM7 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x1000 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin
esptool.py v3.2-dev
Serial port COM7
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: Invalid head of packet (0x08): Possible serial noise or corruption.
For troubleshooting steps visit: https://github.com/espressif/esptool#troubleshooting
CMake Error at run_serial_tool.cmake:56 (message):
  C:/Espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe
  C:/Espressif/frameworks/esp-idf-v4.4/components/esptool_py/esptool/esptool.py
  --chip esp32 failed

FAILED: CMakeFiles/flash
cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v4.4\components\esptool_py && C:\Espressif\tools\cmake\3.20.3\bin\cmake.exe -D IDF_PATH="C:/Espressif/frameworks/esp-idf-v4.4" -D SERIAL_TOOL="C:/Espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe C:/Espressif/frameworks/esp-idf-v4.4/components/esptool_py/esptool/esptool.py --chip esp32" -D SERIAL_TOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="C:/Users/WIN10/esp/hello_world/build" -P C:/Espressif/frameworks/esp-idf-v4.4/components/esptool_py/run_serial_tool.cmake"
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

C:\Users\WIN10\esp\hello_world>

radimkarnis, 2022-08-18:
Hello @pooja079,

the most probable cause of this is that your devkit is not [automatically switching into the download mode](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#automatic-bootloader). This can be verified by using a newer version of [esptool](https://github.com/espressif/esptool) - wrong boot mode detection has been implemented in v3.2. You can download the latest esptool from the repository or from [PyPI](https://pypi.org/project/esptool/), or you can update ESP-IDF.

If your board has boot and reset buttons, you can try [entering the download mode manually](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#manual-bootloader).

Please let me know if this helps. If not, we can try some more troubleshooting.

pooja079, 2022-08-18:
Thank You @radimkarnis ,

Let me try and get back to you

pooja079, 2022-08-18:
Hi,
I'm using
ESP-IDF V4.4
ESP32 LyraT V4.3

radimkarnis, 2022-08-18:
@pooja079, sorry for not explaining.

ESP-IDF uses a utility called [esptool.py](https://github.com/espressif/esptool) under the hood to flash the compiled app binaries to the chip. You can see esptool runs and its output in your log, it is this part:

```
esptool.py v3.2-dev
Serial port COM7
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: Invalid head of packet (0x08): Possible serial noise or corruption.
```

Esptool connects to the chip and if successful, sends the binary over. In order for it to connect, the ESP chip has to be switched into a [download mode](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/entering-bootloader.html). This switch usually happens [automatically](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#automatic-bootloader), but if that for whatever reason fails, you can also switch it [manually](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#manual-bootloader).

> Press the reset button while holding the boot button before flashing, or just hold the boot button while attempting to flash

My advice to you here is to try switching into download mode manually.

If this doesn't help, you can try updating ESP-IDF to a newer version, which will also update esptool.py to a newer version. Updated esptool would either solve the problem or give us more information about the problem.

radimkarnis, 2022-08-18:
I can see you are using `ESP32 LyraT V4.3`. This board requires some jumpers to be installed for automatic upload to work. Please see the details [here](https://espressif-docs.readthedocs-hosted.com/projects/esp-adf/en/latest/design-guide/dev-boards/board-esp32-lyrat-v4.3.html#using-automatic-upload). Also check the schematic to see where the jumpers and `reset`/`boot` buttons are.

radimkarnis, 2022-08-23:
@pooja079 did you manage to flash your device?

JegadeeshXaglabs, 2022-08-29:
Don't boot the ESP32 too many times.

pooja079, 2022-09-16:
> @pooja079 did you manage to flash your device?

Yes, by using jumpers
Thanks

closed 2022-09-22 by dobairoland