espressif/esp-idf #9581: esp_https_ota API might be calling custom http_client_init_cb after performing the request instead of before doing so.

rommo911, 2022-08-18:
## Environment
- Development Kit: ESP32-DevKitCe
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROVER-I
- IDF version (run ``git describe --tags`` to find it): v4.4.1
- Build System: idf.py
- Compiler version (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: [Yes (please give details) : VsCode using IDF extension 
- Power Supply:  usb external 5V

## Problem Description

In the **_esp_https_ota_config_t_** there is a section for adding an **_http_client_init_cb_** to add custom headers to the http request.
i notices it did not work for me to add my custom authorization token when asking for a firmware file from my server 
the i checked **esp_https_ota.c** and found out that the calling of this callback (**http_client_init_cb** ) is done after `esp_http_client_perform()` , this means the custom headers are added after the http is perfomred. is this normal ?  i moved the line of the `http_client_init_cb `calling right befor the `esp_http_client_perform()` and after `esp_http_client_init()` and things worked fine for me.

### Expected Behavior

calling the  `http_client_init_cb ` before ` esp_http_client_perform()` in the   **esp_https_ota.c**
 
### Actual Behavior

 the  `http_client_init_cb ` called after ` esp_http_client_perform()` in the   **esp_https_ota.c**

### Steps to reproduce

step 1: typical use of esp_https_ota api with adding (**http_client_init_cb** )  to the handle.

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

static esp_err_t AddCutomHeaders(esp_http_client_handle_t http_client)
{
    esp_err_t err = ESP_OK;
    err = esp_http_client_set_header(http_client, "Authorization", "Token ***********************************");
    return err;
}
void app_main()
{
    std::string fname = "https://myserverThatUsesCustomHeadersAuthorization/ota.bin";
    esp_http_client_config_t http_ota{};
    memset(&http_ota, 0, sizeof(http_ota));
    http_ota.url = fname.c_str();
    esp_https_ota_config_t ota_config = {};
    ota_config.http_config = &http_ota;
    ota_config.http_client_init_cb = AddCutomHeaders; 
}

```

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/9374966/sdkconfig.txt)

hmalpani, 2022-08-22:
Hello @rommo911 ,
Thanks for reporting the issue. Attaching a patch here. Can you verify if it solves the problem? Thanks!
[IDFGH-8080.txt](https://github.com/espressif/esp-idf/files/9391671/IDFGH-8080.txt)

rommo911, 2022-08-22:
hello @hmalpani , 
Thanks for your reply. yes this resolve the issue.
Thanks!
is it going to be pushed to branch or a tag ? i may have to use it with an IDF docker image ...

hmalpani, 2022-08-22:
@rommo911  Are you also using partial download feature with your example?

rommo911, 2022-08-22:
@hmalpani  nope not at the moment.

hmalpani, 2022-08-24:
@rommo911  Can you verify once if partial download is enabled or not? I tested with both the cases and header was not added only when partial download was enabled.

rommo911, 2022-08-24:
@hmalpani  actually sorry i was mistaken , the partial download was active indeed in my code. i will test again later today to disable it.

closed 2022-09-05 by espressif-bot