espressif/esp-idf #9585: unity support for esp32s3?

chmorgan, 2022-08-18:
It looks like unity port doesn't support esp32s3 yet?

## Environment

- Development Kit:      [esp-box]
- Module or chip used:  ESP32-S3
- IDF version (run ``git describe --tags`` to find it):
    v4.4.1
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:     macOS
- Using an IDE?: no (usually vscode but not for this getting started testing)
- Power Supply:         USB

## Problem Description

Using unit-test-app to test a component I'm working on.

I get the expected output on the terminal after programming via:

```
idf.py flash monitor
```

but it prompts to press enter and won't continue:
```
Press ENTER to see the list of tests.
```

I think this is because unity_port_esp32.c is expecting data to be delivered a particular uart buffer but this isn't the case for esp32s3 on the esp-box where the console is mapped to USB/JTAG via ```CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y```

```
void unity_gets(char *dst, size_t len)
{
    size_t unity_input_from_gdb_len = strlen(unity_input_from_gdb);
    if (unity_input_from_gdb_len > 0 && unity_input_from_gdb_len < len - 1) {
        memcpy(dst, unity_input_from_gdb, unity_input_from_gdb_len);
        dst[unity_input_from_gdb_len] = '\n';
        dst[unity_input_from_gdb_len + 1] = 0;
        memset(unity_input_from_gdb, 0, sizeof(unity_input_from_gdb));
        return;
    }
    /* esp_rom_uart_rx_string length argument is uint8_t */
    if (len >= UINT8_MAX) {
        len = UINT8_MAX;
    }
    /* Flush anything already in the RX buffer */
    uint8_t ignore;
    while (esp_rom_uart_rx_one_char(&ignore) == 0) {  <-----------------------
    }
    /* Read input */
    esp_rom_uart_rx_string((uint8_t *) dst, len);
}
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

igrr, 2022-08-19:
Hi @chmorgan,
Could you please attach your sdkconfig file? I think the input should work fine over the USB_SERIAL_JTAG with `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y`. 

By default the settings are CONFIG_ESP_CONSOLE_SECONDARY_USB_SERIAL_JTAG=y and CONFIG_ESP_CONSOLE_UART_DEFAULT=y, in which case input via USB_SERIAL_JTAG indeed doesn't work, only output does.

chmorgan, 2022-08-19:
@igrr well damnit all. Installed an osx update which caused a reboot, re export.sh, re-ran menuconfig and re-built and now its working. Perhaps I set secondary usb and then the menuconfig rules fixed the configuration when I re-selected usb/tag for the primary console?

I apologize and now without the baseline sdkconfig I can't prove the theory above. Sorry for taking any of your time. Let me close this out.

closed 2022-08-19 by chmorgan