espressif/esp-idf #9586: Accessing flash very slow on modules with Octal flash ESP32-S3-WROOM-2-N32R8V

ThinkTransit, 2022-08-19:
## Environment

- Module or chip used:  ESP32-S3-WROOM-2-N32R8V
- IDF version (run ``git describe --tags`` to find it): v4.4.2-138-g4c550fd79f
- Build System:       idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32s3-elf-gcc (crosstool-NG esp-2022r1) 11.2.0

- Operating System:     Linux
- Using an IDE?: PyCharm
- Power Supply:         external 3.3V

## Problem Description

The ESP32-S3-WROOM-2-N32R8V has 32MB of Octal flash.
When Octal flash is enabled via menuconfig, accessing the flash is very slow. I assume the following message on boot indicates that the flash is operating in some form of "slow" mode.

```

Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
```

```

#
# Serial flasher config
#
CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=115200
# CONFIG_ESPTOOLPY_NO_STUB is not set
CONFIG_ESPTOOLPY_OCT_FLASH=y
CONFIG_ESPTOOLPY_FLASHMODE_OPI=y
CONFIG_ESPTOOLPY_FLASH_SAMPLE_MODE_STR=y
# CONFIG_ESPTOOLPY_FLASH_SAMPLE_MODE_DTR is not set
CONFIG_ESPTOOLPY_FLASHMODE="dout"
# CONFIG_ESPTOOLPY_FLASHFREQ_120M is not set
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y
# CONFIG_ESPTOOLPY_FLASHFREQ_40M is not set
# CONFIG_ESPTOOLPY_FLASHFREQ_20M is not set
CONFIG_ESPTOOLPY_FLASHFREQ="80m"
# CONFIG_ESPTOOLPY_FLASHSIZE_1MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_2MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_4MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_8MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_16MB is not set
CONFIG_ESPTOOLPY_FLASHSIZE_32MB=y
# CONFIG_ESPTOOLPY_FLASHSIZE_64MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_128MB is not set
CONFIG_ESPTOOLPY_FLASHSIZE="32MB"
CONFIG_ESPTOOLPY_FLASHSIZE_DETECT=y
CONFIG_ESPTOOLPY_BEFORE_RESET=y
# CONFIG_ESPTOOLPY_BEFORE_NORESET is not set
CONFIG_ESPTOOLPY_BEFORE="default_reset"
# CONFIG_ESPTOOLPY_AFTER_RESET is not set
CONFIG_ESPTOOLPY_AFTER_NORESET=y
CONFIG_ESPTOOLPY_AFTER="no_reset"
# CONFIG_ESPTOOLPY_MONITOR_BAUD_CONSOLE is not set
# CONFIG_ESPTOOLPY_MONITOR_BAUD_9600B is not set
# CONFIG_ESPTOOLPY_MONITOR_BAUD_57600B is not set
CONFIG_ESPTOOLPY_MONITOR_BAUD_115200B=y
# CONFIG_ESPTOOLPY_MONITOR_BAUD_230400B is not set
# CONFIG_ESPTOOLPY_MONITOR_BAUD_921600B is not set
# CONFIG_ESPTOOLPY_MONITOR_BAUD_2MB is not set
# CONFIG_ESPTOOLPY_MONITOR_BAUD_OTHER is not set
CONFIG_ESPTOOLPY_MONITOR_BAUD_OTHER_VAL=115200
CONFIG_ESPTOOLPY_MONITOR_BAUD=115200
# end of Serial flasher config

```

### Expected Behavior
Flash to operate at normal speed

### Actual Behavior
Access to flash is 4-5 times slower than other ESP modules using quad flash

### Steps to reproduce

1. Enable Octal flash in menuconfig
2. Build and flash to module
3. Read/write data from flash

## Other items if possible
[image]

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/9378305/sdkconfig.txt)

JeremiahGillis, 2022-11-22:
I am on ESP-IDF v4.4.3. What concerns me is the following:

```
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce3808,len:0x3ac
load:0x403c9700,len:0x99c
load:0x403cc700,len:0x265c
entry 0x403c98b8
W (788) FLASH: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
```

I get a warning to not use for mass production. I have designed this module into a product which will be entering production soon. I currently have about 1000 of these modules. Is this `SLOW_RD` just for the bootloader or what?

When is this planned to be addressed since you already sell production modules?

higaski, 2023-11-22:
Just to  clarify... does this log mean that octal settings work but that the bootloader is using something else?

```sh
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x28 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce3810,len:0x17bc
load:0x403c9700,len:0x4
load:0x403c9704,len:0xdc4
load:0x403cc700,len:0x2e1c
entry 0x403c9934
I (33) boot: ESP-IDF v5.3-dev-277-gc8243465e4 2nd stage bootloader
I (33) boot: compile time Nov 22 2023 16:39:26
I (34) boot: Multicore bootloader
I (38) boot: chip revision: v0.1
I (42) boot.esp32s3: Boot SPI Speed : 80MHz
I (47) boot.esp32s3: SPI Mode       : SLOW READ
I (52) boot.esp32s3: SPI Flash Size : 32MB
...
```

Is this a workaround to some kind of bug or something?