espressif/esp-idf #9587: esp-idf version 5.1 - option to "Skip server certificate verification by default" not working

hoangdovan, 2022-08-19:
## Problem Description

With esp-idf version 4.4 or below, for testing or for another purpose, when making https request, if we want to skip verify server certificate, then we can select 2 options in menuconfig ESP-TLS:
  **1. Allow potentially insecure options
  2. Skip server certificate verification by default**

But with latest version 5.1, these options not working anymore, Https request not working, return error!

### Expected Behavior
esp-idf 5.1 should allow making https request with option to skip server certificate same as version 4.4 or below.

laukik-hase, 2022-08-19:
Hello, @hoangdovan! 

Thank you for the issue report. I was investigating this issue but unfortunately could not reproduce it with the `https_request` example on the `v5.1-dev` tag. Could you please provide the following info -

- Target chip or module (`ESP32`, `ESP32-S2`, `ESP32-S3`, etc.)
- ESP-IDF version (run `git describe --tags`) / commit ID
- Example to reproduce the issue (or a minimal code snippet)
- `sdkconfig` file
- Debug logs

hoangdovan, 2022-08-19:
@laukik-hase Thank for response! Here are some informations you requested:

1. Target chip: ESP32-Wrover-Kit
2. ESP-IDF version: 5.1 (Master Branch)
3. Minimal code snippet - I cannot share all code base, but maybe some important related code like below:

esp_http_client_config_t config = {
        .url = url,
        .method = HTTP_GET,
        .user_data = &custom_data,
        .event_handler = _http_event_handle,
        .buffer_size = 1024,
        .timeout_ms = 5000
        // .cert_pem = test_cert
    };

esp_err_t err = esp_http_client_perform(client);

(Nothing special about _http_event_handle or custom_data).
Please note that comment out  "// .cert_pem = test_cert".
This code working well with esp-idf version 4.4 or below.

4. sdkconfig

#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_SOC_BROWNOUT_RESET_SUPPORTED="Not determined"
CONFIG_SOC_TWAI_BRP_DIV_SUPPORTED="Not determined"
CONFIG_SOC_DPORT_WORKAROUND="Not determined"
CONFIG_SOC_CAPS_ECO_VER_MAX=3
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_DAC_SUPPORTED=y
CONFIG_SOC_MCPWM_SUPPORTED=y
CONFIG_SOC_SDMMC_HOST_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_PCNT_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_SDIO_SLAVE_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_EMAC_SUPPORTED=y
CONFIG_SOC_ULP_SUPPORTED=y
CONFIG_SOC_CCOMP_TIMER_SUPPORTED=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_SLOW_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
CONFIG_SOC_RMT_SUPPORTED=y
CONFIG_SOC_SDM_SUPPORTED=y
CONFIG_SOC_SUPPORT_COEXISTENCE=y
CONFIG_SOC_AES_SUPPORTED=y
CONFIG_SOC_MPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_TOUCH_SENSOR_SUPPORTED=y
CONFIG_SOC_DPORT_WORKAROUND_DIS_INTERRUPT_LVL=5
CONFIG_SOC_XTAL_SUPPORT_26M=y
CONFIG_SOC_XTAL_SUPPORT_40M=y
CONFIG_SOC_XTAL_SUPPORT_AUTO_DETECT=y
CONFIG_SOC_ADC_RTC_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DIG_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DMA_SUPPORTED=y
CONFIG_SOC_ADC_PERIPH_NUM=2
CONFIG_SOC_ADC_MAX_CHANNEL_NUM=10
CONFIG_SOC_ADC_ATTEN_NUM=4
CONFIG_SOC_ADC_DIGI_CONTROLLER_NUM=2
CONFIG_SOC_ADC_PATT_LEN_MAX=16
CONFIG_SOC_ADC_DIGI_MIN_BITWIDTH=9
[... 1916 lines cut ...]
I (23351) mbedtls: ssl_msg.c:2369 => write record

I (23351) mbedtls: ssl_msg.c:1789 => flush output

I (23361) mbedtls: ssl_msg.c:1807 message length: 7, out_left: 7

I (28691) mbedtls: ssl_msg.c:1814 ssl->f_send() returned -26752 (-0x6880)

W (28691) mbedtls: ssl_msg.c:2510 mbedtls_ssl_flush_output() returned -26752 (-0x6880)

W (28691) mbedtls: ssl_msg.c:4573 mbedtls_ssl_write_record() returned -26752 (-0x6880)

W (28701) mbedtls: ssl_tls.c:1969  mbedtls_x509_crt_parse_der() returned -10368 (-0x2880)

I (28711) mbedtls: ssl_tls.c:5323 <= handshake

E (28711) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2880
E (28721) esp-tls: Failed to open new connection
E (28731) transport_base: Failed to open a new connection
E (28741) HTTP_CLIENT: Connection failed, sock < 0
I (28741) mos_http_client: HTTP_EVENT_ERROR
I (28741) mos_http_client: esp_http_client_perform() failed.
W (28751) mos_http_client: status_code = 0
I (28751) mos_http_client: HTTP_EVENT_DISCONNECTED
W (28761) mos_http_client: mos_http_request() return err = -1

hoangdovan, 2022-08-19:
@laukik-hase Sorry for messy sdkconfig content display!
And there some more information: my code working perfectly with esp-idf version 4.4 or below. With esp-idf 5.1 it failed when request https. I doubt it's related to mbedtls config problem when using built-in function esp_http_client_perform() in version 5.1, may be option to skip verify server certificate not working in this case? As you can see in debug log, mbedtls_ssl_handshake failed with returned -0x2880. Thank you!

hoangdovan, 2022-08-20:
@laukik-hase to compare with esp-idf version 4.4 here are same debug log, maybe it can be helpful for your investigation:

I (497358) wifi_com: Got IPv6 event: Interface "wifi_com: sta" address: fe80:0000:0000:0000:ae67:b2ff:fe71:62c4, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (592408) cli_analyzer: cli_command_analyze parameters [0] key = 'u', value = 'https://gorest.co.in/public/v2/posts'
I (592408) request_handler: Found target command_type by default!
I (592408) request_handler: Found target request handler!
W (592418) request_handler: execute_request() cmd_data.command_type = 1!
I (592428) request_handler: time = 591857131, collection id = 2, execute_command() at './components/request_handler/request_handler.c', 282: 'Running command = 'http'
'
W (592438) http_functions: response_option != RAW_DATA
I (592448) mos_http_client: mos_http_request() target url: 'https://gorest.co.in/public/v2/posts'
I (592598) mbedtls: ssl_tls.c:5904 => handshake

I (592598) mbedtls: ssl_cli.c:4483 client state: 0

I (592598) mbedtls: ssl_msg.c:2102 => flush output

I (592608) mbedtls: ssl_msg.c:2114 <= flush output

I (592608) mbedtls: ssl_cli.c:4483 client state: 1

I (592618) mbedtls: ssl_msg.c:2102 => flush output

I (592628) mbedtls: ssl_msg.c:2114 <= flush output

I (592628) mbedtls: ssl_cli.c:999 => write client hello

I (592648) mbedtls: ssl_msg.c:2542 => write handshake message

I (592648) mbedtls: ssl_msg.c:2701 => write record

I (592648) mbedtls: ssl_msg.c:2102 => flush output

I (592658) mbedtls: ssl_msg.c:2122 message length: 243, out_left: 243

I (592658) mbedtls: ssl_msg.c:2127 ssl->f_send() returned 243 (-0xffffff0d)

I (592668) mbedtls: ssl_msg.c:2155 <= flush output

I (592678) mbedtls: ssl_msg.c:2870 <= write record

I (592678) mbedtls: ssl_msg.c:2678 <= write handshake message

I (592688) mbedtls: ssl_cli.c:1467 <= write client hello

I (592698) mbedtls: ssl_cli.c:4483 client state: 2

I (592698) mbedtls: ssl_msg.c:2102 => flush output

I (592708) mbedtls: ssl_msg.c:2114 <= flush output

I (592708) mbedtls: ssl_cli.c:2082 => parse server hello

I (592718) mbedtls: ssl_msg.c:3941 => read record

I (592718) mbedtls: ssl_msg.c:1886 => fetch input

I (592728) mbedtls: ssl_msg.c:2043 in_left: 0, nb_want: 5

I (592798) mbedtls: ssl_msg.c:2068 in_left: 0, nb_want: 5

I (592798) mbedtls: ssl_msg.c:2069 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)

I (592798) mbedtls: ssl_msg.c:2089 <= fetch input

I (592808) mbedtls: ssl_msg.c:1886 => fetch input

I (592818) mbedtls: ssl_msg.c:2043 in_left: 5, nb_want: 72

I (592818) mbedtls: ssl_msg.c:2068 in_left: 5, nb_want: 72
[... 636 lines cut ...]
I (596518) mbedtls: ssl_msg.c:2069 ssl->f_recv(_timeout)() returned 92 (-0xffffffa4)

I (596528) mbedtls: ssl_msg.c:2089 <= fetch input

I (596538) mbedtls: ssl_msg.c:1144 => decrypt buf

I (596538) mbedtls: ssl_msg.c:1741 <= decrypt buf

I (596548) mbedtls: ssl_msg.c:4015 <= read record

I (596548) mbedtls: ssl_msg.c:5693 <= read

I (596558) mbedtls: ssl_msg.c:5399 => read

I (596558) mbedtls: ssl_msg.c:3941 => read record

I (596568) mbedtls: ssl_msg.c:1886 => fetch input

I (596568) mbedtls: ssl_msg.c:2043 in_left: 0, nb_want: 5

I (596578) mbedtls: ssl_msg.c:2068 in_left: 0, nb_want: 5

************************************************************
So there is no problem with mbedtls with skip server certificate verification option, Https request can be done successfully. Somehow it got trouble with version 5.1.
Thank you!

laukik-hase, 2022-08-22:
Hello, @hoangdovan!

Thank you for the info. Unfortunately, I am still unable to reproduce the issue with the `esp_http_client` and `https_request` examples with the `sdkconfig` you provided. 

<details>
  <summary> Code snippet </summary>

```
static void https_with_url(void)
{
    char local_response_buffer[MAX_HTTP_OUTPUT_BUFFER] = {0};

    esp_http_client_config_t config = {
        .url = "https://gorest.co.in/public/v2/posts",
        .event_handler = _http_event_handler,
        .method = HTTP_METHOD_GET,
        .user_data = local_response_buffer,
        .buffer_size = 1024,
        .timeout_ms = 5000
    };
    esp_http_client_handle_t client = esp_http_client_init(&config);
    esp_err_t err = esp_http_client_perform(client);

    if (err == ESP_OK) {
        ESP_LOGI(TAG, "HTTPS Status = %d, content_length = %lld",
                esp_http_client_get_status_code(client),
                esp_http_client_get_content_length(client));
    } else {
        ESP_LOGE(TAG, "Error perform http request %s", esp_err_to_name(err));
    }
    esp_http_client_cleanup(client);
}
```

</details>

- Test config and logs: [tls_no_server_verify_bug.zip](https://github.com/espressif/esp-idf/files/9391970/tls_no_server_verify_bug.zip)
- Could you attach the debug logs with the following config options enabled - `CONFIG_LOG_DEFAULT_LEVEL_VERBOSE=y`, `CONFIG_MBEDTLS_DEBUG=y` and `CONFIG_MBEDTLS_DEBUG_LEVEL_VERBOSE=y`
- Mbed TLS returned [`-0x2880`](https://github.com/espressif/mbedtls/blob/9bb5effc3298265f829878825d9bd38478e67514/include/mbedtls/x509.h#L89) which indicates a failure to allocate memory. Could you please track your heap memory using the APIs [`esp_get_free_heap_size()`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/misc_system_api.html#_CPPv422esp_get_free_heap_sizev) and [`esp_get_minimum_free_heap_size()`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/misc_system_api.html#_CPPv430esp_get_minimum_free_heap_sizev) just before and after the `esp_http_client_perform()` call.

hoangdovan, 2022-08-22:
@laukik-hase Thank you! I think with your help, I already found the root cause of problem! As your suggestion, Mbed TLS returned [-0x2880] which indicates a failure to allocate memory. The problem here: my program nearly running out of heap memory with version 5.1, but not with version 4.4 or below!

***
**With version 4.4 logging like this:**
Before esp_http_client_perform() - free_heap = 43868, min_free_heap = 38448
After esp_http_client_perform() - free_heap = 9164, min_free_heap = 2008

**With version 5.1 logging result:**
After esp_http_client_perform() - free_heap = 32048, min_free_heap = 732

**The reason another part of my code require large heap memory. After adjustment to reduce heap memory, then run again with esp-idf 5.1:**

Before esp_http_client_perform() - free_heap = 57388, min_free_heap = 52128
After esp_http_client_perform() - free_heap = 23180, min_free_heap = 16116

**And now I can make HTTPS request successfully!**

One thing I don't know why esp-idf 5.1 require bigger heap memory when request https than version 4.4! And I cannot find any document from Espressif about this matter! It's better if there is document about this information to help developer avoid of this problem.
Thank you!

closed 2022-08-22 by hoangdovan

lekshith, 2024-03-21:
Hi there, 
I have this exact same issue, can you please explain how to free up some heap memory?