espressif/esp-idf #9592: ESP-NOW example does not work with any other channel than 0 and 1

PilnyTomas, 2022-08-19:
## Environment

- Development Kit:     ESP32-C3-DevKitC-02
- Kit version (for WroverKit/PicoKit/DevKitC): v1.1
- Module or chip used:  ESP32-C3-WROOM-02
- IDF version : v5.1-dev-224-g4d14c2ef2d
- Build System:         idf.py
- Compiler version : 11.2.0
- Operating System:     Linux - Fedora
- Using an IDE?: No
- Power Supply:  USB

## Problem Description

ESP-NOW example outputs error and does not work when the channel is to anything but 0 or 1

### Expected Behavior

If this is expected behavior, please update the example and its documentation to prevent confusion.
If the channel really is supposed to be changeable, then please fix the reason.

### Actual Behavior

Does not work.

### Steps to reproduce

1. Go to ESP-NOW example in IDF `examples/wifi/espnow`
2. In menuconfig go to `Example Configuration ---> Channel` and set number 2 or more
3. Flash and run
4. Observer errors:
```
E (5499) ESPNOW: Peer channel is not equal to the home channel, send fail!
E (5499) espnow_example: Send error
```

### Code to reproduce this issue
ESP-NOW example in IDF [`examples/wifi/espnow`](https://github.com/espressif/esp-idf/tree/master/examples/wifi/espnow)

Optionaly you can add extra debug output at the end of the function `example_wifi_init` - (under [line 57](https://github.com/espressif/esp-idf/blob/4d14c2ef2d9d08cd1dcbb68a8bb0d76a666e2b4b/examples/wifi/espnow/main/espnow_example_main.c#L57))
```
    uint8_t primaryChan = 0;
    wifi_second_chan_t secondChan = WIFI_SECOND_CHAN_NONE;
    esp_wifi_get_channel(&primaryChan, &secondChan);
    ESP_LOGI(TAG, "CONFIG_ESPNOW_CHANNEL=%d; WiFi is on channel %d  ", CONFIG_ESPNOW_CHANNEL, primaryChan);
```

MaxwellAlan, 2022-08-24:
Hi @PilnyTomas 

Thanks for the bug reporting. We will optimize ESP-NOW example.
You can add below line after esp_wifi_start:
**esp_wifi_set_channel(CONFIG_ESPNOW_CHANNEL,0);**

[image]

PilnyTomas, 2022-08-24:
Hi, thanks for your reply.
Will you update only the example, or are you planning to implement the ability to change a channel into ESP-NOW API?

MaxwellAlan, 2022-08-25:
Hi @PilnyTomas 
We will only update the example. ESP-NOW's channel should obey the STA/AP's channel.

closed 2022-08-25 by PilnyTomas

garykuipers, 2024-05-09:
That is incorrect, I am currently running on channel 11.

"""
        Serial.println("SETUP ESPNOW");
        WiFi.mode(WIFI_STA);
        WiFi.disconnect(true);
        esp_wifi_set_mode(WIFI_MODE_STA);
        delay(100);

        // Initialize WiFi
        wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
        esp_wifi_init(&cfg);
        esp_wifi_start();
        // Initialize ESP-NOW
        esp_err_t initResult = esp_now_init();
        if (initResult != ESP_OK) {
            Serial.printf("ESP-NOW initialization failed with error: %d\n", initResult);
            return;
        }

        // Set the channel for 2.4 GHz
        esp_err_t result = esp_wifi_set_channel(storage.channel, WIFI_SECOND_CHAN_NONE);
        if (result == ESP_OK) {
            Serial.println("Channel set successfully");
        } else {
            Serial.printf("Failed to set channel with error code: %d\n", result);
            // TODO: handle init error
        }
        Serial.println("SETUP ESPNOW: setup OK");

        // Verify the channel setup (This does not work! The communication works even though this says the channel is different)
//         uint8_t current_channel;
//         esp_wifi_get_channel(&current_channel, NULL);
//         if (current_channel == storage.channel) {
//             Serial.printf("Channel verification successful: Current channel is %d\n", current_channel);
//         } else {
//             Serial.printf("Channel verification failed: Expected %d, but got %d\n", storage.channel, current_channel);
//         }

"""