espressif/esp-idf #9593: ESP32 build successfully but ESP32S3 failed.

aboyuanna, 2022-08-19:
## Environment

- IDF version :V4.4.2
- Operating System: Windows

### Code to reproduce this issue

[https://github.com/aboyuanna/outlet](url)

## Debug Logs

-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.37.2.windows.2") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/HJYVM10/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/HJYVM10/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/HJYVM10/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file C:/Users/HJYVM10/Desktop/outlet/sdkconfig
Loading defaults file C:/Users/HJYVM10/Desktop/outlet/sdkconfig.defaults...
Traceback (most recent call last):
  File "C:/Users/HJYVM10/.espressif/frameworks/esp-idf/tools/kconfig_new/confgen.py", line 640, in <module>
    main()
  File "C:/Users/HJYVM10/.espressif/frameworks/esp-idf/tools/kconfig_new/confgen.py", line 326, in main
    output_function(deprecated_options, config, temp_file)
  File "C:/Users/HJYVM10/.espressif/frameworks/esp-idf/tools/kconfig_new/confgen.py", line 473, in write_json
    config_dict = get_json_values(config)
  File "C:/Users/HJYVM10/.espressif/frameworks/esp-idf/tools/kconfig_new/confgen.py", line 468, in get_json_values
    write_node(n)
  File "C:/Users/HJYVM10/.espressif/frameworks/esp-idf/tools/kconfig_new/confgen.py", line 465, in write_node
    val = int(val)
ValueError: invalid literal for int() with base 10: ''
CMake Error at C:/Users/HJYVM10/.espressif/frameworks/esp-idf/tools/cmake/kconfig.cmake:266 (message):
  Failed to run confgen.py
  (C:/Users/HJYVM10/.espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe;C:/Users/HJYVM10/.espressif/frameworks/esp-idf/tools/kconfig_new/confgen.py;--kconfig;C:/Users/HJYVM10/.espressif/frameworks/esp-idf/Kconfig;--sdkconfig-rename;C:/Users/HJYVM10/.espressif/frameworks/esp-idf/sdkconfig.rename;--config;C:/Users/HJYVM10/Desktop/outlet/sdkconfig;--defaults;C:/Users/HJYVM10/Desktop/outlet/sdkconfig.defaults;--env-file;C:/Users/HJYVM10/Desktop/outlet/build/config.env).
  Error 1
Call Stack (most recent call first):
  C:/Users/HJYVM10/.espressif/frameworks/esp-idf/tools/cmake/build.cmake:525 (__kconfig_generate_config)
  C:/Users/HJYVM10/.espressif/frameworks/esp-idf/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:17 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/HJYVM10/Desktop/outlet/build/CMakeFiles/CMakeOutput.log".

aboyuanna, 2022-08-20:
I have found the reason.

closed 2022-08-20 by aboyuanna

JohnJohnXia, 2023-03-03:
So what's the reason? I have the same problem