espressif/esp-idf #9594: esp.emac: no mem for receive buffer

SERIDJ, 2022-08-19:
hi,

ESP-IDF v4.4.1-dirty
ESP32-WROVER-IE
Operating System: Windows
IDE Visual Studio code

iam running a web sevrer in my esp32 its work fine without SSL (the size of web page around 1Mb) .
so when i ensable HTTPS for my server i got error : esp.emac: no mem for receive buffer

log output detail :

```
(15078) esp_https_server: performing session handshake
I (15138) MB_TCP_SLAVE_PORT: Socket (#53), listener on port: 502, errno=0
I (15138) MB_TCP_SLAVE_PORT: Protocol stack initialized.
I (15188) Modbus server: Modbus slave stack initialized from task.
I (15188) Modbus server: Start modbus test...
I (15588) app_main: Running partition: ota_0
E (16178) esp.emac: no mem for receive buffer
E (16178) esp.emac: no mem for receive buffer
E (16178) esp.emac: no mem for receive buffer
E (16188) esp.emac: no mem for receive buffer
E (16188) esp.emac: no mem for receive buffer
E (16198) esp.emac: no mem for receive buffer
E (16198) esp.emac: no mem for receive buffer
E (16208) esp.emac: no mem for receive buffer
E (16208) esp.emac: no mem for receive buffer
E (16218) esp.emac: no mem for receive buffer
E (16218) esp.emac: no mem for receive buffer
E (16228) esp.emac: no mem for receive buffer
E (16228) esp.emac: no mem for receive buffer
E (16238) esp.emac: no mem for receive buffer
E (16238) esp.emac: no mem for receive buffer
E (16248) esp.emac: no mem for receive buffer
E (16248) esp.emac: no mem for receive buffer
E (16258) esp.emac: no mem for receive buffer
E (16258) esp.emac: no mem for receive buffer
E (16268) esp.emac: no mem for receive buffer
E (16268) esp.emac: no mem for receive buffer
E (16278) esp.emac: no mem for receive buffer
E (16278) esp.emac: no mem for receive buffer
E (16288) esp.emac: no mem for receive buffer
E (16288) esp.emac: no mem for receive buffer
E (16298) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x4310
E (16298) esp.emac: no mem for receive buffer
E (16308) esp.emac: no mem for receive buffer
I (16318) esp-tls-mbedtls: (FFFFBCF0): RSA - The private key operation failed : BIGNUM - Memory allocation failed
E (16318) esp.emac: no mem for receive buffer
E (16328) esp_https_server: esp_tls_create_server_session failed
E (16328) esp.emac: no mem for receive buffer
E (16348) esp.emac: no mem for receive buffer
W (16348) httpd: httpd_accept_conn: session creation failed
W (16358) httpd: httpd_server: error accepting new connection
I (16358) esp_https_server: performing session handshake
E (16378) Dynamic Impl: alloc(4770 bytes) failed
E (16378) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7F00
I (16388) esp-tls-mbedtls: (FFFF8100): SSL - Memory allocation failed
E (16388) esp_https_server: esp_tls_create_server_session failed
W (16398) httpd: httpd_accept_conn: session creation failed
W (16408) httpd: httpd_server: error accepting new connection
I (17438) esp_https_server: performing session handshake
```

how i can use my PSRAM to solve this code ?

king1688, 2022-08-22:
I have the same problem when I complie and run the example under /ethernet/eth2ap examples code. if you got a solution, please let me know, thank you

kostaond, 2022-09-01:
Duplicate of #9308

closed 2022-09-01 by kostaond

thiagohd, 2022-09-30:
Same problem here.

kostaond, 2022-10-31:
Could you please check if [this WIP update](https://github.com/kostaond/esp-idf/tree/feature/emac_buff_optimization) helped to resolve your issue?