espressif/esp-idf #9595: `coex_enable()` crashes if `esp_wifi_set_ps(WIFI_PS_NONE);` has been called

thorrak, 2022-08-19:
## Environment

- Module or chip used:  ESP32
- IDF version: v4.4.2 (via Arduino-ESP32)
- Build System:         CMake (PlatformIO)
- Operating System:     macOS
- Using an IDE?: Yes - VSCode
- Power Supply:         USB

## Problem Description

If I do the following, in this order, I generate a crash:

1. Initialize/connect to WiFi
2. Call `esp_wifi_set_ps(WIFI_PS_NONE);`
3. Initialize Bluetooth (I'm using NimBLE-Arduino, but that in turn ends up [calling](https://github.com/h2zero/NimBLE-Arduino/blob/release/1.4/src/NimBLEDevice.cpp#L880) `esp_bt_controller_enable(ESP_BT_MODE_BLE)`)
4. I get a crash on [this line](https://github.com/espressif/esp-idf/blob/v4.4.2/components/bt/controller/esp32/bt.c#L1848)

By contrast, if I initialize bluetooth first, and _then_ call `esp_wifi_set_ps(WIFI_PS_NONE);` I also get a crash, but with an error message is output: `E (10532) wifi:Error! Should enable WiFi modem sleep when both WiFi and Bluetooth are enabled!!!!!!`

### Expected Behavior

I would expect that an error message would be output by `coex_enable()` if the WiFi sleep mode has been set to `WIFI_PS_NONE`

### Actual Behavior

A crash:

```
abort() was called at PC 0x4017f830 on core 1

Backtrace:0x40083fb1:0x3ffd35e00x40095e35:0x3ffd3600 0x4009bae1:0x3ffd3620 0x4017f830:0x3ffd36a0 0x4017f65b:0x3ffd36c0 0x4013e261:0x3ffd36e0 0x400f48f2:0x3ffd3700 0x400decb5:0x3ffd3750 0x400d95c9:0x3ffd3790 0x401101f6:0x3ffd37b0 

  #0  0x40083fb1:0x3ffd35e0 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:402
  #1  0x40095e35:0x3ffd3600 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:128
  #2  0x4009bae1:0x3ffd3620 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/abort.c:46
  #3  0x4017f830:0x3ffd36a0 in coex_core_enable at ??:?
  #4  0x4017f65b:0x3ffd36c0 in coex_enable at ??:?
  #5  0x4013e261:0x3ffd36e0 in esp_bt_controller_enable at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/controller/esp32/bt.c:1848
  #6  0x400f48f2:0x3ffd3700 in NimBLEDevice::init(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at .pio/libdeps/lcd_ssd1306/NimBLE-Arduino/src/NimBLEDevice.cpp:770
```

Espressif-liuuuu, 2022-08-22:
Hi @thorrak , thanks for your report! 
There is only one antenna and could only be used by WiFi or BT at one time. Since BT may interrupt WiFi traffic, to avoid too much retransmission nor speed loose, WiFi modem sleep is the essential prerequisite for coexistence now, so that AP could buffer the data for our stations.

thorrak, 2022-08-22:
Thanks for the reply, @Espressif-liuuuu 

Although I understand why WiFi modem sleep is required, the issue I see is that `coax_enable()` crashes _silently_ if WiFi modem sleep is disabled. 

I would expect it to raise an error message before crashing, similar to the one that is raised if WiFi modem sleep is disabled after `coex_enable()` is called -- `E (10532) wifi:Error! Should enable WiFi modem sleep when both WiFi and Bluetooth are enabled!!!!!!`

rtheil-growlink, 2023-06-14:
+1 for a proper error message. Just ran into this issue and had just disabled modem sleep. I figured out a way around it, but it took a bit of troubleshooting to figure out what I had changed to cause the crash.

MaxwellAlan, 2024-03-05:
This feat has been supported from below commit:
v5.0: https://github.com/espressif/esp-idf/commit/728cd2489a3fd01e8d27a95260a03e0d0fdb6704
v5.1: https://github.com/espressif/esp-idf/commit/18ffa02b729c340945451251f3e56f20ba18038a
master/v5.2: https://github.com/espressif/esp-idf/commit/111c517da0411bc46421308a62a8b6cb3096628e
Plz have a try :)
Feel free to reopen if still have this issue.

closed 2024-03-05 by MaxwellAlan