espressif/esp-idf #9598: "CMUX: Protocol mismatch: Missed leading SOF, recovering..." warning in esp_modem

DarshanDodal, 2022-08-20:
Hello,
I am facing an issue with the "esp_modem" component. I am getting a warning " CMUX: Protocol mismatch: Missed leading SOF, recovering..." and the success rate of connection after this warning is 60%, 40% of times it does not connect. I have tried increasing the rx tx buffer, but still no luck. And, even if connection happens, I get "Protocol Recovered" log very very rarely.

P.S. I am on the latest commit.

chegewara, 2022-08-22:
I would add log next to this line
```
CMUX: Protocol mismatch: Missed leading SOF, recovering...
```
and print what its comming from uart. Its just guessing, but i have this message when BG96 restart, usually due to low power supply, and its printing `RDY`.

Alvin1Zhang, 2022-09-13:
This is a duplicate of https://github.com/espressif/esp-protocols/issues/118, closed this one and let us track in https://github.com/espressif/esp-protocols/issues/118. Thanks.

closed 2022-09-13 by Alvin1Zhang