espressif/esp-idf #9599: Crazy crash in modified bootloader calling copied local bootloader_utility functions

HomeACcessoryKid, 2022-08-20:
## Environment

- Development Kit:  none
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): 4.4.2
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:  Linux Docker container
- Using an IDE?: No
- Power Supply:         USB

## Problem Description
The overridden bootloader crashes at a specific spot that should not be possible

### Expected Behavior
No crash, behave like example code

### Actual Behavior
The code crashes at a point where it returns from a function, which was itself a successful return from a function.
If you look at the patched code below and then the output reports
```
I (384) IL: 1
I (384) IL: err=0
Fatal exception (0): IllegalInstruction
```
that seems impossible to me. How can anything crash at that point?

In comparison to the example code I observed that in compiling there is one extra warning which I also do not understand
```
/project/LCM4ESP32/bootloader_components/main/bootloader_start.c: In function 'call_start_cpu0':
/project/LCM4ESP32/bootloader_components/main/bootloader_start.c:556:1: warning: 'noreturn' function does return
 }
```
Overall, I feel I have followed instructions for bootloader work but I noticed that to make it compile I had to add this to the CMake file: `PRIV_REQUIRES spi_flash`. I do not understand however why this would then deliver crashing code.
Other than that I am lost, so your  help would be really appreciated :-)

### Steps to reproduce
1. copied bootloader_override example and it works fine 
2. adjusted elements of the example and copied local functions from boot loader_utilities.c into the bootloader_start.c
3. because the code crashed, I added some debug statements in esp_image.c to pin down the crash point
```
root@bf203b1f7ab4:/opt/esp/idf# git diff
diff --git a/components/bootloader_support/src/esp_image_format.c b/components/bootloader_support/src/esp_image_format.c
index ebdf7f334f..21bc17bb04 100644
--- a/components/bootloader_support/src/esp_image_format.c
+++ b/components/bootloader_support/src/esp_image_format.c
@@ -223,6 +223,7 @@ static esp_err_t image_load(esp_image_load_mode_t mode, const esp_partition_pos_
 #endif // BOOTLOADER_BUILD
 
     // Success!
+ESP_LOGI("IL","1");
     return ESP_OK;
 
 err:
@@ -256,7 +257,11 @@ esp_err_t bootloader_load_image(const esp_partition_pos_t *part, esp_image_metad
 #endif // CONFIG_BOOTLOADER_SKIP_...
 #endif // CONFIG_SECURE_BOOT
 
- return image_load(mode, part, data);
+ESP_LOGI("IL","0");
+ //return image_load(mode, part, data);
+ esp_err_t err=image_load(mode, part, data);
+ESP_LOGI("IL","err=%d",err);
+ return err;
 #endif // BOOTLOADER_BUILD
 }
```
[image]

### Code to reproduce this issue
[see 0.0.5 version of my repo](https://github.com/HomeACcessoryKid/LCM4ESP32/commit/8671530e03547553f145436b55d73e8b3132fe2c). No need to compile more than the bootloader. NOTE the modified partition table to wear down my ESP32 in a less critical area.

## Debug Logs

Inside my [repo](https://github.com/HomeACcessoryKid/LCM4ESP32/commit/8671530e03547553f145436b55d73e8b3132fe2c) is a `bootloader-crash.txt` file with additional info. This also shows that using the example code again it works again as expected, which proves that the app-file is working fine and nothing hardware seems to be broken.
```
entry 0x400809e4
I (27) boot: ESP-IDF v4.4.2-dirty 2nd stage bootloader
I (27) boot: compile time 13:25:27
I (27) boot: chip revision: 1
I (27) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (33) boot.esp32: SPI Speed      : 40MHz
I (37) boot.esp32: SPI Mode       : DIO
I (40) boot.esp32: SPI Flash Size : 4MB
I (44) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 otadata          OTA data         01 00 00209000 00002000
I (64) boot:  1 phy_init         RF data          01 01 0020b000 00001000
I (70) boot:  2 lcmcert_1        unknown          65 18 0020c000 00001000
I (77) boot:  3 lcmcert_2        unknown          65 18 0020d000 00001000
I (84) boot:  4 nvs              WiFi data        01 02 0020e000 00012000
I (90) boot:  5 ota_1            OTA app          00 11 00220000 000e0000
I (97) boot:  6 ota_0            OTA app          00 10 00300000 00100000
I (103) boot: End of partition table
I (106) BL4LCM32: This code is called AFTER bootloader initialization: idx=0x0
I (113) BL4LCM32: RTC booting is not valid. Try to load an app as usual
I (120) BL4LCM32: Trying partition index 0 offset 0x300000 size 0x100000
I (126) IL: 0
I (127) boot_comm: chip revision: 1, min. application chip revision: 0
I (134) esp_image: segment 0: paddr=00300020 vaddr=3f400020 size=13988h ( 80264) map
I (170) esp_image: segment 1: paddr=003139b0 vaddr=3ffb0000 size=03804h ( 14340) load
I (176) esp_image: segment 2: paddr=003171bc vaddr=40080000 size=08e5ch ( 36444) load
I (191) esp_image: segment 3: paddr=00320020 vaddr=400d0020 size=6dc58h (449624) map
I (354) esp_image: segment 4: paddr=0038dc80 vaddr=40088e5c size=0b5f4h ( 46580) load
I (374) esp_image: segment 5: paddr=0039927c vaddr=50000000 size=00010h (    16) load
I (384) IL: 1
I (384) IL: err=0
Fatal exception (0): IllegalInstruction
epc1=0x40080877, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
ets Jun  8 2016 00:22:57
```

## Other items if possible

- [ ] sdkconfig file in [repo](https://github.com/HomeACcessoryKid/LCM4ESP32/commit/8671530e03547553f145436b55d73e8b3132fe2c)
- [ ] elf file in the ``build`` folder attached
[bootloader.elf.gz](https://github.com/espressif/esp-idf/files/9387241/bootloader.elf.gz)

## Final comments:
issue #7484 is effectively running into the lack of flexibility and then difficulty of understanding how to modify the bootloader correctly. I support this request to make these local functions addressable from outside.

I have considered using the hooks, but considered it not OK considering I change the essence of the main body of the bootloader.

My main purpose is to replicate the [rboot4lcm](https://github.com/HomeACcessoryKid/rboot4lcm) functions in ESP8266.

First the use of `tempBoot`: the idea to boot from a partition ONCE and not change the otadata sectors. 
Second, the part to count how often the bootloader was started by power cycles at short intervals to provide an input mechanism that does not depend on any GPIO pin. This is used to control the behaviour of the user app or lcm app and achieve e.g. a reset of wifi parameters (this part I still have to start with once this issue is solved)

For more info see [LCM4ESP8266](https://github.com/HomeACcessoryKid/life-cycle-manager)

HomeACcessoryKid, 2022-08-21:
I updated my code to use a copy of the bootloader_support component in my local bootloader-components.
I prefixed names with `lcm_` to set them apart for just the bits that I needed to tweak.
When I had a full copy, it compiled but unexplainable behaviour moved even earlier into esp_image routines.
I removed esp_image.c from my library and I then got compiler errors that the original bootloader_support functions collided with the copied ones DESPITE that I had not REQUIRES bootloader_support anymore in the CMake file.
I removed all the duplicates and it compiled again but crashes identically to the original issue report.
Long story short, there is something fundamentally wrong in the bootloader setup that prevents tweaking to work.
The code of this new attempt can be found at [version 0.0.6 of my repo](https://github.com/HomeACcessoryKid/LCM4ESP32/commit/8d04297072eee1ffb7747512ad479da205ff59bb)

The improvement is at least in not having this `no_return` warning anymore and no need for the `PRIV_REQUIRES spi_flash` in the CMake file

HomeACcessoryKid, 2022-08-21:
Just wondering about the address space (I do not understand this stuff for real)

I took the log from a working bootloader (based on the same setup as the crashing one, but pointing into the routine that came out of the IDF)
```
mode:DIO, clock div:2
load:0x3fff0030,len:5808
load:0x40078000,len:14992
load:0x40080400,len:3744
entry 0x4008068c
I (27) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (27) boot: compile time 12:38:26
I (27) boot: chip revision: 1
I (27) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (33) boot.esp32: SPI Speed      : 40MHz
I (36) boot.esp32: SPI Mode       : DIO
I (40) boot.esp32: SPI Flash Size : 4MB
I (44) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 otadata          OTA data         01 00 00209000 00002000
I (63) boot:  1 phy_init         RF data          01 01 0020b000 00001000
I (70) boot:  2 lcmcert_1        unknown          65 18 0020c000 00001000
I (76) boot:  3 lcmcert_2        unknown          65 18 0020d000 00001000
I (83) boot:  4 nvs              WiFi data        01 02 0020e000 00012000
I (89) boot:  5 ota_1            OTA app          00 11 00220000 000e0000
I (96) boot:  6 ota_0            OTA app          00 10 00300000 00100000
I (103) boot: End of partition table
I (106) boot_comm: chip revision: 1, min. application chip revision: 0
I (112) esp_image: segment 0: paddr=00300020 vaddr=3f400020 size=13988h ( 80264) map
I (149) esp_image: segment 1: paddr=003139b0 vaddr=3ffb0000 size=03804h ( 14340) load
I (155) esp_image: segment 2: paddr=003171bc vaddr=40080000 size=08e5ch ( 36444) load
I (170) esp_image: segment 3: paddr=00320020 vaddr=400d0020 size=6dc58h (449624) map
I (333) esp_image: segment 4: paddr=0038dc80 vaddr=40088e5c size=0b5f4h ( 46580) load
I (352) esp_image: segment 5: paddr=0039927c vaddr=50000000 size=00010h (    16) load
I (362) boot: Loaded app from partition at offset 0x300000
I (362) boot: Disabling RNG early entropy source...
I (372) cpu_start: Pro cpu up.
I (372) cpu_start: Starting app cpu, entry point is 0x40081194
```
Now, if the initial numbers are the memory where the bootloader is running then part of it runs in 40080400+ space

If then the segment 2 loads from 40080000-40088e5b then it would overwrite that space.
If this is the case, why then does the bootloader not crash in the working example I show here?

ESP-Marius, 2022-09-01:
@HomeACcessoryKid  Sorry for the late reply, it seems to me like you already solved most of the earlier issues yourself right? So i'll just comment on the last question you posed:

> If then the segment 2 loads from 40080000-40088e5b then it would overwrite that space.
If this is the case, why then does the bootloader not crash in the working example I show here?

I think the answer here is the `iram_loader_seg`, code needed for the final stages of the secondary bootloader, i.e. the loading of the app, is placed in this segment. So at this point any code that ran earlier from `iram_seg` (0x40080000) is safe to be overwritten with data from the app

HomeACcessoryKid, 2022-09-02:
Thanks @ESP-Marius for your reply, that clarifies that detail. Not that I understand it for now in detail, and maybe this explanation does not apply to the way an additional component gets loaded and causes the running code to get clobbered.

NOTE, my issue itself is not solved. In [version 0.0.6 of my repo](https://github.com/HomeACcessoryKid/LCM4ESP32/commit/8d04297072eee1ffb7747512ad479da205ff59bb) I have made a cleaner setup, but the issue happens there as well. Which proves even more there is something buggy in there.
Can you try to investigate this bug based on [version 0.0.6 of my repo](https://github.com/HomeACcessoryKid/LCM4ESP32/commit/8d04297072eee1ffb7747512ad479da205ff59bb)?

And if you have additional proposals for me to try, do ask.

ESP-Marius, 2022-09-05:
@HomeACcessoryKid I took a look at your `0.0.6` repo and I think I see the problem. It is actually related to my last answer about `iram_loader_seg`. Code that runs at the end of the bootloader needs to be put a place where it will not get overwritten by the APP code during loading, this is why we put it in the section `iram_loader_seg`.

Take `bootloader_utility_load_boot_image` as an example: in `components/bootloader/subproject/main/ld/esp32/bootloader.ld` we put everything from `bootloader_utility` in this segment [here](https://github.com/espressif/esp-idf/blob/master/components/bootloader/subproject/main/ld/esp32/bootloader.ld#L53).

But this depends on the actual names of libraries/functions, so when you compile your replacements they no longer fit this pattern and gets placed in the default section `iram_seg`. So when you start loading the app your code gets overwritten and you get this crash. 

To test this you can edit `bootloader.ld` from:

```
    *libbootloader_support.a:bootloader_utility.*(.literal .text .literal.* .text.*)
    *libbootloader_support.a:bootloader_sha.*(.literal .text .literal.* .text.*)
```

to:
```
    *libbootloader_support.a:bootloader_utility.*(.literal .text .literal.* .text.*)
    *liblcm_bootloader_support.a:lcm_bootloader_utility.*(.literal .text .literal.* .text.*)
    *libbootloader_support.a:bootloader_sha.*(.literal .text .literal.* .text.*)
```

This should force your new functions into the correct IRAM section. With this change your app is able to boot successfully on my end.

ESP-Marius, 2022-09-05:
As far fixing this without touching IDF code:

I think adding your own linkerfile to place your code should work OK. Something like `lcm.ld`:

```

SECTIONS
{
  .lcm_iram_loader.text :
  {
    . = ALIGN (16);
    _lcm_loader_text_start = ABSOLUTE(.);
    *liblcm_bootloader_support.a:lcm_bootloader_utility.*(.literal .text .literal.* .text.*)
    _lcm_loader_text_end = ABSOLUTE(.);
  } > iram_loader_seg
}
```

And then add this to the list of linkerfiles in `bootloader_components/main/CMakeLists.txt`. Note that in this case the new linkerscript needs to be listed first, as it needs to be parsed before any of the IDF bootloader linkersscripts or the function will end up in the wrong section.

HomeACcessoryKid, 2022-09-05:
Thanks a lot for finding this and making it a proposal for a fix without touching IDF.
I will test this later this week since I don't have time at short notice.
But if it works, it would be good to update the README that comes with the bootloader_overwrite example.
This is not an obvious thing to understand if you are relatively new to bootloader programming.

And another way to avoid this altogether is making those basic functions I had to rebuild not be local static as suggested by me and others. But that can be a long term goal.

closed 2022-10-25 by espressif-bot