espressif/esp-idf #9600: websocket server to echo close frame

hyansuper, 2022-08-21:
see the last paragraph of [section-5.5.1](https://www.rfc-editor.org/rfc/rfc6455#section-5.5.1)

>If an endpoint receives a Close frame and did not previously send a
   Close frame, the endpoint MUST send a Close frame in response.  (When
   sending a Close frame in response, the endpoint typically echos the
   status code it received.)

The below client program exits abnormally because the [ws server](https://github.com/espressif/esp-idf/blob/master/examples/protocols/http_server/ws_echo_server/main/ws_echo_server.c) does not by default echo client's close frame. but if I simply set `.handle_ws_control_frames = true` in [here](https://github.com/espressif/esp-idf/blob/master/examples/protocols/http_server/ws_echo_server/main/ws_echo_server.c#L116) so that client's close frame will be dispatched to the echo_handler function and be sent back, now the below program exits cleanly. But this should be done internally without deferring to user code.

Client test program in python
```python
import websockets
import asyncio

async def recv(ws):
    async for data in ws:
        print('recv:', data)

async def main():
    async with websockets.connect('ws://192.168.1.101/ws') as ws:
        recv_task = asyncio.create_task(recv(ws))
        print('send:', 'hello')
        await ws.send('hello')

    await recv_task

asyncio.run(main())
```

Also, the [wss server example shouldn't trim payload of close frame](https://github.com/espressif/esp-idf/blob/master/examples/protocols/https_server/wss_server/main/wss_server_example.c#L87~L88) , In fact, the client send close frame with a 2-byte payload of value 1000 (big endian), just send it back.

ESP-YJM, 2022-08-24:
@hyansuper What IDF version you used, it seems websocket server will response CLOSE frame internally when `handle_ws_control_frames ` not set. You can check code here https://github.com/espressif/esp-idf/blob/master/components/esp_http_server/src/httpd_ws.c#L495. For echo payload code back, i think you can change the code and choose a suitable code to response.

hyansuper, 2022-08-24:
https://github.com/espressif/esp-idf/blob/master/components/esp_http_server/src/httpd_ws.c#L510
yes, the server does respond a close frame with null payload, but a proper response would be a payload of value 1000 for normal close. thanks for pointing out where to modify

closed 2022-08-24 by hyansuper