espressif/esp-idf #9601: Error: fatal error: mbedtls/config.h: No such file or directory

AAl-Hadi, 2022-08-21:
- Development Kit:      Custom ESP32S2
- IDF version v4.4.2
- Operating System:     Windows
- Using an IDE?: VS Code

## Problem Description

Hello,
I had 2 projects (one on ESP32 chip and the second on ESP32S2). They were working fine
But then I migrated to vs code
I installed everything and cloned both projects from GitHub
The ESP32S2 one doesn't compile and gives me the error: 
Error: fatal error: mbedtls/config.h: No such file or directory

The only difference is that before I was using "ESP-IDF v4.4-dirty". And now I use "ESP-IDF v4.4.2-138-g4c550fd79f"
I tried to uninstall everything and install v4.4 instead of v4.4.2 (from VS Code extension), but it'll still install v4.4.2 for some reason

Note: I posted the problem in the forum but got no response
https://www.esp32.com/viewtopic.php?f=13&t=29332#p101878

AAl-Hadi, 2022-08-28:
I found the problem
The file base64.h was included in one of my C files
I don't know how that happened or why it wasn't giving errors before
Anyway, I removed that #include line and the project builds successfully now

closed 2022-08-28 by AAl-Hadi

Teesmo, 2023-02-08:
Hi
I am encountering the same problem when trying to run the esp_mesh_lite mesh_local_control example. I made no changes to the code, I just ran it as is, but I am encountering this problem.
I am using IDF release/v4.4, and no IDE (running IDF in command line).
Board - ESP32-S2 DevKit M-1 v1.0
OS - Ubuntu 22.10

AAl-Hadi, 2023-02-08:
Hi @Teesmo 
My guess is that you have an `#include` some where you shouldn't have to