espressif/esp-idf #9602: vTaskDelay(30000 / portTICK_PERIOD_MS)  assert failed: prvTaskExitError port.c:164 (uxCriticalNesting == ~0UL)

DreamPWJ, 2022-08-21:
When I execute multi-threaded task on esp32 c3 chip, there is an error, and the chip automatically restarts
code `    xTaskCreate(&do_firmware_upgrade, "do_firmware_upgrade", 8192, NULL, 5, NULL); `
[image]
[image]

ESP-Marius, 2022-08-24:
Hi,

Are you able to provide a bit more information about this, preferably a minimal example reproducing the issue.

DreamPWJ, 2022-08-28:
esp_https_ota   Multithreaded tasks are not supported

ESP-Marius, 2022-08-31:
> esp_https_ota Multithreaded tasks are not supported,
You can try it with FreeRTOS multithreading with vTaskDelay @ESP-Marius

Are you trying to do a `esp_https_ota` from multiple tasks at the same time? I dont think that is supported, what is the use-case here? (Not sure how `vTaskDelay` is involved)

closed 2022-09-15 by DreamPWJ

jwestmoreland, 2023-04-18:
This is/was a bug - 
#define portTICK_PERIOD_MS          ( ( TickType_t ) 1000 / configTICK_RATE_HZ )

If configTICK_RATE_HZ is 1024, for instance - the calc here is 0.  vTaskDelay(30000/0); will throw an exception.  Not sure how this drifted into these ports... this was worked out a long time ago in the FreeRTOS ports.  Of course division by zero has been tested but this is an example of how this creeps back in.  Murphy is omnipresent.

GKFernandes, 2023-04-28:
Hello everyone,

I just ran into the same problem trying to compile a simple Serial.println program while using ESP32-C3

I checked to see if my FreeRTOS Tickrate was set to 1000Hz and it is, any clues on a fix to this issue?

Thanks!

homonto, 2023-12-12:
this problem still exists by the end of 2023:
```

06:52:21.893 > [StartCaptivePortal]: Task led_blink created
06:52:21.893 > [StartCaptivePortal]: CP timer STARTED
06:52:21.893 > 
06:52:21.893 > assert failed: _prvTaskExitError port.c:126 (uxCriticalNesting == ~0UL)
06:52:21.893 > Core  0 register dump:
06:52:21.893 > MEPC    : 0x40382880  RA      : 0x40388c30  SP      : 0x3fcac490  GP      : 0x3fc90800  
06:52:21.899 > TP      : 0x3fc89680  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
06:52:21.910 > S0/FP   : 0x3fcac607  S1      : 0x00000080  A0      : 0x3fcac4f4  A1      : 0x3fc92ed1  
06:52:21.915 > A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fc9a000  
06:52:21.926 > A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fcac4e8  S3      : 0x00000001  
06:52:21.933 > S4      : 0x3fcac4e8  S5      : 0x4038be6e  S6      : 0x00000000  S7      : 0x00000000  
06:52:21.937 > S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
06:52:21.948 > T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
06:52:21.954 > MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
06:52:21.965 > MHARTID : 0x00000000  
06:52:21.965 > 
06:52:21.965 > Stack memory:
06:52:21.965 > 3fcac490: 0xa5a5a5a5 0xa5a5a5a5 0x3c0d06d4 0x4038eb28 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x3fc92ed0
06:52:21.976 > 3fcac4b0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00363231 0xa5a5a5a5 0x3fc92fb8 0x3c0d06d4 0x3fc9333c
06:52:21.982 > 3fcac4d0: 0x3c0d0768 0x3fc92fc8 0x3fcac4bc 0x3fc92fcc 0x3c0d0700 0x3fc92ed0 0x00000000 0x00000000
06:52:21.993 > 3fcac4f0: 0xa5000000 0x65737361 0x66207472 0x656c6961 0x5f203a64 0x54767270 0x456b7361 0x45746978
06:52:22.004 > 3fcac510: 0x726f7272 0x726f7020 0x3a632e74 0x20363231 0x43787528 0x69746972 0x4e6c6163 0x69747365
06:52:22.010 > 3fcac530: 0x3d20676e 0x307e203d 0x00294c55 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
06:52:22.021 > 3fcac550: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x42000032 0x4038be40 0x00000000 0x3fc90800
06:52:22.026 > 3fcac570: 0x3fc89680 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
06:52:22.037 > 3fcac590: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
06:52:22.043 > 3fcac5b0: 0x00000000 0x00000000 0x00000000 0xccdb0db1 0x00000000 0x00000000 0x00000000 0x00000000
06:52:22.054 > 3fcac5d0: 0x00000000 0x00000000 0x00000000 0x4038be6e 0x00000000 0x00000000 0x00000000 0x00000000
06:52:22.060 > 3fcac5f0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678 0x00000160
06:52:22.071 > 3fcac610: 0xabba1234 0x00000154 0x3fcac560 0x6edb6172 0x3fc95fcc 0x3fca24fc 0x3fcac618 0x3fc95fc4
06:52:22.076 > 3fcac630: 0x00000018 0xa007023a 0xc015811d 0x3fcac618 0x00000000 0x00000001 0x3fcabe38 0x5f64656c
06:52:22.087 > 3fcac650: 0x6e696c62 0xc6da006b 0x00ac503d 0x00000000 0x3fcac600 0x00000001 0x00000000 0x00000000
06:52:22.098 > 3fcac670: 0x00000000 0x00000000 0x3fc9b644 0x3fc9b6ac 0x3fc9b714 0x00000000 0x00000000 0x00000001
06:52:22.104 > 3fcac690: 0x00000000 0x00000000 0x00000000 0x4209ff90 0x00000000 0x00000000 0x00000000 0x00000000
06:52:22.115 > 3fcac6b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
06:52:22.121 > 3fcac6d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
06:52:22.132 > 3fcac6f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
06:52:22.137 > 3fcac710: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
06:52:22.148 > 3fcac730: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
06:52:22.154 > 3fcac750: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x58000000 0xbaad5678
06:52:22.165 > 3fcac770: 0x00000034 0xabba1234 0x00000028 0x3c0c1938 0x5d029fd7 0xf1c6bd3a 0x4c4f3cef 0xc2945b7e
06:52:22.171 > 3fcac790: 0x00000000 0x000493e0 0x00000001 0x420013ae 0xf32b3c04 0xbaad5678 0x0002ff61 0x3fc9ac54
06:52:22.182 > 3fcac7b0: 0x3fc9ac54 0x12ca712f 0xd8452159 0x9e86657b 0x30f0e64c 0xb834fdbe 0x4727442d 0x38946f84
06:52:22.193 > 3fcac7d0: 0x8d64459c 0x900222b0 0x3b3975a7 0x83d9638d 0xccbfadec 0xd19f54b6 0x31c31408 0xb5418ac1
06:52:22.198 > 3fcac7f0: 0x97fad898 0xa3cb51e6 0xf5b4fe53 0x474cd720 0x62f744f2 0x5d98ff5a 0x510660e0 0x96d95a7c
06:52:22.210 > 3fcac810: 0x3746707d 0x219d689a 0x8fb03f60 0x9a552da0 0x7a2bc855 0x0934abe6 0x3f88e51a 0x2fc1d38d
06:52:22.215 > 3fcac830: 0x349c3b71 0x4b3d553e 0xce6756b5 0x074e3e58 0x6ed8eaa7 0xa3121f4a 0x07ff7e30 0x61addf58
06:52:22.226 > 3fcac850: 0x38b0c673 0x1b64d030 0xdd7f3238 0xa53d023c 0xfadbe99c 0x12d3fa9f 0x48c2943e 0xd9cfe588
06:52:22.232 > 3fcac870: 0x0770a687 0xad6593cb 0xf58f1f1f 0xcdcfb4a2 0xcdc54cd8 0xe523dfc3 0xb14553c6 0xaf5a9893
06:52:22.243 > 
06:52:22.243 > 
06:52:22.243 > 
06:52:22.243 > ELF file SHA256: a43226721b4415a4
06:52:22.243 > 
06:52:22.400 > Rebooting...
06:52:22.400 > ESP-ROM:esp32c3-api1-20210207
06:52:22.400 > Build:Feb  7 2021
06:52:22.401 > rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
06:52:22.401 > Saved PC:0x40382458
06:52:22.402 > SPIWP:0xee
06:52:22.402 > mode:DIO, clock div:1
06:52:22.405 > load:0x3fcd5810,len:0x438
06:52:22.407 > load:0x403cc710,len:0x918
06:52:22.410 > load:0x403ce710,len:0x25f4
06:52:22.415 > entry 0x403cc710
```

KonstantinKondrashov, 2023-12-12:
Hi @homonto!
Thanks for the report. We have not been able to reproduce this issue in the past. Could you provide more information?
- sdkconfig
- IDF version
- and other info that can help to reproduce it, maybe even a simple example.
- describe when and how it happens.

homonto, 2023-12-12:
> Hi @homonto! Thanks for the report. We have not been able to reproduce this issue in the past. Could you provide more information?
> 
> * sdkconfig
> * IDF version
> * and other info that can help to reproduce it, maybe even a simple example.
> * describe when and how it happens.

I found the error in my part of the code - I was calling a function with wrong parameter ;-)
For me topic closed - sorry to make a noise ;-)