espressif/esp-idf #9604: Pulse counter step watchpoint

Diegorro98, 2022-08-22:
### Describe the solution you'd like.

A watchpoint that gets triggered every time that  (in example) an encoder moves one step forward or backward.

Pretty useful for encoders where the position needs to be updated in real time.

### Describe alternatives you've considered.

I've done a separated task which is constantly getting the pulse count and reports it, but I think that it might be a waste on resources knowing that esp-idf is already paying attention to the encoder and updating the count.

I've also tried to set an interruption that triggers when one of the channels of the pulse counter gets triggered, but the results are unprecise because the last position obtained in a loop vs the position obtained in the interruption are always behind.
Also, setting an interruption may cause a modification on the GPIO configuration that can result in a pulse counter malfunction 

It takes more time to implement than a a watchpoint per step (which only would need to set the callback).

suda-morris, 2022-08-23:
Hi @Diegorro98 Do I understand you correctly that you want the PCNT hardware to generate an event when the counter got changed?

Diegorro98, 2022-08-23:
Yes, is exactly that!

suda-morris, 2022-08-23:
Cool, we will report this feature request to our digital team and will let you know when there's any progress.

closed 2024-07-04 by espressif-bot