espressif/esp-idf #9606: [VSCode] [gdb] Debugger not starting on first try

KonssnoK, 2022-08-22:
Using ESP32S3
Using IDF v4.4.2
Using D+ D- internal debugging JTAG.
Using VSCode (latest)

I always have to run the Start debugger command 3 times in order to connect.
That's not a big issue, but it's quite slowing down the process of debugging.
Can you check my scripts?

Thanks!

Launching OpenOCD with
```
{
    // See https://go.microsoft.com/fwlink/?LinkId=733558
    // for the documentation about the tasks.json format
    "version": "2.0.0",
    "tasks": [
        {
            "label": "OpenOCD",
            "type": "shell",
            "isBackground": true,
            "command": "C:\\src\\espressif\\.espressif\\tools\\openocd-esp32\\v0.11.0-esp32-20220411\\openocd-esp32\\bin\\openocd -f board/esp32s3-builtin.cfg",
            "problemMatcher": [
              {
                "pattern": [
                  {
                    "regexp": ".",
                    "file": 1,
                    "location": 2,
                    "message": 3
                  }
                ],
                "background": {
                  "activeOnStart": true,
                  "beginsPattern": ".",
                  "endsPattern": ".",
                }
              }
            ]
        }
    ]
}

```

Launching gdb with
```
{
    "version": "0.2.0",
    "configurations": [
      {
        "name": "GDB",
        "type": "cppdbg",
        "request": "launch",
        "MIMode": "gdb",
        "miDebuggerPath": "${command:espIdf.getXtensaGdb}",
        "program": "${workspaceFolder}/build/${command:espIdf.getProjectName}.elf",
        "windows": {
          "program": "${workspaceFolder}\\build\\${command:espIdf.getProjectName}.elf"
        },
        "cwd": "${workspaceFolder}",
        "environment": [{ "name": "PATH", "value": "${config:idf.customExtraPaths}" }],
        "setupCommands": [
          { "text": "target extended-remote:3333" },
          { "text": "set remote hardware-watchpoint-limit 2"},
          { "text": "mon reset halt" },
          { "text": "thb app_main" },
          { "text": "flushregs" }
        ],
        "externalConsole": false,
        "logging": {
          "engineLogging": true
        }
      }
    ]
  }

```

Outputs:
```
Open On-Chip Debugger  v0.11.0-esp32-20220411 (2022-04-11-08:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (7C:DF:A1:FF:5F:74)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32s3.cpu0: Debug controller was reset.
Info : esp32s3.cpu0: Core was reset.
Info : esp32s3.cpu1: Debug controller was reset.
Info : esp32s3.cpu1: Core was reset.
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Info : esp32s3.cpu0: Target halted, PC=0x420273B6, debug_reason=00000000
Info : Set GDB target to 'esp32s3.cpu0'
Info : esp32s3.cpu1: Target halted, PC=0x420273B6, debug_reason=00000000
Warn : No symbols for FreeRTOS!
Info : esp32s3.cpu0: Target halted, PC=0x403B245E, debug_reason=00000001
Info : Flash mapping 0: 0x20020 -> 0x3c030020, 52 KB
Info : Flash mapping 1: 0x30020 -> 0x42000020, 162 KB
Info : esp32s3.cpu0: Target halted, PC=0x403B245E, debug_reason=00000001
Info : Auto-detected flash bank 'esp32s3.cpu0.flash' size 8192 KB
Info : Using flash bank 'esp32s3.cpu0.flash' size 8192 KB
Info : esp32s3.cpu0: Target halted, PC=0x403B245E, debug_reason=00000001
Info : Flash mapping 0: 0x20020 -> 0x3c030020, 52 KB
Info : Flash mapping 1: 0x30020 -> 0x42000020, 162 KB
Info : Using flash bank 'esp32s3.cpu0.irom' size 164 KB
Info : esp32s3.cpu0: Target halted, PC=0x403B245E, debug_reason=00000001
Info : Flash mapping 0: 0x20020 -> 0x3c030020, 52 KB
Info : Flash mapping 1: 0x30020 -> 0x42000020, 162 KB
Info : Using flash bank 'esp32s3.cpu0.drom' size 56 KB
Info : New GDB Connection: 1, Target esp32s3.cpu0, state: halted
Warn : negative reply, retrying
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32s3.cpu0: Debug controller was reset.
Info : esp32s3.cpu0: Core was reset.
Info : esp32s3.cpu0: Target halted, PC=0x500000EF, debug_reason=00000000
Info : esp32s3.cpu0: Core was reset.
Info : esp32s3.cpu0: Target halted, PC=0x40000400, debug_reason=00000000
Info : esp32s3.cpu1: Debug controller was reset.
Info : esp32s3.cpu1: Core was reset.
Info : esp32s3.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
Info : dropped 'gdb' connection
```

```
1: (7954) Send Event AD7ProgramDestroyEvent
The program 'C:\src\v3\firmware\build\firmware.elf' has exited with code 42 (0x0000002a).

1: (7961) Shell exited, stop debugging
6
1: (7966) Send Event AD7ThreadDestroyEvent
1: (7977) EXCEPTION: System.ObjectDisposedException
1: (7977) EXCEPTION:    at Microsoft.MIDebugEngine.WorkerThread.PostOperation(Operation op)
1: (7977) EXCEPTION:    at Microsoft.MIDebugEngine.DebuggedProcess.ScheduleStdOutProcessing(String line)
1: (7977) EXCEPTION:    at MICore.Debugger.HandleThreadGroupExited(Results results)
1: (7977) EXCEPTION:    at MICore.Debugger.OnNotificationOutput(String cmd)
1: (7977) EXCEPTION:    at MICore.Debugger.ProcessStdOutLine(String line)
1: (7977) EXCEPTION:    at Microsoft.MIDebugEngine.DebuggedProcess.<>c__DisplayClass94_0.<ScheduleStdOutProcessing>b__0()
1: (7977) EXCEPTION:    at Microsoft.MIDebugEngine.WorkerThread.ThreadFunc()
[Inferior 1 (Remote target) detached]

1: (4092) Send Event AD7ProgramDestroyEvent
The program 'C:\src\v3\firmware\build\firmware.elf' has exited with code 42 (0x0000002a).

1: (4101) Shell exited, stop debugging
1: (4108) Send Event AD7ThreadCreateEvent
1: (4109) Send Event AD7ThreadDestroyEvent
1: (4121) EXCEPTION: System.ObjectDisposedException
1: (4121) EXCEPTION:    at Microsoft.MIDebugEngine.WorkerThread.PostOperation(Operation op)
1: (4121) EXCEPTION:    at Microsoft.MIDebugEngine.DebuggedProcess.ScheduleStdOutProcessing(String line)
1: (4121) EXCEPTION:    at MICore.Debugger.HandleThreadGroupExited(Results results)
1: (4121) EXCEPTION:    at MICore.Debugger.OnNotificationOutput(String cmd)
1: (4121) EXCEPTION:    at MICore.Debugger.ProcessStdOutLine(String line)
1: (4121) EXCEPTION:    at Microsoft.MIDebugEngine.DebuggedProcess.<>c__DisplayClass94_0.<ScheduleStdOutProcessing>b__0()
1: (4121) EXCEPTION:    at Microsoft.MIDebugEngine.WorkerThread.ThreadFunc()

```

brianignacio5, 2022-08-29:
Have you tried using the [ESP-IDF Extension](https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension) `espidf` debugger ?

Your launch.json configuration uses the [Microsoft C/C++ extension](https://marketplace.visualstudio.com/items?itemName=ms-vscode.cpptools) `cppdbg` debug configuration and their debug code is not maintained by us. You could open a PR in their repository related to this.

KonssnoK, 2024-02-13:
i think the issue was generated by multiple packages installed in vscode. after moving to a new laptop it didn't happe nanymore

closed 2024-02-13 by KonssnoK