espressif/esp-idf #9610: funny characters are printed out when using more recent IDF

nezen, 2022-08-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

platform: windows10
terminal tool: MobaXterm
IDF5.0  commit: 4d14c2ef2d9d08cd1dcbb68a8bb0d76a666e2b4b

A few days ago I switched to idf5.0 and gave it try, then I noticed funny characters were printed out in the terminal when everything is still normal under idf4.4. It seems those characters are supposed to represent the green color or something. Hopefully, it's not an issue caused by a lack of some feature under windows.

Thanks for your time in advance.

[image]

atanisoft, 2022-08-23:
@nezen These "funny characters" are ANSI escape sequences for colors, check your sdkconfig for https://github.com/espressif/esp-idf/blob/7c2b137ffe8a91ef0967cd5da1fe330ff5912399/components/log/Kconfig#L87-L93. Disable that and you won't have ANSI sequences in your log output.

nezen, 2022-08-24:
@atanisoft Thanks for your help. Disabling that config does help get rid of those sequences. But at the same time, I noticed this option is as well enabled in another menuconfig of my project which has an IDF4.4 dependency and the ANSI color display is just perfect. Not sure if it's due to some changes in the newer IDF.

atanisoft, 2022-08-24:
@nezen a bit unlikely that there was a change in IDF recently in this area.

nezen, 2022-08-24:
@atanisoft OK, thank you:)

ifedotov1984, 2022-08-24:
> @nezen a bit unlikely that there was a change in IDF recently in this area.

scripts that refer to monitor.py in 5.0 have been exactly changed since version 4.4.1

dobairoland, 2022-08-25:
We are sorry for the issue. We will look into this soon.

closed 2022-09-02 by espressif-bot