espressif/esp-idf #9612: The task is launched with a priority higher than the specified one

kotyara12, 2022-08-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Good afternoon.

The problem is not massive, on most of my devices everything works correctly. But only one of them is completely incomprehensible, I have spent several days trying to understand the source of the problem. The matter is that I start a task with a priority of 6U - 10U, and actually the task works with a priority of 23. How so???!!!

Device: Espressif ESP32-WROVER-IE 16MB (ESP32-D0WD-V3)
IDE: VSCode + PlatformIO + ESP-IDF 4.4.1 (this is the maximum version at the moment)

The task is started statically:
```
#define CONFIG_TASK_PRIORITY_SENSORS        6U
...

    static StaticTask_t sensorsTaskBuffer;
    static StackType_t sensorsTaskStack[CONFIG_SENSORS_TASK_STACK_SIZE];
    _sensorsTask = xTaskCreateStaticPinnedToCore(sensorsTaskExec, "sensors", CONFIG_SENSORS_TASK_STACK_SIZE, NULL, CONFIG_TASK_PRIORITY_SENSORS, sensorsTaskStack, &sensorsTaskBuffer, 1);
```

If you check the priority at the same time (by outputting to the log), then it is really equal to 6

Every 10 minutes, the device sends a list of tasks to the broker in JSON form:
```
[
  {
    "id": 3,
    "name": "esp_timer",
    "core": 0,
    "state": "running",
    "current_priority": 22,
    "base_priority": 22,
    "run_time_counter": 0,
    "stack_base": "3ffc1108",
    "stack_minimum": 2488
  },
  {
    "id": 6,
    "name": "IDLE",
    "core": 1,
    "state": "ready",
    "current_priority": 0,
    "base_priority": 0,
    "run_time_counter": 0,
    "stack_base": "3ffc3bb4",
    "stack_minimum": 1724
  },
  {
    "id": 5,
    "name": "IDLE",
    "core": 0,
    "state": "ready",
    "current_priority": 0,
    "base_priority": 0,
    "run_time_counter": 0,
    "stack_base": "3ffc314c",
    "stack_minimum": 1700
  },
  {
    "id": 17,
    "name": "alarm",
    "core": 1,
    "state": "blocked",
    "current_priority": 8,
    "base_priority": 8,
    "run_time_counter": 0,
    "stack_base": "3ffbcf8c",
    "stack_minimum": 3204
  },
  {
    "id": 24,
    "name": "mqtt_task",
    "core": -1,
    "state": "blocked",
    "current_priority": 12,
    "base_priority": 12,
    "run_time_counter": 0,
    "stack_base": "3ffe2c34",
    "stack_minimum": 1948
  },
  {
    "id": 18,
    "name": "sensors",
    "core": 1,
    "state": "blocked",
    "current_priority": 23,
    "base_priority": 23,
    "run_time_counter": 0,
    "stack_base": "3ffb8f58",
    "stack_minimum": 2344
  },
  {
    "id": 9,
    "name": "led_system",
    "core": 1,
    "state": "blocked",
    "current_priority": 13,
    "base_priority": 13,
    "run_time_counter": 0,
    "stack_base": "3ffd2750",
    "stack_minimum": 1088
  },
  {
    "id": 23,
    "name": "pinger",
    "core": 1,
    "state": "blocked",
    "current_priority": 3,
    "base_priority": 3,
    "run_time_counter": 0,
    "stack_base": "3ffb8114",
    "stack_minimum": 940
  },
  {
    "id": 20,
    "name": "tiT",
    "core": -1,
    "state": "blocked",
    "current_priority": 18,
    "base_priority": 18,
    "run_time_counter": 0,
    "stack_base": "3ffdaa4c",
    "stack_minimum": 1476
  },
  {
    "id": 19,
    "name": "sys_evt",
    "core": 0,
    "state": "blocked",
    "current_priority": 20,
    "base_priority": 20,
    "run_time_counter": 0,
    "stack_base": "3ffd9474",
    "stack_minimum": 2556
  },
  {
    "id": 1,
    "name": "ipc0",
    "core": 0,
    "state": "blocked",
    "current_priority": 24,
    "base_priority": 24,
    "run_time_counter": 0,
    "stack_base": "3ffaf7d8",
    "stack_minimum": 456
  },
  {
    "id": 14,
    "name": "siren",
    "core": 1,
    "state": "blocked",
    "current_priority": 13,
    "base_priority": 13,
    "run_time_counter": 0,
    "stack_base": "3ffd4aa0",
    "stack_minimum": 448
  },
  {
    "id": 2,
    "name": "ipc1",
    "core": 1,
    "state": "blocked",
    "current_priority": 24,
    "base_priority": 24,
    "run_time_counter": 0,
    "stack_base": "3ffc0d04",
    "stack_minimum": 460
  },
  {
    "id": 15,
    "name": "flasher",
    "core": 1,
    "state": "blocked",
    "current_priority": 13,
    "base_priority": 13,
    "run_time_counter": 0,
    "stack_base": "3ffd5108",
    "stack_minimum": 488
  },
  {
    "id": 11,
    "name": "led_warn",
    "core": 1,
    "state": "blocked",
    "current_priority": 13,
    "base_priority": 13,
    "run_time_counter": 0,
    "stack_base": "3ffd3c20",
    "stack_minimum": 1168
  },
  {
    "id": 12,
    "name": "tg_send",
    "core": 1,
    "state": "blocked",
    "current_priority": 10,
    "base_priority": 10,
    "run_time_counter": 0,
    "stack_base": "3ffb40c0",
    "stack_minimum": 2896
  },
  {
    "id": 13,
    "name": "http_send",
    "core": 1,
    "state": "blocked",
    "current_priority": 5,
    "base_priority": 5,
    "run_time_counter": 0,
    "stack_base": "3ffb6198",
    "stack_minimum": 1880
  },
  {
    "id": 10,
    "name": "led_alarm",
    "core": 1,
    "state": "blocked",
    "current_priority": 13,
    "base_priority": 13,
    "run_time_counter": 0,
    "stack_base": "3ffd31b8",
    "stack_minimum": 1112
  },
  {
    "id": 8,
    "name": "re_events",
    "core": 0,
    "state": "blocked",
    "current_priority": 19,
    "base_priority": 19,
    "run_time_counter": 0,
    "stack_base": "3ffd04f0",
    "stack_minimum": 1776
  },
  {
    "id": 21,
    "name": "wifi",
    "core": 0,
    "state": "blocked",
    "current_priority": 23,
    "base_priority": 23,
    "run_time_counter": 0,
    "stack_base": "3ffdc5f4",
    "stack_minimum": 4492
  },
  {
    "id": 7,
    "name": "Tmr Svc",
    "core": 0,
    "state": "blocked",
    "current_priority": 1,
    "base_priority": 1,
    "run_time_counter": 0,
    "stack_base": "3ffc461c",
    "stack_minimum": 1524
  },
  {
    "id": 16,
    "name": "buzzer",
    "core": 1,
    "state": "blocked",
    "current_priority": 13,
    "base_priority": 13,
    "run_time_counter": 0,
    "stack_base": "3ffd5730",
    "stack_minimum": 480
  }
]
```
And there you can see that the task does not work with a given priority, but with 23 !!! It's impossible

```
  {
    "id": 18,
    "name": "sensors",
    "core": 1,
    "state": "blocked",
    "current_priority": 23,
    "base_priority": 23,
    "run_time_counter": 0,
    "stack_base": "3ffb8f58",
    "stack_minimum": 2344
  },
```

AxelLin, 2022-08-23:
https://github.com/espressif/esp-idf/issues/7580

kotyara12, 2022-08-23:
Thank you very much. Just not sure what to do in this case.

AxelLin, 2022-08-23:
It needs fix,this issue has been reported by many people.
@KonstantinKondrashov

KonstantinKondrashov, 2022-09-05:
It is fixed on master https://github.com/espressif/esp-idf/commit/aea2fe08162e1ba86ad0189178c1f1cf29238bbd. We will also backport it.