espressif/esp-idf #9613: WPS example doesn't work

alex-ea, 2022-08-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am having issues with the WPS example. It looks like the WPS part of the code works fine, but it then fails when trying to connect to the AP. 
From Debug mode, I was able to locate that the issue happened on line 110 of wps.c: `esp_wifi_connect();` returns  `ESP_ERR_WIFI_CONN `

IDE version: ESP-IDF version 4.4.2
Board: esp32-c3-DevKitC-02v1.1

```
I (31) boot: ESP-IDF v4.4.2-dirty 2nd stage bootloader
I (31) boot: compile time 13:59:50
I (31) boot: chip revision: 3
I (33) boot.esp32c3: SPI Speed      : 80MHz
I (38) boot.esp32c3: SPI Mode       : DIO
I (43) boot.esp32c3: SPI Flash Size : 2MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (86) boot: End of partition table
I (90) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=155b8h ( 87480) map
I (112) esp_image: segment 1: paddr=000255e0 vaddr=3fc8f000 size=02b78h ( 11128) load
I (115) esp_image: segment 2: paddr=00028160 vaddr=40380000 size=07eb8h ( 32440) load
I (124) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=7efe4h (520164) map
I (208) esp_image: segment 4: paddr=000af00c vaddr=40387eb8 size=06fa0h ( 28576) load
I (213) esp_image: segment 5: paddr=000b5fb4 vaddr=50000010 size=00010h (    16) load
I (219) boot: Loaded app from partition at offset 0x10000
I (220) boot: Disabling RNG early entropy source...
I (436) cpu_start: Pro cpu up.
I (445) cpu_start: Pro cpu start user code
I (445) cpu_start: cpu freq: 160000000
I (445) cpu_start: Application information:
I (447) cpu_start: Project name:     wps_example
I (453) cpu_start: App version:      1
I (457) cpu_start: Compile time:     Aug 23 2022 13:59:42
I (463) cpu_start: ELF file SHA256:  55ccb8f2422e64eb...
I (469) cpu_start: ESP-IDF:          v4.4.2-dirty
I (475) heap_init: Initializing. RAM available for dynamic allocation:
I (482) heap_init: At 3FC95CD0 len 0002A330 (168 KiB): DRAM
I (488) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (495) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (502) spi_flash: detected chip: generic
I (506) spi_flash: flash io: dio
W (510) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (523) sleep: Configure to isolate all GPIO pins in sleep state
I (530) sleep: Enable automatic switching of GPIO sleep configuration
I (537) cpu_start: Starting scheduler.
I (546) pp: pp rom version: 9387209
I (546) net80211: net80211 rom version: 9387209
I (556) wifi:wifi driver task: 3fc9ea8c, prio:23, stack:6656, core=0
I (556) system_api: Base MAC address is not set
I (566) system_api: read default base MAC address from EFUSE
I (576) wifi:wifi firmware version: eeaa27d
I (576) wifi:wifi certification version: v7.0
I (576) wifi:config NVS flash: enabled
I (576) wifi:config nano formating: disabled
I (586) wifi:Init data frame dynamic rx buffer num: 32
I (586) wifi:Init management frame dynamic rx buffer num: 32
I (596) wifi:Init management short buffer num: 32
I (596) wifi:Init dynamic tx buffer num: 32
I (606) wifi:Init static tx FG buffer num: 2
I (606) wifi:Init static rx buffer size: 1600
I (606) wifi:Init static rx buffer num: 10
I (616) wifi:Init dynamic rx buffer num: 32
I (616) wifi_init: rx ba win: 6
I (626) wifi_init: tcpip mbox: 32
I (626) wifi_init: udp mbox: 6
I (626) wifi_init: tcp mbox: 6
I (636) wifi_init: tcp tx win: 5744
I (636) wifi_init: tcp rx win: 5744
I (646) wifi_init: tcp mss: 1440
I (646) wifi_init: WiFi IRAM OP enabled
I (646) wifi_init: WiFi RX IRAM OP enabled
I (656) phy_init: phy_version 912,d001756,Jun  2 2022,16:28:07
I (696) wifi:mode : sta (7c:df:a1:da:72:94)
I (696) wifi:enable tsf
I (696) example_wps: WIFI_EVENT_STA_START
I (696) example_wps: start wps...
I (696) wpa: manufacturer: ESPRESSIF, model number: ESP32, model name: ESPRESSIF IOT, device name: ESP STATION
I (706) wpa: wifi_wps_enable

I (3206) wpa: WPS: scan done
I (5716) wpa: WPS: scan done
I (8226) wpa: WPS: scan done
I (10736) wpa: WPS: scan done
I (13246) wpa: WPS: scan done
I (15746) wpa: WPS: scan done
I (18256) wpa: WPS: scan done
I (20766) wpa: WPS: scan done
I (23266) wpa: WPS: scan done
I (25776) wpa: WPS: scan done
I (28286) wpa: WPS: scan done
I (30796) wpa: WPS: scan done
I (33306) wpa: WPS: scan done
I (35816) wpa: WPS: scan done
I (38326) wpa: WPS: scan done
I (40826) wpa: WPS: scan done
I (43336) wpa: WPS: scan done
I (45846) wpa: WPS: scan done
I (48356) wpa: WPS: scan done
I (50856) wpa: WPS: scan done
I (53366) wpa: WPS: scan done
I (55876) wpa: WPS: scan done
I (58386) wpa: WPS: scan done
I (60896) wpa: WPS: scan done
I (63396) wpa: WPS: scan done
I (65906) wpa: WPS: scan done
I (68416) wpa: WPS: scan done
I (70916) wpa: WPS: scan done
I (73426) wpa: WPS: scan done
I (75936) wpa: WPS: scan done
I (78456) wpa: WPS: scan done
I (80956) wpa: WPS: scan done
I (83466) wpa: WPS: scan done
I (85976) wpa: WPS: scan done
I (88486) wpa: WPS: scan done
sm BSSid: 00:00:00:00:00:00 scan BSSID 24:a7:dc:c9:70:52
I (90996) wpa: WPS: scan done
I (91736) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (92396) wifi:state: init -> auth (b0)
I (92396) wifi:state: auth -> assoc (0)
I (92406) wifi:state: assoc -> run (10)
W (92416) wifi:<ba-add>idx:0 (ifx:0, 24:a7:dc:c9:70:52), tid:0, ssn:0, winSize:64
I (94426) wifi:state: run -> init (2c0)
W (94426) wifi:<ba-del>idx
I (94426) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (94426) example_wps: WIFI_EVENT_STA_DISCONNECTED
E (94426) wifi:sta is connecting, return error
I (95266) example_wps: WIFI_EVENT_STA_WPS_ER_SUCCESS
I (95266) wpa: wifi_wps_disable
E (95266) wifi:sta is connecting, return error
```

alex-ea, 2022-09-02:
Just checked out master and it seems the issue is fixed. 
@kapilkedawat Any plans to add the patch to version 4.4?

kapilkedawat, 2022-09-02:
@alex-ea changes are already merged on v4.4 as well using https://github.com/espressif/esp-idf/commit/4d9c4fd7e13fd003b625cdf3b109cf43cb4d455e , Please have a try.

kapilkedawat, 2022-09-04:
> > @alex-ea changes are already merged on v4.4 as well using [4d9c4fd](https://github.com/espressif/esp-idf/commit/4d9c4fd7e13fd003b625cdf3b109cf43cb4d455e) , Please have a try.
> 
> Could you confirm if v4.3 needs fix?

v4.3 doesn't require this fix.

closed 2022-09-05 by Alvin1Zhang