espressif/esp-idf #9614: Error with gen_esp32part.py

josesimoes, 2022-08-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

Being able to use gen_esp32part.py to generate partition table from a CSV file.

### What is the actual behavior?

An error is thrown.

### Steps to reproduce.

run this command:

`gen_esp32part.py --flash-size 4MB E:/GitHub/nf-interpreter/targets/ESP32/_IDF/esp32/partitions_nanoclr_4mb.csv E:/GitHub/nf-interpreter/build/partitions_4mb.bin`

### Build or installation Logs.

```plain
[build]   File "E:\GitHub\esp-idf\components\partition_table\gen_esp32part.py", line 254
[build]     def verify_size_fits(self, flash_size_bytes: int) -> None:
[build]                                                ^
[build] SyntaxError: invalid syntax
```

### More Information.

I believe there's an error in the declaration of verify_size_fits.
If I replace it with `def verify_size_fits(self, flash_size_bytes):`  it works just fine.

The current version was probably meant to be a comment or similar...

igrr, 2022-08-23:
@josesimoes is it possible that you are using an old version of Python, like python 3.5? IDF v4.4.x [supports python 3.6 or later](https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32/get-started/linux-setup.html#python-compatibility), and type annotations syntax should work in python 3.6.

josesimoes, 2022-08-23:
@igrr that was the cause indeed. 
All this started after I've run the install & export scripts for the 4.4.2, some failed, in particular the python step and I've missed that.

Thanks!

closed 2022-08-23 by josesimoes