espressif/esp-idf #9615: Why are you using int64_t for esp_timer_get_time output?

amirhb29, 2022-08-23:
Why are you using int64 for timer counter when we have 64 bits and we can uint64 to cover more time before overflowing the counter?

igrr, 2022-08-23:
@amirhb29 Good question! Since esp_timer counter is counting microseconds, the difference between a 63-bit and 64-bit counter is the difference between device uptime of ~292 thousand years and 584 thousand years. We thought that for most devices built with ESP-IDF, this difference wouldn't matter!

In fact, on ESP32-C3 and later chips, the underlying SYSTIMER peripheral has "just" a 52-bit counter, and it counts at 16 MHz frequency. This translates to approximately 9 years of _uninterrupted_ uptime, which corresponds to a 48-bit microsecond counter.

On the other hand, returning the int64_t makes it much easier to do time difference calculations: you can calculate `t_1 - t_0` without having to check whether `t_1` is greater than `t_0` and without potentially incorrect casts between uint64_t and int64_t.

ESP-Marius, 2022-09-06:
Closing this one as the original question has been answered. Feel free to reopen if there is anything that is still unclear.

closed 2022-09-06 by ESP-Marius