espressif/esp-idf #9621: BLE_MESH: Failed to set mesh/seq data

hipal31, 2022-08-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.3.3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32C3

### Power Supply used.

External 3.3V

### What is the expected behavior?

While operating in BLE-Mesh all data should be saved properly in NVS. All that data of sequence number and RPL data like unicast, seq no, old_iv must be saved successfully in NVS. Otherwise it'll leads to failure of message.

### What is the actual behavior?

I'm getting error something like 
`I (199855) BLE_MESH: Client message 0x00008202 with timeout 2000ms
I (199865) BLE_MESH: send, app_idx 0x0000 src 0x0913 dst 0x0057
I (199875) BLE_MESH: send, len 6: 8202011a0a0a
I (199905) BLE_MESH: Settings store, waiting 0 seconds
E (200275) REEVA_NVS: Store, nvs_set_blob failed, err 4357--ESP_ERR_NVS_NOT_ENOUGH_SPACE
E (200475) BLE_MESH: Failed to set mesh/seq data (err 4357)
I (200605) BLE_MESH: Settings store, waiting 0 seconds
E (200805) BLE_MESH: Failed to set mesh/rpl/0057 data (err 4357)
E (200805) BLE_MESH: Failed to store RPL entry 0x0057`

Due to which sequence number of local node is not updated in nvs that leads to failure of message receive at destination node (may be cause of replay prevention)

### Steps to reproduce.

1. In 1 node, run onoff_client example
2. IN 50 node , run onoff_server(take 8 element of onoff_server model) example
3. then provision all nodes
4. After that from onoff_client node operate all the elements of 50 node one by one.
5. After somewhat operations this error will be displayed.

### Debug Logs.

```plain
I (145815) BLE_MESH: Client message 0x00008201 with timeout 2000ms
I (145815) BLE_MESH: send, app_idx 0x0000 src 0x0913 dst 0x000a
I (145815) BLE_MESH: send, len 2: 8201
I (145825) BLE_MESH: Settings store, waiting 0 seconds
I (145985) BLE_MESH: Settings store, waiting 0 seconds
I (146085) BLE_MESH: Client message 0x00008201 with timeout 2000ms
I (146095) BLE_MESH: send, app_idx 0x0000 src 0x0913 dst 0x000b
I (146095) BLE_MESH: send, len 2: 8201
I (146095) BLE_MESH: Settings store, waiting 0 seconds
I (146535) BLE_MESH: Client message 0x00008201 with timeout 2000ms
[... 275 lines cut ...]
I (199855) CLOUD: addr=0057 type=onoff
I (199855) CLOUD: onoff=1
I (199855) BLE_MESH: Client message 0x00008202 with timeout 2000ms
I (199865) BLE_MESH: send, app_idx 0x0000 src 0x0913 dst 0x0057
I (199875) BLE_MESH: send, len 6: 8202011a0a0a
I (199905) BLE_MESH: Settings store, waiting 0 seconds
E (200275) REEVA_NVS: Store, nvs_set_blob failed, err 4357--ESP_ERR_NVS_NOT_ENOUGH_SPACE
E (200475) BLE_MESH: Failed to set mesh/seq data (err 4357)
nvs_stats Count: UsedEntries = (251), FreeEntries = (127), AllEntries = (378), namespace_count = (6)
I (200605) BLE_MESH: Settings store, waiting 0 seconds
E (200805) BLE_MESH: Failed to set mesh/rpl/0057 data (err 4357)
E (200805) BLE_MESH: Failed to store RPL entry 0x0057
I (200805) BLE_MESH: recv, app_idx 0x0000 src 0x0057 dst 0x0913
I (200815) BLE_MESH: recv, len 3: 820401
I (200815) REEVA_BLE: Generic client, event 1, error code 0, opcode is 0x8202
I (200825) REEVA_BLE: ESP_BLE_MESH_GENERIC_CLIENT_SET_STATE_EVT
I (200825) REEVA_BLE: ADDR : 0x0057
I (200835) REEVA_BLE: ESP_BLE_MESH_MODEL_OP_GEN_ONOFF_SET, onoff 1
I (200845) CLOUD: Topic: REEVA/BLE_MESH_GATEWAY/1C9DC2C25FC2/E/1
I (200845) CLOUD: Message: {"addr":87,"type":"onoff","onoff":1}
```

### More Information.

Also follow this [partition table](https://github.com/espressif/esp-idf/files/9413616/partitions.csv) and [sdkconfig](https://github.com/espressif/esp-idf/files/9413629/sdkconfig.txt)

JiaLWang, 2022-08-24:
Hi @hipal31 
it seems like nvs space is not enough. You can increase it and have a try.
menuconfig: Component config → ESP BLE Mesh Support -> Store BLE Mesh configuration persistently​.
Thanks.

hipal31, 2022-08-24:
Hi @JiaLWang 
I'll increase nvs size and test it.
I want to know, what should be the maximum size of nvs for storing ble-mesh data?

JiaLWang, 2022-08-25:
Hi @hipal31 
The information stored by provisioner for each node is about 60 bytes. (Without considering node name and node composition data.)

You can increase the size of the nvs of partitions.cvs. (The path refer to menuconfig: (Top) → Partition Table -> Custom partition CSV file).

[image]

Thanks.

hipal31, 2022-08-25:
Hi @JiaLWang 
I increased size of NVS to 300KB. And I think it'll work for me. However my client node is not a provisioner. So for a mesh node, how much size of data is stored for a unicast?

JiaLWang, 2022-08-25:
Hi @hipal31 
For a mesh node, the default value of nvs will be enough.

hipal31, 2022-08-25:
Hi @JiaLWang 
In my mesh node, I'm gettiing error : `BLE_MESH: Failed to set mesh/seq data`
As per  your comment, this error should not be there.

JiaLWang, 2022-08-25:
Hi @espressif-bot 
The device runs onoff_client example is a Provisioner, not a Node.

hipal31, 2022-08-25:
Hi @JiaLWang 
I'm not using config client in my node.

Here is list of elements that are present in my node.
```
static esp_ble_mesh_model_t root_models[] = {
    ESP_BLE_MESH_MODEL_CFG_SRV(&config_server),
    ESP_BLE_MESH_MODEL_GEN_ONOFF_CLI(&onoff_cli_pub, &onoff_client),
    ESP_BLE_MESH_MODEL_GEN_LEVEL_CLI(&level_cli_pub, &level_client),
    ESP_BLE_MESH_MODEL_LIGHT_LIGHTNESS_CLI(&light__cli_pub, &light_client),
    ESP_BLE_MESH_MODEL_LIGHT_CTL_CLI(&ctl_cli_pub, &ctl_client),
    ESP_BLE_MESH_MODEL_LIGHT_HSL_CLI(&rgb_cli_pub, &rgb_client),
};

static esp_ble_mesh_elem_t elements[] = {
    ESP_BLE_MESH_ELEMENT(0, root_models, vnd_models),
};
```

JiaLWang, 2022-08-25:
Hi @hipal31 
Maybe the nvs space using by others, not only mesh. But you can create a specific NVS partition for BLE Mesh in partitions.csv.
And enable the option in the menuconfig. 
`(Top) → Component config → ESP BLE Mesh Support -> Use a specific NVS partition for BLE Mesh`

hipal31, 2022-08-26:
Hi @JiaLWang,
I'll try that.