espressif/esp-idf #9623: 24-bit support for esp_lcd_rgb_panel_config_t::data_width

dmartauz, 2022-08-24:
### Describe the solution you'd like.

Ability to drive RGB displays with 24-bit data bus, According to documentation, only 8 and 16-bit buses are supported currently.

Example of such display: https://www.waveshare.com/10.1inch-Capacitive-Touch-LCD-D.htm

suda-morris, 2022-08-25:
Hi @dmartauz The ESP32S3 RGB LCD controller doesn't support 24bit data lines. It's not a software limitation. 

You can choose an LCD that supports the** SRGB(Serial RGB)** interface if you have to use the RGB888 color format.

There might be an ESP chip that supports up to 24 color data lines in the future.

dmartauz, 2022-08-25:
@suda-morris Thank you for clarification. I will stay with 16-bit displays for now :)

closed 2022-08-29 by suda-morris